mongoose-os-apps/shelly-homekit #364: Shelly 1L - 2 switches support

macmishka, 2020-12-26:
Hey.

It would be very handy if you could support both SW inputs of Shelly 1L. Currently the only SW1 is supported which makes it impossible to use the device with 2-way or 2-gang switches. I set my 2-way switch via Homebridge but I would be happy to make it more "natively" through the firmware.

macmishka, 2020-12-27:
Moreover, I'd prefer to use the second SW as detached switch for my Hue lamps, but it's impossible for now since SW2 isn't supported.

andyblac, 2020-12-27:
add your support here https://github.com/mongoose-os-apps/shelly-homekit/issues/346
as this is duplicate i'll close

closed 2020-12-27 by andyblac

rojer, 2020-12-27:
@macmishka actually, it is possible now - second switch is exported as a separate HAP accessory and can be configures as a stateless switch or one of the sensor types. so, in fact, detached mode is the only available mode for the second switch at the moment. what's missing is support for "attached to SW1" mode, so to speak.

freddair, 2021-01-11:
@rojer initially on fw 2.7.0 I had 2 inputs as stateless switches, but for some reason Input 2 is gone and now I only see SW1 (with input mode detached) and Input1... not sure if it was since 2.7.1 or since I set SW1 HAP mode to disabled... I now re-enabled HAP mode to "switch" but Input 2 is still not there... any idea?

andyblac, 2021-01-11:
> @rojer initially on fw 2.7.0 I had 2 inputs as stateless switches, but for some reason Input 2 is gone and now I only see SW1 (with input mode detached) and Input1... not sure if it was since 2.7.1 [...]

to use 'input2' you need to be on 'detatched'. if you dont need to relay you can set HAP to disabled
[image]

andyblac, 2021-01-11:
> > @rojer initially on fw 2.7.0 I had 2 inputs as stateless switches, but for some reason Input 2 is gone and now I only see SW1 (with input mode detached) and Input1... not sure if it was since 2.7.1 or since I set SW1 HAP mode to disabled... I now re-enabled HAP mode to "switch" but Input 2 is still not there... any idea?
> 
> to use 'input2' you need to be on 'detatched'. if you dont need the relay you can set HAP to disabled
> [image]

oh sorry, i see this is about 1L, let me test it.

andyblac, 2021-01-11:
@freddair you're correct i dont see any Switch2 to be configured.

freddair, 2021-01-15:
So what can we do to get Input 2 available again?
[image]

rojer, 2021-01-16:
sorry, looks like there's been a regression. i submitted a fix (https://github.com/mongoose-os-apps/shelly-homekit/commit/a1044b907ad5f02a8b0e3820494255691319a169), it will be in the next release.