mongoose-os-apps/shelly-homekit #365: 2.7.0 WiFi stability improvements

lucaspinelli85, 2020-12-28:
Hi, i don't understand well how work this improvements, because after updating to 2.7.0 some shelly remain connected to the furthest router, not the closest one with the highest power causing unreachability and disconnection (for example after reboot nearest router). I try to turn off and on shelly again, but it remains unreachable and wifi in webUI go to disabled, or connected to the furthest router!
So, my question is: when it scan the network to connect to the strongest SSID signal? Only when it turns on or every "x" time?
If it is already connected to a network, and subsequently will there a stronger one, does it notice and connect to the strongest or does it always remain connected to the previous one?
(SSID and password are always the same on all wifi network)

lucaspinelli85, 2021-01-08:
Any answer?

andyblac, 2021-01-08:
i think this down to the wifi SDK, as even shelly official firmare is having issue with WiFi aswell atm.

rojer, 2021-01-09:
@lucaspinelli85 2.7.0 and then 2.7.1 fixed a number of issues and updated SD but it has not solved the connectivity issues, it seems.
we are well aware of that and looking at ways to debug the problems.

hfagelnour, 2021-01-10:
Could u pls and as an interim (I know it’s a bad tackle) solution follow suite the newest stock firmware release by power cycling the devices retaining all settings if a device isn’t connected to the WiFi properly.

rojer, 2021-01-24:
this has been implemented in https://github.com/mongoose-os-apps/shelly-homekit/commit/f2ead0dddcbfe7e2efedaf3e4b2666a543b69389 and pushed out in 2.7.2

closed 2021-01-24 by rojer