mongoose-os-apps/shelly-homekit #366: 2.7.0 24V problem

belrpr, 2020-12-28:
Hi,

Since upgrading my Shelly 1 from 2.6.1 to 2.7.0 the shelly wouldn't boot on 24V.
Or not reliable at least. If I disconnect the power 10 times maybe 1 time it boot.

My shelly is connected exact like this youtube:
https://www.youtube.com/watch?v=eaD6lWh3r1s

Now I flashed back the 2.6.1 firmware and now it just boot normally every time.

When I use the 2.7.0 firmware with a adapter for 12V (And changing the jumper) it works also fine.
But in my configuration it is easier to tap the 24V from the garage door motor than using another adapter.

rojer, 2020-12-28:
interesting. can you try installing this firmware - http://rojer.me/files/shelly/2.6.1-newsdk/
it's a rebuild of 2.6.1 with new Espressif SDK
let's see if that was the reason for power instability

belrpr, 2020-12-28:
Ok will test this on a spare shelly.
Tried the firmware 2.7.0 on 2 different shelly's to rule out hardware problems.

marcus-aa, 2020-12-31:
Related:
https://github.com/mongoose-os-apps/shelly-homekit/issues/251#issuecomment-745298841

simonfield76, 2021-01-01:
Is the older firmware available also for Shelly 1?

Inviato da iPhone

jchristianj, 2021-01-03:
Same issue, at least related here: Shelly 1 as garage door opener, FW2.7.0. Device is supplied with 24VDC from the motor drive unit (specs say up to 50mA which is sufficient). 
Upon activating the Shelly (open or close command via HomeKit) the relay closes, the drive motor starts. This starting motor load causes the supply voltage to drop below 24VDC. Each time a command is given, the shelly reboots, all HomeKit connections are lost and HK notification is not reliable. So I can not  use it as a garage door opener supplied by the drive unit. BTW I downgraded to 2.6.1 and it is happening here as well.

jsiegenthaler, 2021-01-03:
Add a series diode (1N4007) and a big fat electrolytic capacitor (like a 4700uF or more) to the input of the Shelly1 to help stabilise sudden voltage drops. This may (or may not) be sufficient to protect the Shelly1 from the voltage surges that occur when the garage door motor starts. The bigger the capacitor, the longer it will hold a charge and the more it will buffer the Shelly1 supply voltage.

jchristianj, 2021-01-04:
Had an old 230VAC/12VDC power supply in my drawer. I changed the wiring and jumper to use the separate 12V supply, now it seems to be fine. I'll observe and let you know if there are still any issues. Thanks!

jsiegenthaler, 2021-01-04:
@jchristianj  that basically confirms the issues where due to voltage spikes caused by the motors switching on and off. Any microcontroller requires a clean and steady power supply... which is why I suggested the diode and capacitor as a filter

belrpr, 2021-01-13:
Sorry for the late feedback but 2.6.1 has it too but is a bit stabler.
How would you need to connect the diode and capcitor to make a stabler connection?

andyblac, 2021-03-03:
can this issue be closed now ?

jchristianj, 2021-03-03:
For me it’s ok.

belrpr, 2021-03-04:
Hi,

I can confirm that since I used an external 12V adapter the problem never occured.
So this may be closed and thx for the assistance!

closed 2021-03-04 by belrpr