mongoose-os-apps/shelly-homekit #368: Can't connect to shelly devices

leilok, 2020-12-29:
There is some devices I can't connect with, trough web interface (IP address).

I can make ping to the devices, but when I try to connect with safari, doesn't connect.

Any suggestion?

andyblac, 2020-12-29:
try cutting power to device to reboot

leilok, 2020-12-29:
> try cutting power to device to reboot

Tried already....

And reboot router...

The only one I can connect with, was the last one I installed...

I mean, the devices I want to connect with, has old HomeKit firmware version... and they are visible to "Home" iOS app... But I I wanted to update the firmware to avoid the problems of stability.

andyblac, 2020-12-29:
anyway you can move router closer ?

leilok, 2020-12-29:
> anyway you can move router closer ?

I have a wifi mesh system installed... and one of the devices is 1 m away from the master router... is not a problem of distance from de router...

Anyway, the device I can connect with is the one far away from this master router... in fact, this one is connected to a secondary node...

Thnaks foro your quickly reply!!!

andyblac, 2020-12-29:
that might be your issue, mesh systems sometime do not play nice with mDNS, check your router / mesh setting make sure multicast routing support is enabled.

leilok, 2020-12-29:
> that might be your issue, mesh systems sometime do not play nice with mDNS, check your router / mesh setting make sure multicast routing support is enabled.

No option to change nothing about mdns in my config!!!

Anyway... if there is a problem with mDNS, why one of the devices is available to access with web interface???

and the most important thing, for me (because I think that maybe is an issue of old firmware), how can I update the devices, if I can't do it with web interface???

The most strange for me is that I can make ping with Mac OS Terminal... but I can't upgrade with the rojer server firmware!!!!

andyblac, 2020-12-29:
> No option to change nothing about mdns in my config!!!
>
> Anyway... if there is a problem with mDNS, why one of the devices is available to access with web interface???
>
> and the most important thing, [...]

ping is a low level reply (handled by drivers). that why suguested you move min router closer so it connects directly with it, you should thn beable to access the webUI.

leilok, 2020-12-29:
One question... when I installed the 2 devices I can’t connect with, I made it with my old wifi router...
Some time later, I installed my wifi mesh network, but I used the same wifi name and password, so the Shelly connected to the wifi mesh network directly.... 
maybe that is the reason I can’t connect to the Shelly?

andyblac, 2020-12-29:
> One question... when I installed the 2 devices I can’t connect with, I made it with my old wifi router...
> Some time later, I installed my wifi mesh network, but I used the same wifi name and [...]

as long as you used the same SSID and passwword, should not be an issue.

andyblac, 2021-01-25:
is this still an issue, or can it be closed ?

leilok, 2021-01-25:
Same problem... not solved yet.

andyblac, 2021-01-26:
what exactly happens in web browser?

leilok, 2021-01-27:
I can’t connect to Shelly device.... IP address in the web browser doesn’t
open the HomeKit web interface.

andyblac, 2021-01-27:
did you try my script ? see it that see's all your devices.

leilok, 2021-01-27:
> did you try my script ? see it that see's all your devices.

I didn't know the exist of your script....
I will try this afternoon and reply here asap
Thanks!!!

leilok, 2021-01-27:
> did you try my script ? see it that see's all your devices.

I tried your script... and only found the 3 devices accesible by web browser... one with Homekit version, 2 with stock version.

I used your script to update all of them the the last version...

But the script didn't find the 2 devices I can't access to....

what can I do now???

Thanks for your time, Andy!!!

andyblac, 2021-01-27:
> I tried your script... and only found the 3 devices accesible by web browser... one with Homekit version, 2 with stock version.
>
> I used your script to update all of them the the last version...
>
> But [...]

hmm, so lets recap the 2 devices that are older firmware are still seen in HomeKit, just now by the webUI or by the script, is that correct ?, if so what version of the firmware are they on ?

leilok, 2021-01-28:
> > > did you try my script ? see it that see's all your devices.
> > 
> > 
> > I tried your script... and only found the 3 devices accesible by web browser... one with Homekit version, 2 with stock version.
> > I used your script to update all of them the the last version...
> > But the script didn't find the 2 devices I can't access to....
> > what can I do now???
> > Thanks for your time, Andy!!!
> 
> hmm, so lets recap the 2 devices that are older firmware are still seen in HomeKit, just not by the webUI or by the script, is that correct ?, if so what version of the firmware are they on ?
> 
> can you open Home app, long press on the accessory and go to settings, scroll right to bottom can take a screen grab of the last bit at bottom please.

that's correct... one device is Shelly 1, and another one is Shelly 2.5... 

Please check version in the attached pictures
[image]
[image]

andyblac, 2021-01-28:
i'm not sure what that firmware is, but it is this firmware. as it say's version 3.2, we are only 2.7.2 atm, are you sure you did not flash another firmware?

any way, atm i think the only way to update / reflash to this firmware is to do via serial connection see here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection

leilok, 2021-01-28:
> i'm not sure what that firmware is, but it is not this firmware. as it say's version 3.2, we are only 2.7.2 atm also in Shelly1 the screenshot says "SHelly1" there is no typo in this firmware, are you sure you did not flash another firmware?
> 
> any way, atm i think the only way to update / reflash to this firmware is to do via serial connection see here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection

Sure I didn't install another version... because I thought this HomeKit firmware was the only one available to connect shelly to Homekit

andyblac, 2021-01-28:
> Sure I didn't install another version... because I thought this HomeKit firmware was the only one available to connect shelly to Homekit

no there is a few that does it.

andyblac, 2021-01-28:
i think those 2 have this firmware https://github.com/RavenSystem/esp-homekit-devices

leilok, 2021-01-28:
sorry... I was thinking on it, and maybe you were right... Let me explain.

The first order I placed to buy shelly devices was in a website who sell the devices with the HomeKit firmware installed. 

In the attached picture is the device I updated yesterday with your script.

The next order I bought them with the stock version, and I installed "your" HomeKit version... so probably the firmware wich was included in those devices are not your firmware.

Anyway... is there any way to
[image]
 take the device to AP mode, and try to connect to them directly, and try to install a "your" HomeKit firmware?

andyblac, 2021-01-28:
ye, the 1st 2 you bought, has HAA on them, the only way to flash them with this firmware is via serial connection.

leilok, 2021-01-28:
Ok... so this issue is solved...

So I have to buy a connector... I am trying to understand wich one do I have to buy, but I don't understand pretty much...

Or maybe is easy to me forget both of them and install new one!!!!

lol

Thanks Andy!!!

andyblac, 2021-01-28:
here something like this https://www.amazon.it/DSD-TECH-Adattatore-seriale-Compatibile/dp/B07BBPX8B8/ref=sr_1_1_sspa?__mk_it_IT=ÅMÅŽÕÑ&dchild=1&keywords=DSD+TECH+USB+to+TTL&qid=1611829743&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExN0FQS0taVUMxNFZFJmVuY3J5cHRlZElkPUExMDI1MjY0MVNOSzA4T0tVTUZDTCZlbmNyeXB0ZWRBZElkPUEwMzc3NTMxMjNLMkQ4QTY2TUVTQyZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

but DO NOT connect it to MAIN POWER whilst flashing.

andyblac, 2021-01-28:
oh and cable https://www.amazon.it/Ganvol-40-Pezzi-solderless-breadboard-Raspberry/dp/B01LVVIOUO/ref=sr_1_7_sspa?__mk_it_IT=ÅMÅŽÕÑ&dchild=1&keywords=male+to+Female+Breadboard+Jumper+Wires&qid=1611829952&s=electronics&sr=1-7-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzV0ZIVDIwQlQwSkUmZW5jcnlwdGVkSWQ9QTAwMDkyOTlRRk9aOU05S1BFTUUmZW5jcnlwdGVkQWRJZD1BMDc0NTAxNjI1RTlDOThFNEY0OVkmd2lkZ2V0TmFtZT1zcF9tdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

andyblac, 2021-01-28:
i'll close this issue, but if you need help flashing etc, please feel free to open a new issue 😄

closed 2021-01-28 by andyblac