mongoose-os-apps/shelly-homekit #369: Shelly Duo Support

hughfr4nc15, 2020-12-30:
Hello,

I have several Shelly Relay Switches running flawlessly with your firmware. (thanks for that!)
Today I bought some Shelly Duo bulbs and cannot make them work via Homebridge plugin... It would be great if you guys released your firmware for the Shelly Duo bulbs.... any plans for that?

Thanks in advance and Happy Holidays!

ari-ido-999, 2021-01-24:
Hello, supporting shelly bulbs will be excellent

zeltak, 2021-03-04:
would also love this!

Z

berkus, 2021-03-24:
Coming from #579, would also love to see lightbulbs support!

appliko, 2021-07-23:
Buongiorno, any update on Duo and Vintage bulbs support ?

hughfr4nc15, 2021-09-26:
Hello. Is there any planned release date for bulb support?
Thanks in advance!

juangamnik, 2021-11-13:
I would appreciate support too. Especially for the Shelly duo rgbw gu10… since the homebridge plug-in I currently use, does not support white colors, so that you can only use the color LEDs, which are not as bright (using 1.5W instead of the 4.7W of the white color).

GizaButler83, 2021-12-08:
Same here, can only control RGB in Homekit. White Channel is now working correctly and can only be used via shelly App/Browser GUI

timoschilling, 2022-01-18:
Will be implement in #885

closed 2022-01-25 by rojer

rojer, 2022-01-26:
pushed in 2.11.0-alpha9, please help test it (#868).

PR #885 (2022-01-25): Bulb support
Add Support for Shelly Bulbs.

- [x] Duo (closes #369)
- [x] Color Bulb (closes #823)
- [x] Vintage (closes #719)
- [x] ~remove initial state input setting~ (done by #853)

I have tested this code on Duo E27, Vintage A60, Color Bulb E27 /  GU10.
Shelly Support says that the E27 and GU10 have the same pinout and all Vintages have the same pinout.