mongoose-os-apps/shelly-homekit #371: Shelly 1 Unresponsive

notorioustubze, 2020-12-31:
Hi, woke up this morning to one of my devices showing as unresponsive so I did all the usual: Killed the power, reset the router but no joy. So I've reset the shelly to ap mode and can connect using the default ip and can see the Homekit set up screen as you would normally, I configure my wifi settings and the it says the shelly is rebooting but It never connects to my network. Obviously I can't see the return to stock option as there is no wifi and I can't hard reset it as it doesn't have the reset button like other devices. Is there anything else I can try?

andyblac, 2020-12-31:
sounds like wifi issues, are you on a mesh wifi ?, try moving your main router closer to the device temporarily to see if that helps.

andyblac, 2020-12-31:
also try readig this https://github.com/mongoose-os-apps/shelly-homekit/wiki/Trouble-Shooting

notorioustubze, 2020-12-31:
Hi Andy, popped the Shelly out the ceiling and it went through straight away. Sorry to have wasted your time.

andyblac, 2020-12-31:
> Hi Andy, popped the Shelly out the ceiling and it went through straight away. Sorry to have wasted your time.
> […](#)

np, i am here to help.

closed 2020-12-31 by andyblac