mongoose-os-apps/shelly-homekit #372: Impossible connection to Homekit with firmware 2.7.0

sebpi33, 2020-12-31:
Hi,

I try without success to connect new Shelly 25 to my Homekit with firmware version 2.7.0 (build 20201222-230927/2.7.0-g1041f8b-master).
I already have to 2 Shellly 25 added recently (last month) in firmware version 2.6.1 (build 20201128-161048/2.6.1-g3ba3275-master).

With this new version during Homekit add accessory process it too much longer to find my shelly 25 than the previous adding that I performed with firmware 2.6.1. And after setting up the code Homekit work and after 1 or 2 minutes give me up a status like this "No possible to connect to this accessory. This accessory is not reachable".

My iPhone is connect to the same wifi (2.4GHz => 2 bands 2.4GHz and 5GHz are not on the same network) and both are connecting to the same wifi routeur (I have several routeurs to establish my local network).
I try to rerun my phone and the shelly 25 relay. Same behaviour :(
Nevertheless no worries to access to embedded server of the relay (with my iPhone and with my Mac).

Any ideas ? Thanks in advance for helping.
SeB

sebpi33, 2020-12-31:
After downgrade to firmware 2.6.1 no worries it works ! Issue with new version ?

rojer, 2020-12-31:
this is very interesting, i wish i could see console logs from the device that is connecting... but it's kind of hard to obtain.
maybe if you do the following very quickly (so log is not rotated):
1) go to AP mode, connect
2) enable debug log
3) enabled wifi and have the device try to connect
4) give it some time, then return to AP mode by holding the physical button for 10 seconds
5) connect to AP and download the debug log (via link in the UI)

hopefully the log wouldn't have rotated and we should see connection attempts and error codes

sebpi33, 2021-01-01:
Hi rojer,

First of all I wish you an happy New Year :)
Sorry but what is "AP mode" ?

andyblac, 2021-01-01:
> Sorry but what is "AP mode" ?

when you reset the device and you can connect to directly

andyblac, 2021-01-25:
still need assistance ? or this issue be closed ?

andyblac, 2021-01-31:
as the OP has not reponded i'll close the issue.

closed 2021-01-31 by andyblac