mongoose-os-apps/shelly-homekit #374: 2.7 issues

singas77, 2020-12-31:
closed roller shutter is actually open 
activation results in shelly relay clicking but no actuations 
how can I go back to 2.6 that worked ?

andyblac, 2020-12-31:
get from here https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.6.1 upload via webUI

singas77, 2020-12-31:
how to http for shelly http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip

andyblac, 2020-12-31:
https://github.com/mongoose-os-apps/shelly-homekit/wiki/System-Settings

singas77, 2020-12-31:
ill try that was working perfect until update meeeeow

singas77, 2020-12-31:
-1 Archived file size is unknown

singas77, 2020-12-31:
got it

andyblac, 2020-12-31:
> got it

working again ?

singas77, 2020-12-31:
ive gone back to 2.6 ill test it now

singas77, 2020-12-31:
not really hear shelly clicking but no motion

andyblac, 2020-12-31:
> not really hear shelly clicking but no motion

you may need to recalibrate it.

singas77, 2020-12-31:
calibration never worked correcttly for me 50% open was more like 80% shut hit calibrate now all I hear is click click click have to reboot

singas77, 2020-12-31:
was working so good before ??

andyblac, 2020-12-31:
> was working so good before ??

sorry, i dont have / used a roller door setup, you'll hve to wait for someone with more knowledge to advise.

singas77, 2020-12-31:
yeah cheers don't know what's going on cheers

andyblac, 2020-12-31:
I have renamed issue, to try to get people that have used Roller Shutters to catch there attention.

rojer, 2020-12-31:
nothing really changed from 2.6 to 2.7 wrt roller shutter mode, so it's strange.
nevertheless, we'll need logs. enable debug logs and post some here.

singas77, 2020-12-31:
85999258 shelly_hap_window_c:531 P = 0.00 -> 0.00
86008870 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
86133811 shelly_output.cpp:56    Output 2: on -> off (moving)
86143848 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
86256001 mgos_sys_config.c:232   Loading conf2.json
86279337 mgos_sys_config.c:232   Loading conf3.json
86412251 mgos_sys_config.c:174   Saved to conf9.json
86457630 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
86476844 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
86575058 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
86674168 shelly_output.cpp:56    Output 1: off -> on (move)
86684245 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
86774471 shelly_hap_window_c:531 P = 0.00 -> 0.00
86784523 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
86876891 shelly_hap_window_c:329 WC 1: Tgt pos 50.00 -> 50.51 (fixup)
86885986 shelly_output.cpp:56    Output 1: on -> off (moving)
86896982 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
86998419 mgos_sys_config.c:232   Loading conf2.json
87021756 mgos_sys_config.c:232   Loading conf3.json
87154839 mgos_sys_config.c:174   Saved to conf9.json
87171877 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
87190527 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
88601279 shelly_main.cpp:554     Up 88.59, HAP 0/2/9 ns 2, RAM: 22580/41008; st 81; 4.1: c:1 mp:0.00 mt_ms:1081 cp:50.51 tp:50.51 lemd:0 lhmd:0
95098272 shelly_debug.cpp:203    Tailing
96601014 shelly_main.cpp:554     Up 96.59, HAP 0/2/9 ns 2, RAM: 22572/41008; st 81; 4.1: c:1 mp:0.00 mt_ms:1081 cp:50.51 tp:50.51 lemd:0 lhmd:0
104601124 shelly_main.cpp:554     Up 104.59, HAP 0/2/9 ns 2, RAM: 22572/41008; st 81; 4.1: c:1 mp:0.00 mt_ms:1081 cp:50.51 tp:50.51 lemd:0 lhmd:0
112601153 shelly_main.cpp:554     Up 112.59, HAP 0/2/9 ns 2, RAM: 22572/41008; st 80; 4.1: c:1 mp:0.00 mt_ms:1081 cp:50.51 tp:50.51 lemd:0 lhmd:0

rojer, 2020-12-31:
ok, i don't see anything inherently wrong here - moved to 50% position as instructed (actual position 50.51, within the margin of error). what did you expect to happen?

singas77, 2020-12-31:
can't move at all either from switch or web ui ?

singas77, 2020-12-31:
maybe its shelly ? 
do calibrate it goes cal1 pos1 then stop can't even use manual switch

singas77, 2020-12-31:
4452031 esp_main.c:158          SDK: state: 2 -> 3 (0)
4463156 esp_main.c:158          SDK: state: 3 -> 5 (10)
4470707 esp_main.c:158          SDK: add 0
4478091 esp_main.c:158          SDK: aid 4
4485667 esp_main.c:158          SDK: cnt 
4514006 esp_main.c:158          SDK: 
4524713 esp_main.c:158          SDK: connected with Wifi4, channel 10
4537544 esp_main.c:158          SDK: ip:192.168.1.99,mask:255.255.255.0,gw:192.168.1.1
4549348 esp_main.c:158          SDK: ip:192.168.1.99,mask:255.255.255.0,gw:192.168.1.1
4564787 mgos_wifi.c:133         WiFi STA: Connected, BSSID 8c:15:c7:da:72:74 ch 10 RSSI -48
4616824 mgos_net.c:90           WiFi STA: connected
4632320 mgos_net.c:101          WiFi STA: ready, IP 192.168.1.99, GW 192.168.1.1, DNS 192.168.1.1
4649806 mgos_mongoose.c:66      New heap free LWM: 23592
4664452 mgos_http_server.c:180  0x3fff5b34 HTTP connection from 192.168.1.98:64217
4680915 shelly_debug.cpp:203    Tailing
4699614 mgos_mongoose.c:66      New heap free LWM: 21648
5034073 HAPPlatformTCPStrea:120 0x3fff5e0c 192.168.1.13:51048 Incoming HAP connection, ns 0/0/9
5049400 HAPPlatformTCPStrea:274 0x3fff5e0c 192.168.1.13:51048 Accepted HAP connection, ns 0/1/9 ts 0x3fff2b34
6110891 mgos_sntp.c:95          SNTP query to time.google.com
6129707 mgos_mongoose.c:66      New heap free LWM: 20040
6142947 mgos_mongoose.c:66      New heap free LWM: 19904
6157535 mgos_mongoose.c:66      New heap free LWM: 18856
6403953 mgos_sntp.c:58          SNTP reply from 216.239.35.8: time 1609430806.519205, local 6.391491, delta 1609430800.127714
8678921 shelly_main.cpp:554     Up 8.67, HAP 0/1/9 ns 1, RAM: 24396/41008; st 73; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
14486415 esp_main.c:158          SDK: pm open,type:0 0
16678889 shelly_main.cpp:554     Up 16.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 72; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
24678775 shelly_main.cpp:554     Up 24.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 72; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0

singas77, 2020-12-31:
that's calibrate shelly clicking no movement

rojer, 2020-12-31:
looks like this is taken before calibration is initiated? i don't see any sign of calibration or movement here.

singas77, 2020-12-31:
8679856:1609430808789802 shelly_main.cpp:554     Up 8.67, HAP 0/1/9 ns 1, RAM: 24396/41008; st 73; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
14487348 esp_main.c:158          SDK: pm open,type:0 0
16679853 shelly_main.cpp:554     Up 16.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 72; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
24679740 shelly_main.cpp:554     Up 24.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 72; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
32679754 shelly_main.cpp:554     Up 32.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 71; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
40679783 shelly_main.cpp:554     Up 40.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 71; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
48679719 shelly_main.cpp:554     Up 48.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 71; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
56679739 shelly_main.cpp:554     Up 56.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 70; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
64680604 shelly_main.cpp:554     Up 64.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 70; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
72679910 shelly_main.cpp:554     Up 72.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 70; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
80680066 shelly_main.cpp:554     Up 80.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 69; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
88679717 shelly_main.cpp:554     Up 88.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 69; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
96679778 shelly_main.cpp:554     Up 96.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 69; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
104679876 shelly_main.cpp:554     Up 104.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 69; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
112679869 shelly_main.cpp:554     Up 112.67, HAP 0/1/9 ns 1, RAM: 23892/41008; st 68; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
119329250 mgos_http_server.c:180  0x3fff5f9c HTTP connection from 192.168.1.98:64395
119340969 mgos_http_server.c:180  0x3fff41cc HTTP connection from 192.168.1.98:64396
119415329 shelly_debug.cpp:203    Tailing
120680182 shelly_main.cpp:554     Up 120.67, HAP 0/1/9 ns 1, RAM: 23016/41008; st 68; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
122652950 mgos_http_server.c:187  0x3fff5f9c GET /
122695780 mgos_http_server.c:187  0x3fff5f9c GET /axios.min.js.gz
122722300 mgos_http_server.c:180  0x3fff60cc HTTP connection from 192.168.1.98:64398
122734239 mgos_http_server.c:187  0x3fff60cc GET /logo.png
122773018 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.98:64395
122794859 mgos_mongoose.c:66      New heap free LWM: 17992
128679201 shelly_main.cpp:554     Up 128.67, HAP 0/1/9 ns 1, RAM: 23276/41008; st 67; 4.1: c:1 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
129329910 mgos_http_server.c:180  0x3fff5b34 HTTP connection from 192.168.1.98:64399
129346192 mg_rpc.c:305            Shelly.SetConfig via HTTP 192.168.1.98:64399
129376564 mgos_sys_config.c:232   Loading conf2.json
129400869 mgos_sys_config.c:232   Loading conf3.json
129585647 mgos_sys_config.c:174   Saved to conf9.json
129612451 shelly_hap_window_c:309 WC 1: State: idle -> precal0 (0 -> 10)
129710646 shelly_hap_window_c:431 Begin calibration
129739349 mgos_sys_config.c:232   Loading conf2.json
129763592 mgos_sys_config.c:232   Loading conf3.json
129894719 mgos_sys_config.c:174   Saved to conf9.json
129954648 shelly_output.cpp:56    Output 1: off -> on (precal0)
129964729 shelly_hap_window_c:309 WC 1: State: precal0 -> cal0 (10 -> 11)
129979856 shelly_hap_window_c:447 WC 1: P0 = 6.433
130734337 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.98:64399
130780344 shelly_hap_window_c:447 WC 1: P0 = 226.957
131580053 shelly_hap_window_c:447 WC 1: P0 = 227.738
131779821 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.98:64399
131801020 mgos_mongoose.c:66      New heap free LWM: 17928
132379977 shelly_hap_window_c:447 WC 1: P0 = 227.665
132775082 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.98:64399
133180258 shelly_hap_window_c:447 WC 1: P0 = 227.823
133779139 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.98:64399
133980042 shelly_hap_window_c:447 WC 1: P0 = 227.463
134512555 shelly_debug.cpp:203    Tailing
134779485 shelly_hap_window_c:447 WC 1: P0 = 227.573
135579723 shelly_hap_window_c:447 WC 1: P0 = 227.311
136379854 shelly_hap_window_c:447 WC 1: P0 = 227.915
136679155 shelly_main.cpp:554     Up 136.67, HAP 0/1/9 ns 1, RAM: 22632/41008; st 67; 4.1: c:0 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
137179650 shelly_hap_window_c:447 WC 1: P0 = 227.768
137979520 shelly_hap_window_c:447 WC 1: P0 = 228.146
138779568 shelly_hap_window_c:447 WC 1: P0 = 228.165
139579539 shelly_hap_window_c:447 WC 1: P0 = 227.872
140379739 shelly_hap_window_c:447 WC 1: P0 = 228.061
141179774 shelly_hap_window_c:447 WC 1: P0 = 228.213
141980198 shelly_hap_window_c:447 WC 1: P0 = 227.768
142779462 shelly_hap_window_c:447 WC 1: P0 = 227.634
143579769 shelly_hap_window_c:447 WC 1: P0 = 228.110
144379622 shelly_hap_window_c:447 WC 1: P0 = 227.610
144679099 shelly_main.cpp:554     Up 144.67, HAP 0/1/9 ns 1, RAM: 22632/41008; st 67; 4.1: c:0 mp:0.00 mt_ms:1080 cp:46.39 tp:46.39 lemd:0 lhmd:0
145179631 shelly_hap_window_c:447 WC 1: P0 = 228.085
[... cut ...]
196672624 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
196773328 mgos_sys_config.c:232   Loading conf2.json
196797284 mgos_sys_config.c:232   Loading conf3.json
196966039 mgos_sys_config.c:174   Saved to conf9.json
196983886 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
197003789 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
200679640 shelly_main.cpp:554     Up 200.67, HAP 0/1/9 ns 1, RAM: 23612/41008; st 67; 4.1: c:1 mp:231.64 mt_ms:32680 cp:49.50 tp:49.50 lemd:0 lhmd:0

singas77, 2020-12-31:
that's calibration up down then tries 2nd up stops 1/3 open now can't use manual switch or web ui to do ant operations

singas77, 2020-12-31:
915530938 shelly_hap_window_c:352 WC 1: HAPSetTgtPos 81.00 cur 0.00 tgt 0.00 lhmd 0
915540981 shelly_hap_window_c:329 WC 1: Tgt pos 0.00 -> 81.00 (HAP)
915552207 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
915582505 shelly_output.cpp:56    Output 1: off -> on (move)
915593104 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
915681766 shelly_hap_window_c:531 P = 180.63 -> 0.00
915691878 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
915783543 shelly_hap_window_c:318 WC 1: Cur pos 0.00 -> 7.40, P = 228.26
915794417 shelly_hap_window_c:568 Obstruction: p = 228.26 t = 80
915805210 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
915880864 shelly_output.cpp:56    Output 1: on -> off (stop)
915918055 mgos_sys_config.c:232   Loading conf2.json
915984982 mgos_sys_config.c:232   Loading conf3.json
916123104 mgos_sys_config.c:174   Saved to conf9.json
916140192 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
916156696 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
916255058 shelly_hap_window_c:309 WC 1: State: idle -> error (0 -> 100)
916355425 shelly_hap_window_c:329 WC 1: Tgt pos 81.00 -> 7.40 (error)
916366036 shelly_hap_window_c:309 WC 1: State: error -> idle (100 -> 0)
917396865 shelly_hap_window_c:352 WC 1: HAPSetTgtPos 0.00 cur 7.40 tgt 7.40 lhmd 0
917407757 shelly_hap_window_c:329 WC 1: Tgt pos 7.40 -> 0.00 (HAP)
917418031 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
917454878 shelly_output.cpp:56    Output 2: off -> on (move)
917465645 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
917554872 shelly_hap_window_c:531 P = 180.51 -> 0.00
917564975 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
917657274 shelly_hap_window_c:568 Obstruction: p = 232.54 t = 80
917667162 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
917753984 shelly_output.cpp:56    Output 2: on -> off (stop)
917783416 mgos_sys_config.c:232   Loading conf2.json
917806883 mgos_sys_config.c:232   Loading conf3.json
917981800 mgos_sys_config.c:174   Saved to conf9.json
917999863 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
918024083 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
918122503 shelly_hap_window_c:309 WC 1: State: idle -> error (0 -> 100)
918222820 shelly_hap_window_c:309 WC 1: State: error -> idle (100 -> 0)
922798304 shelly_main.cpp:554     Up 922.78, HAP 0/8/9 ns 8, RAM: 21396/41008; st 64; 4.1: c:1 mp:0.00 mt_ms:1081 cp:0.00 tp:0.00 lemd:0 lhmd:2
927148250 shelly_hap_window_c:352 WC 1: HAPSetTgtPos 71.00 cur 0.00 tgt 0.00 lhmd 2
927200003 shelly_hap_window_c:329 WC 1: Tgt pos 0.00 -> 71.00 (HAP)
927210298 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
927234092 shelly_output.cpp:56    Output 1: off -> on (move)
927244942 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
927322326 shelly_hap_window_c:531 P = 160.55 -> 0.00
927332457 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
927424249 shelly_hap_window_c:568 Obstruction: p = 228.07 t = 80
927434155 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
927521334 shelly_output.cpp:56    Output 1: on -> off (stop)
927550014 mgos_sys_config.c:232   Loading conf2.json
927573421 mgos_sys_config.c:232   Loading conf3.json
927749487 mgos_sys_config.c:174   Saved to conf9.json
927767555 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
927791223 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
927889524 shelly_hap_window_c:309 WC 1: State: idle -> error (0 -> 100)
927989951 shelly_hap_window_c:329 WC 1: Tgt pos 71.00 -> 7.40 (error)
928000578 shelly_hap_window_c:309 WC 1: State: error -> idle (100 -> 0)
929268524 shelly_hap_window_c:352 WC 1: HAPSetTgtPos 0.00 cur 7.40 tgt 7.40 lhmd 0
929278547 shelly_hap_window_c:329 WC 1: Tgt pos 7.40 -> 0.00 (HAP)
929324473 shelly_hap_window_c:309 WC 1: State: idle -> move (0 -> 20)
929349512 shelly_output.cpp:56    Output 2: off -> on (move)
929360314 shelly_hap_window_c:309 WC 1: State: move -> rampup (20 -> 22)
929390526 shelly_hap_window_c:531 P = 40.00 -> 0.00
929400659 shelly_hap_window_c:309 WC 1: State: rampup -> moving (22 -> 23)
929491126 shelly_hap_window_c:568 Obstruction: p = 220.37 t = 78
929501041 shelly_hap_window_c:309 WC 1: State: moving -> stop (23 -> 24)
929588458 shelly_output.cpp:56    Output 2: on -> off (stop)
929616930 mgos_sys_config.c:232   Loading conf2.json
929640394 mgos_sys_config.c:232   Loading conf3.json
929810148 mgos_sys_config.c:174   Saved to conf9.json
929827263 shelly_hap_window_c:309 WC 1: State: stop -> stopping (24 -> 25)
929849438 shelly_hap_window_c:309 WC 1: State: stopping -> idle (25 -> 0)
[... 159 lines cut ...]
1031081587 shelly_hap_window_c:318 WC 1: Cur pos 22.88 -> 23.19, P = 231.82
1031881738 shelly_hap_window_c:318 WC 1: Cur pos 25.33 -> 25.64, P = 231.69
1032681561 shelly_hap_window_c:318 WC 1: Cur pos 27.78 -> 28.09, P = 230.99
1033481549 shelly_hap_window_c:318 WC 1: Cur pos 30.23 -> 30.53, P = 231.68
1034285123 shelly_hap_window_c:318 WC 1: Cur pos 32.68 -> 32.99, P = 231.84
1034798871 shelly_main.cpp:554     Up 1034.79, HAP 0/8/9 ns 8, RAM: 20604/41008; st 67; 4.1: c:1 mp:235.09 mt_ms:32679 cp:34.51 tp:50.00 lemd:0 lhmd:2
1035081463 shelly_hap_window_c:318 WC 1: Cur pos 35.12 -> 35.43, P = 231.69
1035881553 shelly_hap_window_c:318 WC 1: Cur pos 37.57 -> 37.88, P = 231.62
[... cut ...]
1050798557 shelly_main.cpp:554     Up 1050.79, HAP 0/8/9 ns 8, RAM: 20656/41008; st 67; 4.1: c:1 mp:235.09 mt_ms:32679 cp:49.51 tp:49.51 lemd:0 lhmd:2
1058798972 shelly_main.cpp:554     Up 1058.79, HAP 0/8/9 ns 8, RAM: 20656/41008; st 67; 4.1: c:1 mp:235.09 mt_ms:32679 cp:49.51 tp:49.51 lemd:0 lhmd:2
1066798738 shelly_main.cpp:554     Up 1066.79, HAP 0/8/9 ns 8, RAM: 20656/41008; st 67; 4.1: c:1 mp:235.09 mt_ms:32679 cp:49.51 tp:49.51 lemd:0 lhmd:2
1074798705 shelly_main.cpp:554     Up 1074.79, HAP 0/8/9 ns 8, RAM: 20656/41008; st 67; 4.1: c:1 mp:235.09 mt_ms:32679 cp:49.51 tp:49.51 lemd:0 lhmd:2

rojer, 2020-12-31:
ok, this provides more detail.
in the first log i see successful calibration - can you please confirm that during calibration it reaches both fully closed and fully open position?
you say after calibration it stops at about 30% open - it's supposed to stop at 50 but i'm not so concerned about that, we know there are accuracy issues with positioning (there's a separate issue about that).

in the second log i see failed movement attempts due to activation of obstruction detection safety check. but it doesn't make much sense, since the power value is well within the margin of calibrated average power value.
then there is another successful calibration but no more movement  attempts, so i'm not entirely sure what the calibrated power value is.

can you please collect and post an uninterrupted log of full calibration plus positioning attempts immediately after that?

singas77, 2020-12-31:
yes up / down then 3rd attempt stops 30%

singas77, 2020-12-31:
[image]

I can not

singas77, 2020-12-31:
[image]

singas77, 2020-12-31:
[image]

can't do anything now?

singas77, 2020-12-31:
but it bloody working lmfao 

unreal

andyblac, 2020-12-31:
> but it bloody working lmfao
> 
> unreal

looked like shelly was not fully booted up, in above pic

singas77, 2020-12-31:
won't let me from web might have to cut the power 
PS Happy new year everyone

andyblac, 2020-12-31:
you too , 3 miniets till BREXIT !! pmsl, 1hr 3mins till new year

singas77, 2020-12-31:
im starting to think its the shelly
yeah im about to have breakfast now in Australia lokl

andyblac, 2020-12-31:
nice, can i have a bacon butty please 😄

singas77, 2020-12-31:
im leaving this for now we got stuff on today ill get the logs when I can thanks for everyones assists

andyblac, 2021-01-25:
@singas77 any updates ?

singas77, 2021-01-25:
Sorry I work away from home I will grab a log and upload in a few days 
I reverted firmware back from latest and had no issues thou

singas77, 2021-02-05:
202455586 shelly_rpc_service.:223 SetConfig ok, 0
202486857 mgos_sys_config.c:232   Loading conf2.json
202510871 mgos_sys_config.c:232   Loading conf3.json
202644829 mgos_sys_config.c:174   Saved to conf9.json
202670164 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
202891746 shelly_hap_window_c:491 WC 1: P1 = 237.799
203176900 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
203691493 shelly_hap_window_c:491 WC 1: P1 = 237.921
203973998 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
204177545 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
204491492 shelly_hap_window_c:491 WC 1: P1 = 237.494
205178329 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
205291486 shelly_hap_window_c:491 WC 1: P1 = 238.006
206091392 shelly_hap_window_c:491 WC 1: P1 = 237.811
206179059 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
206892016 shelly_hap_window_c:491 WC 1: P1 = 238.543
207180904 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
207691439 shelly_hap_window_c:491 WC 1: P1 = 239.226
208181119 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
208491952 shelly_hap_window_c:491 WC 1: P1 = 239.000
209180228 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
209292054 shelly_hap_window_c:491 WC 1: P1 = 239.104
209321393 shelly_main.cpp:543     Up 209.31, HAP 0/7/12 ns 7, RAM: 16680/37616; st 58; 4.1: c:0 mp:235.09 mt_ms:32679 cp:100.00 tp:100.00 lemd:0 lhmd:0
210091295 shelly_hap_window_c:491 WC 1: P1 = 238.829
210181183 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
210891400 shelly_hap_window_c:491 WC 1: P1 = 239.232
211181109 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
211691512 shelly_hap_window_c:491 WC 1: P1 = 239.445
212181939 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55266
212491546 shelly_hap_window_c:491 WC 1: P1 = 239.476
212842316 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55243
213291345 shelly_hap_window_c:491 WC 1: P1 = 239.469
213843436 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55243
214091440 shelly_hap_window_c:491 WC 1: P1 = 239.628
214842834 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55243
214891965 shelly_hap_window_c:491 WC 1: P1 = 239.768
215691433 shelly_hap_window_c:491 WC 1: P1 = 239.512
215843898 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.98:55243
216163670 mgos_http_server.c:180  0x3fff8134 HTTP connection from 192.168.1.98:55272
216263445 shelly_debug.cpp:203    Tailing
216490895 shelly_hap_window_c:491 WC 1: P1 = 239.634
217290935 shelly_hap_window_c:491 WC 1: P1 = 239.043
217320934 shelly_main.cpp:543     Up 217.31, HAP 0/7/12 ns 7, RAM: 16104/37616; st 59; 4.1: c:0 mp:235.09 mt_ms:32679 cp:100.00 tp:100.00 lemd:0 lhmd:0
218090905 shelly_hap_window_c:491 WC 1: P1 = 238.884
218891055 shelly_hap_window_c:491 WC 1: P1 = 238.598
219690965 shelly_hap_window_c:491 WC 1: P1 = 238.073
220391286 shelly_output.cpp:56    Output 2: on -> off (cal1)
220404057 shelly_hap_window_c:495 WC 1: calibration done, move_time 33179, move_power 236.926
220414914 shelly_hap_window_c:322 WC 1: State: cal1 -> postcal1 (14 -> 15)
220491957 shelly_hap_window_c:331 WC 1: Cur pos 100.00 -> 0.00, P = -1.00
220524241 mgos_sys_config.c:232   Loading conf2.json
220548614 mgos_sys_config.c:232   Loading conf3.json
220683023 mgos_sys_config.c:174   Saved to conf9.json
220696345 shelly_hap_window_c:342 WC 1: Tgt pos 100.00 -> 50.00 (postcal1)
220707827 shelly_hap_window_c:322 WC 1: State: postcal1 -> idle (15 -> 0)
220745565 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
220844945 shelly_output.cpp:56    Output 1: off -> on (move)
220856932 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
220945890 shelly_hap_window_c:544 P = 186.12 -> 236.93
220956439 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
221748006 shelly_hap_window_c:331 WC 1: Cur pos 2.04 -> 2.35, P = 233.82
222547743 shelly_hap_window_c:331 WC 1: Cur pos 4.45 -> 4.76, P = 233.14
223347481 shelly_hap_window_c:331 WC 1: Cur pos 6.87 -> 7.17, P = 232.76
224148960 shelly_hap_window_c:331 WC 1: Cur pos 9.28 -> 9.58, P = 232.48
224947715 shelly_hap_window_c:331 WC 1: Cur pos 11.69 -> 11.99, P = 232.27
225320886 shelly_main.cpp:543     Up 225.31, HAP 0/7/12 ns 7, RAM: 16716/37616; st 59; 4.1: c:1 mp:236.93 mt_ms:33179 cp:12.89 tp:50.00 lemd:0 lhmd:0
225750964 shelly_hap_window_c:331 WC 1: Cur pos 14.10 -> 14.41, P = 232.29
226547712 shelly_hap_window_c:331 WC 1: Cur pos 16.51 -> 16.81, P = 231.66
227347608 shelly_hap_window_c:331 WC 1: Cur pos 18.92 -> 19.22, P = 232.10
228147869 shelly_hap_window_c:331 WC 1: Cur pos 21.33 -> 21.63, P = 232.36
228947778 shelly_hap_window_c:331 WC 1: Cur pos 23.74 -> 24.05, P = 232.51
229750447 shelly_hap_window_c:331 WC 1: Cur pos 26.16 -> 26.47, P = 232.62
230547534 shelly_hap_window_c:331 WC 1: Cur pos 28.57 -> 28.87, P = 232.66
231347659 shelly_hap_window_c:331 WC 1: Cur pos 30.98 -> 31.28, P = 232.34
232147810 shelly_hap_window_c:331 WC 1: Cur pos 33.39 -> 33.69, P = 232.64
232947724 shelly_hap_window_c:331 WC 1: Cur pos 35.80 -> 36.10, P = 233.10
233321003 shelly_main.cpp:543     Up 233.31, HAP 0/7/12 ns 7, RAM: 16716/37616; st 59; 4.1: c:1 mp:236.93 mt_ms:33179 cp:37.01 tp:50.00 lemd:0 lhmd:0
233754669 shelly_hap_window_c:331 WC 1: Cur pos 38.21 -> 38.53, P = 233.37
[... cut ...]
237469247 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
237567211 mgos_sys_config.c:232   Loading conf2.json
237592803 mgos_sys_config.c:232   Loading conf3.json
237727644 mgos_sys_config.c:174   Saved to conf9.json
237739751 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
237758930 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
241320455 shelly_main.cpp:543     Up 241.31, HAP 0/7/12 ns 7, RAM: 16772/37616; st 59; 4.1: c:1 mp:236.93 mt_ms:33179 cp:49.67 tp:49.67 lemd:0 lhmd:0

singas77, 2021-02-05:
so when open it reports as open and when closed it reports as closed so that's all good the only issue is calibration stops at 10% open I guess apart from that all ok

andyblac, 2021-03-03:
@rojer any advise ?

rojer, 2021-04-03:
sorry, still haven't had the time to review

pivale, 2021-04-22:
hey, I just had the same situation.

Yesterday evening I closed my shutter, but in home app it still stated open.
Today morning, I had to recalibrate, since I couldn't open the shutter (both on Home and physical switch) because it was "open" already.

I have all logs since February :P

This is the latest log file I have, it should have yesterday's problem.
[shellyswitch25-sala-persiana.log.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6356274/shellyswitch25-sala-persiana.log.zip)

lucaspinelli85, 2021-06-30:
I keep having this problem, the shutter status is not updated between app and reality

timoschilling, 2023-01-29:
closed because of inactivity and newer firmware releases

closed 2023-01-29 by timoschilling