mongoose-os-apps/shelly-homekit #375: Doorbell

DavidK1981, 2021-01-01:
It would be cool to use a Shelly with the HomeKit Firmeware as a Doorbell like this Homebridge Plugin enables a Shelly 1 to do so:[Shelly-Doorbell](https://gitlab.com/ca-iot/homebridge-shelly-doorbell#readme).
I try to avoid using Homebridge so a direct integration would make perfect sense.

BlancoDeMallloca, 2021-01-01:
This sounds great. Should work like a garage door opener Shelly 1. My doorbell uses a pulse for the ring and the close Sensor should be the ring button. Ring Button is also a pulse comming to SW on Shelly 1

@rojer: I think this could be easy as it should only be a new Mode for Shelly 1. Am I right?

DerKoller-eu, 2021-01-01:
> My doorbell uses a pulse for the ring and the close Sensor should be the ring button. Ring Button is also a pulse comming to SW on Shelly 1
> 
My Doorbell works exactly like that and I can’t hear it when I‘m in my Cellar‘s Office. So having a ring on my HomePod would be great if someone is at the door. I also think it should just be another mode for the Shelly like the Garagedoor.

Cojaeck, 2021-01-08:
I already use this. The input has to be in detached mode. And the HAP type must be set to occupancy. Then it appears in HomeKit as a sensor that can e.g. trigger your iPhone to show a notification. Additionally you can set a routine which activates a light or a bell. I have another Shelly connected to a buzzer which gets activated for two seconds (via auto off function) when the occupancy sensor gets active.
So I think there is no need to do anything. Everything is already there.

DerKoller-eu, 2021-01-08:
The advantage of Doorbell mode is that all homepods in the House ring immediately, no delay and every device in the home also immediately gets a push notification. You only need one Shelly 1 for this and also your old doorbell still rings. You also get two switches in home app to enable or disable the mechanical bell as well as the digital bell. This can be useful to disable the bells based on time, occupancy or routines.

simonfield76, 2021-01-08:
Is it possible to use it on the gate as well? 
I've attached a shelly 1 in parallel to the button to use it as gate opener but attaching it to the ring button would be nice!

Cojaeck, 2021-01-08:
> The advantage of Doorbell mode is that all homepods in the House ring immediately, no delay and every device in the home also immediately gets a push notification. You only need one Shelly 1 for [...]

Ok. Interesting. In my environment (no homepods) this wouldn't change anything. And the switching is done via switching routines on/off.

Cojaeck, 2021-01-08:
> Is it possible to use it on the gate as well? 
> I've attached a shelly 1 in parallel to the button to use it as gate opener but attaching it to the ring button would be nice!

When you set the Input to detached you should be able to do this. Then the switch operates separated from the input (as two separate devices). But the input is not isolated (the output of a Shelly 1 is). So the input must receive a signal from the same source as the supply voltage.

rojer, 2021-01-08:
doorbell in HAP spec has only one required characteristic - programmable switch event, exactly the same as the stateless switch.
so adding doorbell mode for input would be super easy, just a matter of plumbing.
@ZeevoX this would be a good first issue to tackle on the firmware side.

DerKoller-eu, 2021-01-08:
> So the input must receive a signal from the same source as the supply voltage.

And as remark:
As the standard doorbells often work with 12 Volt AC (in Austria at least) make sure to change that to 12 Volt DC or 24 Volt AC (If using 12 Volt DC don’t forget to set the jumper on the Shelly to the correct setting). Otherwise your Shelly gets toasted.

basroovers, 2021-01-23:
I now have a Shelly 1 connected to receive button presses from outside and ring the doorbell. Works like a charm but needs Homebridge and some work.

I’d rather have a simpler approach using on-Shelly firmware.

BlancoDeMallloca, 2021-02-06:
@ZeevoX: Could you already take a look into this? It would be great to have the doorbell mode. Thanks for your support

ZeevoX, 2021-02-12:
@BlancoDeMallloca and everyone else

Regarding the support status of doorbells in HomeKit. The service type is still present on [Apple's developer docs](https://developer.apple.com/documentation/homekit/hmservicetypedoorbell). However, it is not mentioned even once in the code for the [HomeKit ADK](https://github.com/Apple/HomeKitADK/). Back in July 2018 [Apple removed the entire doorbell category from its accessory list](https://9to5mac.com/2018/07/18/homekit-doorbell-removed/). 

I implemented a doorbell for Shelly-HomeKit but it shows up as "not supported" in the HomeKit app:

[image]

Apparently doorbells no longer function as stand-alone devices, they can be attached to a camera to produce a video doorbell, for reference see [this issue comment](https://github.com/homebridge/HAP-NodeJS/issues/433#issuecomment-294969084) on another HAP library. Or, as the HAP Specification vaguely puts it:
> The Doorbell service describes a doorbell and is the primary service of the Video Doorbell Profile.

For those that have [homebridge-shelly-doorbell](https://gitlab.com/ca-iot/homebridge-shelly-doorbell/) could you share some insight on how it works with HomeKit? Does it show a blank camera stream?

basroovers, 2021-02-12:
I would be fine it fakes a camera somehow. I just want it for the notifications.

jsiegenthaler, 2021-02-12:
I use https://github.com/Sunoo/homebridge-camera-ffmpeg with a GIF stored on my raspberry pi for the image and it works well on my iPhone with iOS 14.4 (and prior versions) . It also works with a static image or with no image. You could always add a default image, one of the existing icons and maybe some text, if you can pack that into the tiny memory space.

BlancoDeMallloca, 2021-02-12:
@ZeevoX and all: Thanks for the time spent on this issue. 

For me its also fine to have a predefined image or an icon. As @basroovers already wrote the main feature is the notification. 

Trank you very much

DerKoller-eu, 2021-02-12:
> For those that have [homebridge-shelly-doorbell](https://gitlab.com/ca-iot/homebridge-shelly-doorbell/) could you share some insight on how it works with HomeKit? Does it show a blank camera stream?

I attached two screenshots how it looks like in my home app.

i have a Shelly 1 with the stock firmware and Homebridge-Shelly-doorbell plug-in from @sl1nd installed.

The Bell itself doesn’t show up in HomeKit but the Homepods ring if someone pushes the button. Additionally you get two switches where I can turn the mechanical gong at my door as well as the digital gong for the HomePods on or off. Very useful for my „Good Night“ Automation.
[image]
[image]

timoschilling, 2021-02-12:
@ZeevoX The post that are you liked to is outdated or wrong, doorbells are back and they worked, only the Home app tile says unsupported.
Hier is it listeded in Apple Docs https://www.apple.com/ios/home/accessories/#section-bell
[image]

The Home app says it's unsupported, but that only means there is no interface in the Home app. If you use Home+ or the Eve app you can attach automatisation for the doorbell.

And if you trigger a doorbell the Home app gives a push notification and all HomePods are plaining a doorbell sound.
[image]
The push notification cam outside the Home app, too.

ZeevoX, 2021-02-12:
@timoschilling Yes! Thank you for pointing this out, indeed it says unsupported but pressing it causes a notification.

[image]

But I guess since there is no UI that means there is no switch for digital and mechanical gong like in the screenshot @DerKoller-eu posted?

DerKoller-eu, 2021-02-13:
> But I guess since there is no UI that means there is no switch for digital and mechanical gong like in the screenshot @DerKoller-eu posted?

The switch is created by the Homebridg plug-in. S as this is a nice feature I would love to see this in a Shelly HomeKit Firmware as well..

rojer, 2021-02-13:
this can be done in a separate update.

rojer, 2021-02-14:
@DerKoller-eu @BlancoDeMallloca i've cut 2.8.0-alpha1, please help test - https://github.com/mongoose-os-apps/shelly-homekit/issues/498

closed 2021-02-14 by rojer

BlancoDeMallloca, 2021-02-14:
> @DerKoller-eu @BlancoDeMallloca i've cut 2.8.0-alpha1, please help test - #498

Thank you very much. Need to go to the workshop to get add. installation material. Will test it asap.

spiegelt, 2021-03-12:
I've been using a shelly1 for my bell as a classic switch with an auto-off timer for a long time.  After the update I have now set up the shelly as the detached / doorbell / pulse as described.  I also get push notifications when the doorbell rings.  But how can I make the gong ring?

SimonWecklein, 2021-03-12:
I have the same issue either I get a notofication with detached or the gong rings with momentary. Is it possible to get both?

timoschilling, 2021-03-12:
Wait for a later version or create a automatisation wich turns on the switch if the doorbell triggers

victor-vz, 2021-05-16:
I have a Eufy 2k Homekit Camera and would like to use it in combination with a shelly (If possible without Homebridge) and a mechanical ring as a Video Doorbell. The Notification should show the Video Stream and announce who is at the Door via HomePods. Is it possible to trick HomeKit to think that Shelly plus Eufy 2k is a VideoDoor bell? I saw a GH post where the Shelly with HomeKit could be used with an IP camera...

AndrewJLockhart, 2021-06-27:
> I've been using a shelly1 for my bell as a classic switch with an auto-off timer for a long time. After the update I have now set up the shelly as the detached / doorbell / pulse as described. I also get push notifications when the doorbell rings. But how can I make the gong ring?

Have I missed something obvious in the GUI, I have a Shelly1, but can’t see the doorbell option anywhere.

jsiegenthaler, 2021-06-27:
Shelly1 acts as a Relay and a Switch input. 
You set it up in HomeKit as a doorbell using whatever method fits your system the best.
 
I used the https://github.com/Sunoo/homebridge-camera-ffmpeg#readme plugin to make a video doorbell and used the Shelly1 to send a http command from the SW input for the doorbell trigger. Works brilliantly. I use a GIF on my raspberry pi as the video feed.

Here's how it works:
**Ringing the Doorbell**
Pressing the physical doorbell button buzzes the doorbell buzzer, and activates the relay wired to the buzzer
The relay closes the SW contact on the Shelly1
The Shelly1 calls a URL command to the ffmpeg plugin "Front Door Doorbell Trigger" door bell input (a switch with auto-off)
The ffmpeg plugin sends the "Someone rang the doorbell" notification to HomeKit
The iOS devices show the notification, and it also pops up on Apple TV

**Opening the Door**
Switching on the "Front Door Opener" in Home app activates the Shelly1 relay
The Shelly1 relay closes and connects the Door opener wires on the physical door opener
The door buzzer sounds and the person outside the door can open the door
The Shelly1 turns off automatically after 4 seconds (the relay goes off, and the door opener buzzer stops)

**Automatic Door Opener**
When a person in my house comes home, a delay switch "Recent Arrival" is turned on. This turns off after a set time.
When the ffmpeg door bell input is turned on (by the Shelly1), it runs a simple automation: If Recent Arrival is On, then unlock the "Front Door" (lock mechanism), and turn on the "Front Door Opener" (switch). 
This allows the person to enter the building

**Displaying Door Lock/Unlock Status**
When the "Front Door Opener" switch turns on, it triggers an automation: When Front Door Opener Turns On, Unlock Apartment Front Door"
When the "Front Door Opener" switch turns off (auto timeout on the Shelly1), it triggers an automation: When Front Door Opener Turns Off, Lock Apartment Front Door"
The "Front Door" lock mechanism in the room "Apartment" sends notifications when the door is locked or unlocked, and it displays the lock/unlock state.

AndrewJLockhart, 2021-06-27:
That’s a nice setup, but my question was related specifically to the Shelly Mongoose HomeKit firmware and how this is supported natively, without going via Homebridge.

There is reference to a doorbell in the firmware, but I can’t seem to set it up as such.

kloibs, 2022-05-13:
Hi guys, I know this is from a while ago, but I’m looking to implement something along these lines for my apartment in Austria. 

If I understand correctly I should be able to wire a Shelly 1 into the bell itself so that when someone rings from downstairs or from my front door, it triggers a standard HomeKit doorbell notification and rings on the HomePods.

Would I be able to turn the physical bell on or off as well? Thanks in advance!

jsiegenthaler, 2022-05-14:
Hi @kloibs 
Your understanding is correct. 
I use a single Shelly 1 in my apartment.
1. the SW input is triggered by a relay which in turn is triggered by the bell voltage (when the bell is ringing)
2. the relay output triggers the door opener
The Shelly 1 one only has a single relay. If you want to trigger the door opener (Türöffner), and also have the ability to turn off the ding-dong bell, then you need 2 x Shelly 1s.

kloibs, 2022-05-14:
@jsiegenthaler thanks for the tip! I’ve got a Nuki opener installed already, so just looking for the doorbell connectivity. Should just be able to switch the Shelly‘s switch to 12V and be good to go!

jsiegenthaler, 2022-05-14:
@kloibs Excellent! If your apartment intercom has an old fashioned electric buzzer, triggered by an AC voltage, then be careful with what voltage relay you use as the voltage drops under load. I measured 14V AC under no load, so added a 12V AC relay, but it was slow to switch on when the buzzer was going. Turned out the AC voltage sunk to about 8V AC when the buzzer was buzzing. So I swapped the relay to a 6V AC relay and now it is reliable.

kloibs, 2022-05-15:
@jsiegenthaler I‘m planning on getting a Shelly 1 tomorrow to implement this, but I don’t have a multimeter unfortunately. My apartment building in Salzburg was built in the 70s, so now I’m nervous about the voltage, even though my landlord said he was rather confident about the voltage being 12v.

Do you recommend proceeding or should I go on the side of caution and try to find a multimeter?

jsiegenthaler, 2022-05-15:
@kloibs my apartment in Switzerland is of the same age. It'll be a 12V system likely with the doorbell being 12V AC. But you do need to confirm with a multimeter. You should definitely proceed, and a multimeter will be very useful.

kloibs, 2022-05-16:
@jsiegenthaler maybe I‘m too stupid to be implementing Shellys in my setup, but the Shelly would still need to be powered somehow…. It wouldn’t be able to be powered via the doorbell wiring as I originally thought, correct?

kloibs, 2022-05-18:
[image]
[image]

kloibs, 2022-05-18:
Sorry to keep bothering everyone. I was successfully able to get a Shelly Uni connected to power, but unsure as to where to connect either of the inputs. Whenever they’re connected to the lines that go to the bell itself, it triggers a continuous loop of doorbell rings within HomeKit. Any suggestions?

Tillman-Z, 2022-11-30:
> 

Hi!
Was this implemented already?
Currently I only see "Switch", "Outlet", "Lock" and "Valve" as HAP service types (running in mode "switch").
Or am I missing something? :)

timoschilling, 2022-11-30:
Change the input type to detached, then you get a separate config section for the input where you can set the type to doorbell.

pingcheater, 2024-12-12:
Hi guys,
first of all, many thanks for what you're doing! It's great work and I appreciate it very much! Got several Shellys around my house running your fw!

And since I've damaged one of the chimes in my house some while ago, I wanted to make my hompepod ring, when someone rings at the door. Now I finally got to do it and seem to be very close. Here's my setup:

A non-smart doorbell on a 12V AC line with a button at the door closing and opening contact. Since there are several cable wire running up to the chime, I've connected one of the unused ones to the 12V AC as well, so I can feed my Shelly. 

At the Chime, I've got a Shelly Uni, getting 12V AC and using the button signal as analog input. I've got the shelly to install the HomeKit FW and have set it up as in the screenshot:
 
[image]

But I don't have any way to setup the homepod to react to this. Where did I go wrong?

Thanks alot!

PR #495 (2021-02-14): Add HAP doorbell input type
* Refactored out bulk of `stateless_switch` into `stateless_switch_base` to avoid code duplication
* Added simple `doorbell` service based on `stateless_switch_base`
* Fix #375