mongoose-os-apps/shelly-homekit #377: 2.7.0 update of Shelly I3 fails

sebastianheise, 2021-01-02:
I'm not too sure if it's just me, or if this is occurring to anyone else. 
I've updated all my Shellys to 2.7.0 and while Shelly1 and Shelly2.5 were updating fine, the i3 update failed. 

The upload part times out with a lost network connection.
I have also tried uploading the zip manually - with the same result.

[image]

[image]

sebastianheise, 2021-01-07:
Soo... looks like I might be the only one seeing this? That's odd. Okay, I'll investigate a little more.

sebastianheise, 2021-02-03:
No. Sorry. I have tried a few more times to update, but is fails. No clue why.

sebastianheise, 2021-02-03:
(Not sure why I didn't receive a notification of your response though...)

lieberjott, 2021-02-03:
Seem that I'm having the same issue but with the difference, that I'm trying to flash a brand new Shelly I3 with stock firmware.

Stock firmware tells me:
"The current Firmware version of your Shelly device is 20200421-183805/master@d834f204 No newer firmware available."

I've tried to do the OTA update/flashing to shelly-homekit several times, but 'nothing' happens, this is the output after doing so:

status: | "updating"
-- | --
has_update | false
new_version | ""
old_version | "20200421-183805/master@d834f204"

sebastianheise, 2021-02-06:
No, no static IP. But it's also not "fetching". I see the same, if I choose to upload the file. As if there wasn't enough memory on the device or something. I've tried to revert to stock firmware - no luck. Feels like it's "bricked" - except that it works fine. Just no firmware updates anymore. It's very strange....

sebastianheise, 2021-02-06:
Is there anything in regards of debug logs, that I could try to get off the device? Anything that might help discovering what's going wrong?

rojer, 2021-02-07:
unfortunately stock firmware does not provide any feedback or logs, update either works or it doesn't...

sebastianheise, 2021-03-06:
It's not the stock firmware I'm trying to update from. I'm on `Version: 2.6.1 (build 20201128-161102/2.6.1-g3ba3275-master)`. 
I have tried to update, I have tried to revert to stock, no luck. I always get the above timeout/error.

sebastianheise, 2021-03-06:
........okayyyyy..... 
I do not know what went wrong, but I just used an incognito window in Chrome to do the update and it worked. 
Previously I used Safari and a normal window in Chrome and it timed out with that network error.

Feel free to close. 🤷

closed 2021-03-06 by andyblac