mongoose-os-apps/shelly-homekit #379: Publish automatic latest release

rojer, 2021-01-03:
Add a GitHub action to automatically build a release from master
Github release should be called 'latest'
Version should be incremented with number of commits from latest tag.
This can be taken from `git describe output`:
For example, right now we are 7 commits past 2.7.0:

```
$ git describe --tags
2.7.0-7-g3cfc021
```

Current latest release should have version `2.7.0-latest7`

This will ensure version check comparison during automatic update check works.

timoschilling, 2021-01-04:
The GitHub release should be called a prerelease if it's not a stable one.

ZeevoX, 2021-01-29:
I have started work on a pull request but have not implemented the part of the GitHub workflow to upload the produced firmware files, as I would like to ask for clarification on how this ought to be done. 

If we have a 'latest' GitHub release that would imply tagging every commit on master so that it always pointed to the latest commit. Should the script instead attach files to the last tag? (e.g. attaching `2.7.0-latest7` to the `2.7.0` release) Or could the script instead upload the firmwares to `rojer.me` like @rojer manually did [here](https://github.com/mongoose-os-apps/shelly-homekit/issues/50#issuecomment-753633174)?

timoschilling, 2021-01-29:
Maybe we can remove rojer.me and direct work with the GitHub release api?

andyblac, 2021-01-29:
thats how it was originally, but was moved to Roger.me for better controll.

timoschilling, 2021-01-29:
What was the concrete reason / problem?

andyblac, 2021-01-29:
not sure was rojer decision.

andyblac, 2021-01-29:
think it was so he could control beta releases.

rojer, 2021-01-30:
we need a plain HTTP location for stock -> HK update as stock firmware can't flash from HTTPS location (at least not from github).
we could use github urls for HK -> HK updates but for me it's just easier to have everything in one place.

andyblac, 2021-01-30:
it also makes it easier to maintain script.