mongoose-os-apps/shelly-homekit #381: Feature Request: physical input to reboot

oliversaa, 2021-01-03:
Good Day

Very rarely a shelly becomes unresponsive. Rebooting the router does not do the trick; the only way to breath life back into the shelly is to cut power for a second or two. This means I have to play with the fuse.

It would be awesome if e.g. 8x switching a lamp on/off would trigger a reboot of the shelly - if the shelly is connected to a wall-switch.

Thank you very much for your consideration!

andyblac, 2021-01-04:
has been decussed before @rojer prefers to find the real reason main issue, instead of a work around, maybe you can provide hime with more deatails and logs from your device. but it sounds like this issue here https://github.com/mongoose-os-apps/shelly-homekit/issues/129

rojer, 2021-01-04:
that's right. ideally we fix the root cause, or at least if we find manifestation of this, we can have firmware apply remediation such as restart wifi or reboot itself. in wither case we need information, in this case logs.

andyblac, 2021-01-25:
@oliversaa can yu provide an update ?

andyblac, 2021-01-31:
as the OP has not responded, i'll close this issue.

closed 2021-01-31 by andyblac