mongoose-os-apps/shelly-homekit #382: Shelly gets out of sync in HomeKit

ursusHU, 2021-01-04:
This is more or less a question. With the latest firmware (2.7.0) I get the issue that the status of the switch isn’t appearing correctly in HomeKit, and therefore also automations are failing. It is not a common problem, but appearing time to time. Should I look for an external cause, or anyone else experiencing this?

andyblac, 2021-01-24:
how many ios devices do you have in the home ?

ursusHU, 2021-01-25:
2 Phones only

simonfield76, 2021-01-25:
In my experience it looks like a wifi issue.

Inviato da iPhone

andyblac, 2021-03-03:
@ursusHU for automations you need a homekit hub, do this help, can i close this issue ?

ursusHU, 2021-03-03:
@andyblac I have hubs. You asked about ios devices (Phones)
Anyway with The current version I don’t experience the issue.

closed 2021-03-03 by ursusHU