mongoose-os-apps/shelly-homekit #384: Internal temperature in original firmware

marmod, 2021-01-04:
Hi

it is possible to read internal temperature??
if yes, how to do it

simonfield76, 2021-01-04:
With the Shelly 2.5 you can see the internal temperature only via web interface but at the moment it seems not possible to expose it to HomeKit

[cid:2CBED507-F6A6-41B2-ABE1-7A30309B6D15-L0-001]

Inviato da iPhone

rojer, 2021-01-04:
indeed, you can see it via web interface (and not just for 2.5) but not in homekit. not a bad idea to expose it as a temperature sensor, perhaps off by default.

andyblac, 2021-01-04:
> indeed, you can see it via web interface (and not just for 2.5) but not in homekit. not a bad idea to expose it as a temperature sensor, perhaps off by default.

off as default ye, personally i do not see this being of any use as a sensor in homekit.

rojer, 2021-01-04:
agree, probably not useful to a wide audience, hence off by default.

marmod, 2021-01-04:
roy 
how did you read temp inside shelly???? it is possible in the same way read tem in original firmware

marmod, 2021-01-04:
ok I found it

pivale, 2021-01-05:
I would say that the shelly will rarely display internal temp = ambient temp, and it can be misleading and even triggering automations when not supposed to. Enclosed in walls or any other space, the temperature will rise, specially on load.
Can't really see a benefit on having this since it won't be accurate...

Dabr422, 2021-01-05:
I agree that the use for the internal temperature sensor is limited. But some  Shelly devices also support connection of external temp sensors in stock mode. Maybe it would be a good idea to add that feature. I can see use for such setup.

andyblac, 2021-01-05:
> I agree that the use for the internal temperature sensor is limited. But some Shelly devices also support connection of external temp sensors in stock mode. Maybe it would be a good idea to add that feature. I can see use for such setup.

that already has a issue submitted.

timoschilling, 2022-05-02:
> I would say that the shelly will rarely display internal temp = ambient temp, and it can be misleading and even triggering automations when not supposed to. Enclosed in walls or any other space, the [...]

I see it in the same way. @rojer do you still thinks a good feature? I expect too many bug reports. For real temperature measuring is implemented since #874

rojer, 2022-05-06:
not useful as ambient temperature sensor but could be useful as a way to monitor device's own health.

kloibs, 2022-06-13:
Hey all, also would be curious for this to be possible…. Similar how a Shelly installed in an outlet can display current power usage in the Eve app. 

Thought potentially the same for the internal operating temp. Cheers!