mongoose-os-apps/shelly-homekit #385: Shelly 1pm after reverting to stock. SW input is not working.

agarwalsndp, 2021-01-04:
Here is the steps I followed for a Shelly 1 as well as Shelly 1pm
1. Flashed with Shelly-homekit v2.7.0.
2. Reverted to stock via web interface
3. The manual input (SW input) was not working. Tried updating button type in settings still did not work.
4. Factory reset Shelly 1pm  and Shelly 1 via stock app. (Shelly 1pm still didn't work, Shelly 1 started recognizing SW input)
5. Flashed again with Shelly-homekit v2.7.0 and SW input is working fine here. 
6. Repeated steps 1-5 again with same results.

I did notice once after step 2 that Shelly 1pm was started showing input voltage as '220v'. Don't remember seeing that again.

rojer, 2021-01-05:
hm, this is very strange but i can confirm - i reverted 1pm device to stock and input stopped working. it took some fiddling with configuration, changing button and power-on state to restore it. and looks like device rebooted a few times in the process...
very strange.

rojer, 2021-01-05:
ok, i think i know what the problem is. it has to do with different interrupt modes for hardware timer used by stock vs homekit firmware: stock uses NMI interrupt, HK uses normal edge-triggered interrupt.
it's a technicality but until recently there was a bug in Mongoose OS that prevented switching between the modes correctly.
in the other direction it prevented upgrading from stock to homekit for 1L (#323).
after upgrade is applied, firmware performs soft reset of the CPU to reboot but it does not restore interrupt configuration registers into initial state and things end up going awry wen stock tries to configure NMI mode with edge mode already active.
this bug was fixed upstream in https://github.com/cesanta/mongoose-os/commit/41f01a2c76d626249d42bfdc4ec8f068e7251d00 - hopefully it will eventually trickle down into new stock version (cc @kzyapkov)

meanwhile, what you need to do is after reverting to stock firmware, power the device completely off and back on. this will reset the interrupt configuration register and stock firmware should operate correctly from then on (at least this this worked for me).

agarwalsndp, 2021-01-05:
Thank you for the info. Will give it a try.

andyblac, 2021-01-25:
@agarwalsndp can you report back ?

agarwalsndp, 2021-01-26:
I can confirm that after disconnecting power from panel the SW input starts working in stock firmware

closed 2021-01-26 by agarwalsndp