mongoose-os-apps/shelly-homekit #388: Feature request flash script one step from new stock to HomeKit

timoschilling, 2021-01-06:
It would be nice if a new stock devise could be flash from outdated stock to shelly-homekit firmware in one step.

Current flow:
1. `flash-shelly.py -a host`
2. update from Stock 0.0.0 to Stock 1.9.3
3. `flash-shelly.py -a host` 
4. update from Stock 1.9.3 to HomeKit 2.7.1

Desired flow:
1. `flash-shelly.py -a host`
2. update from Stock 0.0.0 to Stock 1.9.3 to HomeKit 2.7.1

andyblac, 2021-01-06:
i'll see what i can do.

rojer, 2021-01-06:
this is a good idea because at some point updating from latest stock (or at least from stock beyond 1.9) will become a hard requirement.
this will happen when we change flash layout for the firmware and only recent stock firmware supports changing layout.

andyblac, 2021-01-06:
> this is a good idea because at some point updating from latest stock (or at least from stock beyond 1.9) will become a hard requirement.
> this will happen when we change flash layout for the firmware and only recent stock firmware supports changing layout.

atm, flashing to hk, if stock is not current it will auto update to vastest 1st, it will not allow you to flash direct to HK from a none current stock device.

andyblac, 2021-01-06:
something like ?

```
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py shelly1l-84CCA8ADF044

Host: http://shelly1l-84CCA8ADF044.local
Device Name: None
Device ID: shelly1l-84CCA8ADF044
IP: 192.168.3.62
Model: Shelly1L
Sys Temp: 0˚c
Current: Stock 1.9.2
Latest: Stock 1.9.3
This device needs to be upgraded to latest stock firmware version, before you can flash to HomeKit (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8ADF044 to reboot...
Successfully flashed shelly1l-84CCA8ADF044 to 1.9.3

Host: http://shelly1l-84CCA8ADF044.local
Device Name: None
Device ID: shelly1l-84CCA8ADF044
IP: 192.168.3.62
Model: Shelly1L
Sys Temp: 0˚c
Current: Stock 1.9.3
Latest: HomeKit 2.7.1
Do you wish to flash shelly1l-84CCA8ADF044 to firmware version 2.7.1 (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8ADF044 to reboot...
Successfully flashed shelly1l-84CCA8ADF044 to 2.7.1
```

andyblac, 2021-01-06:
this is better:
```
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py shelly1l-84CCA8ADF044

Host: http://shelly1l-84CCA8ADF044.local
Device Name: None
Device ID: shelly1l-84CCA8ADF044
IP: 192.168.3.62
Model: Shelly1L
Sys Temp: 0˚c
Current: Stock 1.9.2
Latest: Stock 1.9.3
This device needs to be upgraded to latest stock firmware version, before you can flash to HomeKit (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8ADF044 to reboot...
Successfully flashed shelly1l-84CCA8ADF044 to 1.9.3
Do you wish to contintue to flash shelly1l-84CCA8ADF044 to HomeKit firmware version 2.7.1 (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8ADF044 to reboot...
Successfully flashed shelly1l-84CCA8ADF044 to 2.7.1
```

andyblac, 2021-01-06:
submited update in PR https://github.com/mongoose-os-apps/shelly-homekit/pull/389

lucaspinelli85, 2021-01-06:
Maybe is off topic but I can't update from script, I followed the instructions but this Is result of terminal: -bash: i.e: command not found
Where I wrong? help me thanks

andyblac, 2021-01-06:
Can you show me your screen, I can understand what you did?

lucaspinelli85, 2021-01-06:
Ok, i do this:
-install phyton 3.9 and double click to install certificates.command
[image]
-download folder
[image]
-right click to "tool" folder and "new terminal in folder"
-paste the command
[image]

andyblac, 2021-01-07:
It seems it has lost execution bit, try

chmod 755 flash-script.py 

Then retry script

andyblac, 2021-01-07:
wiki was out of date, has been updated, sorry about that.

timoschilling, 2021-01-07:
Why not removing the `upgraded to latest stock firmware version` step? The user wants to install hk so do it, there is no need to explicit ask for installing latest stock first.

andyblac, 2021-01-07:
> Why not removing the `upgraded to latest stock firmware version` step? The user wants to install hk so do it, there is no need to explicit ask for installing latest stock first.

you can overide all questions via `-y`

pivale, 2021-01-07:
> Why not removing the `upgraded to latest stock firmware version` step? The user wants to install hk so do it, there is no need to explicit ask for installing latest stock first.

My understanding is that we might not always be able to update directly to homekit, depending on the stock firmware.
So the safest approach is to always have stock firmware as updated as possible before applying homekit firmware.

andyblac, 2021-01-07:
> My understanding is that we might not always be able to update directly to homekit, depending on the stock firmware.
> So the safest approach is to always have stock firmware as updated as possible [...]

correct, before you can safely flash to HomeKit the stock needs to be on a minimum (atm) of 1.9.0

rojer, 2021-01-07:
actually, at the moment flashing from any stock version is possible but in the future having 1.9+ will be a requirement

andyblac, 2021-01-07:
> actually, at the moment flashing from any stock version is possible but in the future having 1.9+ will be a requirement

not necessarily true, https://github.com/mongoose-os-apps/shelly-homekit/issues/390

lucaspinelli85, 2021-01-08:
> wiki was out of date, has been updated, sorry about that.

in wiki there is: i.e ./flash-script.py -a, I try it but don't work! (I don't know what means i.e at beginning?)
It work only with this command: ./flash-shelly.py -ay (without i.e ?)
But sometimes it stopped with this log, why?:

Host: http://Babbo-Natale.local
Device Name: Babbo-Natale
Device ID: shellyplug-s-9A36E9
IP: 192.168.0.32
Model: ShellyPlugS
Current: HomeKit 2.7.0
Latest: HomeKit 2.7.1
Downloading Firmware...
Now Flashing...
waiting for Babbo-Natale to reboot...
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 445, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 440, in _make_request
    httplib_response = conn.getresponse()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/http/client.py", line 268, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 755, in urlopen
    retries = retries.increment(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/util/retry.py", line 531, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/packages/six.py", line 735, in reraise
    raise value
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 699, in urlopen
    httplib_response = self._make_request(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 447, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urllib3/connectionpool.py", line 336, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPConnectionPool(host='192.168.0.32', port=80): Read timed out. (read timeout=3)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/lucaspinelli/Desktop/shelly-homekit-master/tools/./flash-shelly.py", line 694, in <module>
    device_scan(args.hosts, action, args.dry_run, args.quiet_run, args.silent_run, args.mode, args.exclude, args.version, args.variant, args.hap_setup_code)
  File "/Users/lucaspinelli/Desktop/shelly-homekit-master/tools/./flash-shelly.py", line 614, in device_scan
    probe_device(device, action, dry_run, quiet_run, silent_run, mode, exclude, version, variant, hap_setup_code)
  File "/Users/lucaspinelli/Desktop/shelly-homekit-master/tools/./flash-shelly.py", line 589, in probe_device
    parse_info(deviceinfo, action, dry_run, quiet_run, silent_run, flashmode, exclude, hap_setup_code, requires_upgrade)
  File "/Users/lucaspinelli/Desktop/shelly-homekit-master/tools/./flash-shelly.py", line 558, in parse_info
    write_flash(device_info, hap_setup_code)
  File "/Users/lucaspinelli/Desktop/shelly-homekit-master/tools/./flash-shelly.py", line 405, in write_flash
    onlinecheck = device_info.get_current_version()
  File "/Users/lucaspinelli/Desktop/shelly-homekit-master/tools/./flash-shelly.py", line 194, in get_current_version
    info = self.get_device_info()
  File "/Users/lucaspinelli/Desktop/shelly-homekit-master/tools/./flash-shelly.py", line 177, in get_device_info
    fp = requests.get(self.device_url, timeout=3)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPConnectionPool(host='192.168.0.32', port=80): Read timed out. (read timeout=3)

andyblac, 2021-01-08:
log is truncated can you put in a text file please.

andyblac, 2021-01-08:
is it the latest version of the script ? can you try 
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5789297/flash-shelly.py.zip)

andyblac, 2021-01-08:
think i need to add a version number to script, so it easier to keep track.

lucaspinelli85, 2021-01-08:
[log.rtf.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5789303/log.rtf.zip)

lucaspinelli85, 2021-01-08:
but I repeat the command and now it update alle the 42 shelly! It happens only 2 times

andyblac, 2021-01-08:
> log.rtf.zip

yeah that bud has been fixed, please version in zip.

andyblac, 2021-01-08:
> but I repeat the command and now it update alle the 42 shelly! It happens only 2 times

it sometimes happens when shelly is slow to respond, the zip version should fix stop it crashing. think it waiting a pull request for fix

andyblac, 2021-01-08:
please report back if it still happens :) in furure.

lucaspinelli85, 2021-01-08:
ok, now I can't replicate the log, but next time I'll upload, I'll download again the zip from link in wiki page (is it updated with last fix?)
Please correct the commands in wiki page (delete i.e if not necessary) and if you can a more intuitive description for those people unfamiliar with programming code, how to combine options arguments together too! (report also the example "-ay" to avoid press y for each, is very useful!). Personally I don't understand what is the second column on description, for example the -a, --all, or the -q, --qiet? Is only a short description?!

andyblac, 2021-01-08:
> ok, now I can't replicate the log, but next time I'll upload, I'll download again the zip from link in wiki page (is it updated with last fix?)
> Please correct the commands in wiki page (delete i.e [...]

not yet latest version is waiting a purll request, latest vesion is in zip further up in this thread.

andyblac, 2021-01-08:
> ok, now I can't replicate the log, but next time I'll upload, I'll download again the zip from link in wiki page (is it updated with last fix?)
> Please correct the commands in wiki page (delete i.e [...]

it all in the help.  `-h`

andyblac, 2021-01-08:
```
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py -h
usage: flash-shelly.py [-h] [-m {homekit,keep,revert}] [-a] [-q] [-l] [-e [EXCLUDE ...]] [-n] [-y] [-V VERSION] [-c HAP_SETUP_CODE] [--variant VARIANT] [--file FILE] [-v {0,1}] [hosts ...]

Shelly HomeKit flashing script utility

positional arguments:
  hosts

optional arguments:
  -h, --help            show this help message and exit
  -m {homekit,keep,revert}, --mode {homekit,keep,revert}
                        Script mode.
  -a, --all             Run against all the devices on the network.
  -q, --quiet           Only include upgradeable shelly devices.
  -l, --list            List info of shelly device.
  -e [EXCLUDE ...], --exclude [EXCLUDE ...]
                        Exclude hosts from found devices.
  -n, --assume-no       Do a dummy run through.
  -y, --assume-yes      Do not ask any confirmation to perform the flash.
  -V VERSION, --version VERSION
                        Force a particular version.
  -c HAP_SETUP_CODE, --hap-setup-code HAP_SETUP_CODE
                        Configure HomeKit setup code, after flashing.
  --variant VARIANT     Prerelease variant name.
  --file FILE           File / URL to flash.
  -v {0,1}, --verbose {0,1}
                        Enable verbose logging level.
```

just keep adding commands is easier ie `-a -y -q` etc

andyblac, 2021-01-08:
second collum is just another way to action some people like `-a` and some are to use long commands `--all` thats all

andyblac, 2021-01-09:
this has been merged, closing

closed 2021-01-09 by andyblac