mongoose-os-apps/shelly-homekit #390: Problems flashing 2.7.0 from stock

Dabr422, 2021-01-06:
Hi,
Just added my 10th Shelly1 but can't get it to overwrite the stock software with v2.7.0. I've followed the same process as all the previous times I did this:
1. Update to latest stock firmware. 
2. Set a fixed IP from stock firmware, in this case 192.168.1.231.
3. Calling http://192.168.1.231/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip.
I then see **{"status":"updating","has_update":false,"new_version":"20201228-092119/v1.9.3@ad2bb4e3","old_version":"20201228-092119/v1.9.3@ad2bb4e3"}** and nothing else is happening (as expected). When I refresh the page after a while the stock firmware is still showing up. I've tried this at least 10 times now with the same result. Any idea?

Thanks
Daniel

daneo82, 2021-01-06:
Same for me, I have several Shellys (1 & 2.5) in place, all up to date, but now the new one (Shelly 1 v3) is not going to move.

Benjamin66617, 2021-01-06:
Same for me. Doesn't work!!

daneo82, 2021-01-06:
I used local http file server (search goolge for HFS), manually downloaded 2.7 from here and than it worked.
Could be an issue with 2.7.1 or with the fileserver.
Tried to access file server manually worked, no real conclusion to this, maybe 2.7.1?
After 2.7.0 installed update to 2.7.1 worked fine.

andyblac, 2021-01-06:
I have been flashing back and forth my Shelly 1L from
Stock to HomeKit np today whilst working on script.

Dabr422, 2021-01-06:
Interesting. I just renamed the topic to 2.7.1. I didn’t notice that a new version came today and I have tried to update with the latest version this afternoon, that is, most likely 2.7.1.

andyblac, 2021-01-06:
```
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py shelly1l-84CCA8ADF044

Host: http://shelly1l-84CCA8ADF044.local
Device Name: None
Device ID: shelly1l-84CCA8ADF044
IP: 192.168.3.62
Model: Shelly1L
Sys Temp: 0˚c
Current: Stock 1.9.2
Latest: Stock 1.9.3
This device needs to be upgraded to latest stock firmware version, before you can flash to HomeKit (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8ADF044 to reboot...
Successfully flashed shelly1l-84CCA8ADF044 to 1.9.3
Do you wish to contintue to flash shelly1l-84CCA8ADF044 to HomeKit firmware version 2.7.1 (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8ADF044 to reboot...
Successfully flashed shelly1l-84CCA8ADF044 to 2.7.1
```

andyblac, 2021-01-06:
@Dabr422 did you update to stock 1.9.3 1st ?

Dabr422, 2021-01-06:
Actually no. I first tried to flash the homekit firmware before updating stock firmware. But when that didn't work I updated it and factory reset it. Same result when trying again to apply homekit firmware.
However, as it so happens I bought two units last time. I've now tried the other one, where I did update stock firmware to latest version first. And this time it worked. I don't understand the difference. compared to when I updated ut and reset it but something seems to be different between these two approaches. The question now is how I can make the first unit up and running.

andyblac, 2021-01-06:
the latest fw requires the latest stock formware, it is to do with how flash storage it setup iirc.

Dabr422, 2021-01-06:
I see. Do you know how I can restore my unit so that it can be used, or is it useless now?

andyblac, 2021-01-06:
please explain where you are at atm.

Dabr422, 2021-01-06:
Hi again,
Now it works :-). I repeated the whole process again with resetting to factory default and then included it in the Shelly app once more. (I did not update to latest stock firmware this time, because it said it already has the latest version and can therefore not be upgraded.) For some reason I was now able to download the homekit firmware after that. Many thanks for your assistance and dedication!

andyblac, 2021-01-06:
> Hi again,
> Now it works :-). I repeated the whole process again with resetting to factory default and then included it in the Shelly app once more. (I did not update to latest stock firmware this [...]

glad you got it working, i have a script that does it all for you, once you have linked it to your home wifi.  download the git zip here https://github.com/mongoose-os-apps/shelly-homekit/archive/master.zip

and read this https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#Script

Dabr422, 2021-01-06:
Perfect, thanks :-)

andyblac, 2021-01-06:
i'll close this issue.

closed 2021-01-06 by andyblac

rojer, 2021-01-06:
thanks Andy!