mongoose-os-apps/shelly-homekit #392: UI bug with selecting text in inputs

timoschilling, 2021-01-07:
Way to reproduce:
1. Open UI of a devise (tested with PlugS, i3 and 2.5)
2. Select the text of a text input for example devise name
3. wait one or two second and the text will be deselected

This happens until the first change in the inputs happens, after that selected text keeps selected.

andyblac, 2021-01-07:
yeah, noticed this too, but if you go to end of the text field it stays there, and allows you to edit the field with no issues.

timoschilling, 2021-01-07:
Same happens if you place the cursor inside the text, it will jump to the end of the input after a short moment.

timoschilling, 2021-01-07:
Yes there is a work around but I would like to report this to make users more happy.

andyblac, 2021-01-07:
it is probably due to the new UI update with auto refresh, i have no idea if is a bug or normal behaviour.

rojer, 2021-01-07:
yes, we should not be updating elements if there is active selection and/or value hasn't changed.
@ZeevoX please take a look

commit cad46e1c (2021-01-20): Fix UI bug with selected text in inputs (fix #392) (#417)

Do not update elements if there is an active sleection and/or their
  value has not changed since the last refresh
closed 2021-01-20 by rojer