mongoose-os-apps/shelly-homekit #395: Firmware Upgrade Script Issue

hughfr4nc15, 2021-01-08:
Hello,

All my Shelly's have your latest firmware (apart from the Duo's), I only noticed the script after upgrading them all but wanted to try it anyway.
I installed it on Windows 10 and Linux Mint (VM) and in both cases it cannot find the any of my Shelly devices:

```
hughfr4nc15@HTPC-VirtualBox:~/Desktop/shelly-homekit-master/tools$ ./flash-shelly.py -la
Scanning for Shelly devices...

Devices found: 0 Upgradeable: 0
hughfr4nc15@HTPC-VirtualBox:~/Desktop/shelly-homekit-master/tools$ 
```

Any idea why this is happening?

Thanks in advance.

andyblac, 2021-01-08:
can you run with `-v=1` and copy paste output to a file and post here.

andyblac, 2021-01-08:
another question, do you use VLAN's is your PC on the same network subnet as your shelly devcies ?

hughfr4nc15, 2021-01-08:
Hello @andyblac,

Thanks for the answer.
Changed the Linux VM adapter to bridged for it to be in the same subnet as the Shelly's and that worked.
Weird that my WIndows 10 that is the actual machine where the VM is running and is also in the same subnet does not detect the Shelly's.

Here's the Windows output:
`C:\Users\Hugo Francisco\Downloads\shelly-homekit-master\tools>flash-shelly.py -la -v=1
app
OS: Windows
manual_hosts: []
action: list
mode: homekit
dry_run: False
quiet_run: False
silent_run: False
hap_setup_code: False
version: False
exclude: None
variant: False
verbose: 1

stock_release_info:{'isok': True, 'data': {'SHPLG-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-1.zip', 'version': '20201124-091454/v1.9.0@57ac4ad8'}, 'SHPLG-S': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-S.zip', 'version': '20201228-092848/v1.9.3@ad2bb4e3'}, 'SHPLG-U1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-U1.zip', 'version': '20201124-092346/v1.9.0@57ac4ad8'}, 'SHPLG2-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG2-1.zip', 'version': '20201228-092942/v1.9.3@ad2bb4e3'}, 'SHSW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-1.zip', 'version': '20201228-092119/v1.9.3@ad2bb4e3'}, 'SHSW-21': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-21.zip', 'version': '20201228-092214/v1.9.3@ad2bb4e3'}, 'SHSW-25': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-25.zip', 'version': '20201228-092242/v1.9.3@ad2bb4e3'}, 'SHSW-PM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-PM.zip', 'version': '20201228-093009/v1.9.3@ad2bb4e3'}, 'SHSW-L': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-L.zip', 'version': '20201228-092147/v1.9.3@ad2bb4e3'}, 'SHAIR-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHAIR-1.zip', 'version': '20201124-092534/v1.9.0@57ac4ad8'}, 'SHSW-44': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-44.zip', 'version': '20201019-110845/v1.6.6_4Pro@bad181e4'}, 'SHUNI-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHUNI-1.zip', 'version': '20201124-093042/v1.9.0@57ac4ad8'}, 'SHEM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM.zip', 'version': '20201124-092817/v1.9.0@57ac4ad8'}, 'SHEM-3': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM-3.zip', 'version': '20201124-092854/v1.9.0@57ac4ad8'}, 'SHSEN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSEN-1_build.zip', 'version': '20200717-112934/v1.7.7@a02db0b8'}, 'SHGS-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHGS-1.zip', 'version': '20201124-093005/v1.9.0@57ac4ad8'}, 'SHSM-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSM-01.zip', 'version': '20201124-091539/v1.9.0@57ac4ad8'}, 'SHHT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHHT-1.zip', 'version': '20201124-091711/v1.9.0@57ac4ad8'}, 'SHWT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHWT-1.zip', 'version': '20201128-102432/v1.9.2@e83f7025'}, 'SHDW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-1.zip', 'version': '20201124-091845/v1.9.0@57ac4ad8'}, 'SHDW-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-2.zip', 'version': '20201228-092556/v1.9.3@ad2bb4e3'}, 'SHSPOT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSPOT-1.zip', 'version': '20201124-091104/v1.9.0@57ac4ad8'}, 'SHCL-255': {'url': 'http://api.shelly.cloud/firmware/compatibility/SHCL-255/0.2.0-73a40eef.zip', 'version': '20180116-161402/v0.2.0@73a40eef'}, 'SHBLB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBLB-1.zip', 'version': '20201124-090807/v1.9.0@57ac4ad8'}, 'SHCB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHCB-1.zip', 'version': '20210107-122133/1.9_GU10_RGBW@07531e29'}, 'SHRGBW2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBW2.zip', 'version': '20201124-092125/v1.9.0@57ac4ad8'}, 'SHRGBWW-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBWW-01.zip', 'version': '20201124-090953/v1.9.0@57ac4ad8'}, 'SH2LED-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SH2LED-1.zip', 'version': '20201124-091028/v1.9.0@57ac4ad8'}, 'SHDM-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-1.zip', 'version': '20201228-093108/v1.9.3@ad2bb4e3'}, 'SHDM-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-2.zip', 'version': '20201228-093141/v1.9.3@ad2bb4e3'}, 'SHDIMW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDIMW-1.zip', 'version': '20201124-092741/v1.9.0@57ac4ad8'}, 'SHVIN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHVIN-1.zip', 'version': '20201124-090843/v1.9.0@57ac4ad8'}, 'SHBDUO-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBDUO-1.zip', 'version': '20201124-090917/v1.9.0@57ac4ad8'}, 'SHBTN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBTN-1.zip', 'version': '20201124-092005/v1.9.0@57ac4ad8'}, 'SHIX3-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHIX3-1.zip', 'version': '20201124-092930/v1.9.0@57ac4ad8'}}}

homekit_release_info:[['X2.7.0-.*', {'version': '2.7.0-beta3', 'rel_notes': 'https://github.com/mongoose-os-apps/shelly-homekit/issues/330', 'rel_notesX': '', 'urls': {'Shelly1': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1.zip', 'Shelly1L': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1L.zip', 'Shelly1PM': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1PM.zip', 'Shelly2': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly2 [... cut ...] helly-homekit-Shelly25.zip', 'ShellyI3': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyI3.zip', 'Shelly-i3': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyI3.zip', 'ShellyPlug': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyPlug.zip', 'ShellyPlugS': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyPlugS.zip'}}]]

device_scan
Scanning for Shelly devices...

Devices found: 0 Upgradeable: 0

C:\Users\Hugo Francisco\Downloads\shelly-homekit-master\tools>`

andyblac, 2021-01-08:
yeah, atm script only supports same subnet, maybe i'll add multi subnet support at a later date.

hughfr4nc15, 2021-01-08:
Same subnet is fine by me. I actually installed the VM to test the script and forgot to change the NIC to bridged.

Still don't understand why Windows 10 is not working. It is in the same subnet as the Shelly's.

andyblac, 2021-01-08:
> Same subnet is fine by me. I actually installed the VM to test the script and forgot to change the NIC to bridged.
> 
> Still don't understand why Windows 10 is not working. It is in the same subnet as the Shelly's.

strange, it as been tested on windows should work.

morettoni, 2021-01-11:
I have the same issue running the script in my mac. Seems that sometimes the device detection fails without any reason. The mac and the Shellys are connected on the same WiFi network but I get this result:

`luca@MacBook-Pro ~ % flash_shelly.py -al -v=1
app
OS: Darwin
manual_hosts: []
action: list
mode: homekit
dry_run: False
quiet_run: False
silent_run: False
hap_setup_code: False
version: False
exclude: None
variant: False
verbose: 1

stock_release_info:{'isok': True, 'data': {'SHPLG-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-1.zip', 'version': '20201124-091454/v1.9.0@57ac4ad8'}, 'SHPLG-S': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-S.zip', 'version': '20201228-092848/v1.9.3@ad2bb4e3'}, 'SHPLG-U1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-U1.zip', 'version': '20201124-092346/v1.9.0@57ac4ad8'}, 'SHPLG2-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG2-1.zip', 'version': '20201228-092942/v1.9.3@ad2bb4e3'}, 'SHSW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-1.zip', 'version': '20201228-092119/v1.9.3@ad2bb4e3'}, 'SHSW-21': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-21.zip', 'version': '20201228-092214/v1.9.3@ad2bb4e3'}, 'SHSW-25': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-25.zip', 'version': '20201228-092242/v1.9.3@ad2bb4e3'}, 'SHSW-PM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-PM.zip', 'version': '20201228-093009/v1.9.3@ad2bb4e3'}, 'SHSW-L': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-L.zip', 'version': '20201228-092147/v1.9.3@ad2bb4e3'}, 'SHAIR-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHAIR-1.zip', 'version': '20201124-092534/v1.9.0@57ac4ad8'}, 'SHSW-44': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-44.zip', 'version': '20201019-110845/v1.6.6_4Pro@bad181e4'}, 'SHUNI-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHUNI-1.zip', 'version': '20201124-093042/v1.9.0@57ac4ad8'}, 'SHEM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM.zip', 'version': '20201124-092817/v1.9.0@57ac4ad8'}, 'SHEM-3': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM-3.zip', 'version': '20201124-092854/v1.9.0@57ac4ad8'}, 'SHSEN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSEN-1_build.zip', 'version': '20200717-112934/v1.7.7@a02db0b8'}, 'SHGS-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHGS-1.zip', 'version': '20201124-093005/v1.9.0@57ac4ad8'}, 'SHSM-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSM-01.zip', 'version': '20201124-091539/v1.9.0@57ac4ad8'}, 'SHHT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHHT-1.zip', 'version': '20201124-091711/v1.9.0@57ac4ad8'}, 'SHWT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHWT-1.zip', 'version': '20201128-102432/v1.9.2@e83f7025'}, 'SHDW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-1.zip', 'version': '20201124-091845/v1.9.0@57ac4ad8'}, 'SHDW-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-2.zip', 'version': '20201228-092556/v1.9.3@ad2bb4e3'}, 'SHSPOT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSPOT-1.zip', 'version': '20201124-091104/v1.9.0@57ac4ad8'}, 'SHCL-255': {'url': 'http://api.shelly.cloud/firmware/compatibility/SHCL-255/0.2.0-73a40eef.zip', 'version': '20180116-161402/v0.2.0@73a40eef'}, 'SHBLB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBLB-1.zip', 'version': '20201124-090807/v1.9.0@57ac4ad8'}, 'SHCB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHCB-1.zip', 'version': '20210107-122133/1.9_GU10_RGBW@07531e29'}, 'SHRGBW2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBW2.zip', 'version': '20201124-092125/v1.9.0@57ac4ad8'}, 'SHRGBWW-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBWW-01.zip', 'version': '20201124-090953/v1.9.0@57ac4ad8'}, 'SH2LED-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SH2LED-1.zip', 'version': '20201124-091028/v1.9.0@57ac4ad8'}, 'SHDM-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-1.zip', 'version': '20201228-093108/v1.9.3@ad2bb4e3'}, 'SHDM-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-2.zip', 'version': '20201228-093141/v1.9.3@ad2bb4e3'}, 'SHDIMW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDIMW-1.zip', 'version': '20201124-092741/v1.9.0@57ac4ad8'}, 'SHVIN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHVIN-1.zip', 'version': '20201124-090843/v1.9.0@57ac4ad8'}, 'SHBDUO-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBDUO-1.zip', 'version': '20201124-090917/v1.9.0@57ac4ad8'}, 'SHBTN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBTN-1.zip', 'version': '20201124-092005/v1.9.0@57ac4ad8'}, 'SHIX3-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHIX3-1.zip', 'version': '20201124-092930/v1.9.0@57ac4ad8'}}}

homekit_release_info:[['X2.7.0-.*', {'version': '2.7.0-beta3', 'rel_notes': 'https://github.com/mongoose-os-apps/shelly-homekit/issues/330', 'rel_notesX': '', 'urls': {'Shelly1': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1.zip', 'Shelly1L': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1L.zip', 'Shelly1PM': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1PM.zip', 'Shelly2': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly2.zip', 'Shelly25': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly25.zip', 'ShellyI3': 'https://rojer.me/files/shelly/2.7.0-beta3 [... cut ...] /2.7.1/shelly-homekit-Shelly25.zip', 'ShellyI3': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyI3.zip', 'Shelly-i3': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyI3.zip', 'ShellyPlug': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyPlug.zip', 'ShellyPlugS': 'https://rojer.me/files/shelly/2.7.1/shelly-homekit-ShellyPlugS.zip'}}]]

device_scan
Scanning for Shelly devices...

Devices found: 0 Upgradeable: 0
luca@MacBook-Pro ~ % 
`

Thanks in advance for any hint!

andyblac, 2021-01-11:
> I have the same issue running the script in my mac. Seems that sometimes the device detection fails without any reason. The mac and the Shellys are connected on the same WiFi network but I get this [...]
are your devices on the same subnet ?

can you try the lastest beta script 

[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5797368/flash-shelly.py.zip)

can you with following command, i should create a file called debug.log can you post it i'll take a look.
`./flash-shelly.py -la -m=keep -v=5 --log-file=debug.log`

morettoni, 2021-01-11:
> > I have the same issue running the script in my mac. Seems that sometimes the device detection fails without any reason. The mac and the Shellys are connected on the same WiFi network but I get this result:
> > ```
> > luca@MacBook-Pro ~ % flash_shelly.py -al -v=1
> > app
> > OS: Darwin
> > manual_hosts: []
> > action: list
> > mode: homekit
> > dry_run: False
> > quiet_run: False
> > silent_run: False
> > hap_setup_code: False
> > version: False
> > exclude: None
> > variant: False
> > verbose: 1
> > 
> > stock_release_info:{'isok': True, 'data': {'SHPLG-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-1.zip', 'version': '20201124-091454/v1.9.0@57ac4ad8'}, 'SHPLG-S': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-S.zip', 'version': '20201228-092848/v1.9.3@ad2bb4e3'}, 'SHPLG-U1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-U1.zip', 'version': '20201124-092346/v1.9.0@57ac4ad8'}, 'SHPLG2-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG2-1.zip', 'version': '20201228-092942/v1.9.3@ad2bb4e3'}, 'SHSW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-1.zip', 'version': '20201228-092119/v1.9.3@ad2bb4e3'}, 'SHSW-21': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-21.zip', 'version': '20201228-092214/v1.9.3@ad2bb4e3'}, 'SHSW-25': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-25.zip', 'version': '20201228-092242/v1.9.3@ad2bb4e3'}, 'SHSW-PM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-PM.zip', 'version': '20201228-093009/v1.9.3@ad2bb4e3'}, 'SHSW-L': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-L.zip', 'version': '20201228-092147/v1.9.3@ad2bb4e3'}, 'SHAIR-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHAIR-1.zip', 'version': '20201124-092534/v1.9.0@57ac4ad8'}, 'SHSW-44': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-44.zip', 'version': '20201019-110845/v1.6.6_4Pro@bad181e4'}, 'SHUNI-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHUNI-1.zip', 'version': '20201124-093042/v1.9.0@57ac4ad8'}, 'SHEM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM.zip', 'version': '20201124-092817/v1.9.0@57ac4ad8'}, 'SHEM-3': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM-3.zip', 'version': '20201124-092854/v1.9.0@57ac4ad8'}, 'SHSEN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSEN-1_build.zip', 'version': '20200717-112934/v1.7.7@a02db0b8'}, 'SHGS-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHGS-1.zip', 'version': '20201124-093005/v1.9.0@57ac4ad8'}, 'SHSM-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSM-01.zip', 'version': '20201124-091539/v1.9.0@57ac4ad8'}, 'SHHT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHHT-1.zip', 'version': '20201124-091711/v1.9.0@57ac4ad8'}, 'SHWT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHWT-1.zip', 'version': '20201128-102432/v1.9.2@e83f7025'}, 'SHDW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-1.zip', 'version': '20201124-091845/v1.9.0@57ac4ad8'}, 'SHDW-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-2.zip', 'version': '20201228-092556/v1.9.3@ad2bb4e3'}, 'SHSPOT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSPOT-1.zip', 'version': '20201124-091104/v1.9.0@57ac4ad8'}, 'SHCL-255': {'url': 'http://api.shelly.cloud/firmware/compatibility/SHCL-255/0.2.0-73a40eef.zip', 'version': '20180116-161402/v0.2.0@73a40eef'}, 'SHBLB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBLB-1.zip', 'version': '20201124-090807/v1.9.0@57ac4ad8'}, 'SHCB-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHCB-1.zip', 'version': '20210107-122133/1.9_GU10_RGBW@07531e29'}, 'SHRGBW2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBW2.zip', 'version': '20201124-092125/v1.9.0@57ac4ad8'}, 'SHRGBWW-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHRGBWW-01.zip', 'version': '20201124-090953/v1.9.0@57ac4ad8'}, 'SH2LED-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SH2LED-1.zip', 'version': '20201124-091028/v1.9.0@57ac4ad8'}, 'SHDM-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-1.zip', 'version': '20201228-093108/v1.9.3@ad2bb4e3'}, 'SHDM-2': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDM-2.zip', 'version': '20201228-093141/v1.9.3@ad2bb4e3'}, 'SHDIMW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDIMW-1.zip', 'version': '20201124-092741/v1.9.0@57ac4ad8'}, 'SHVIN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHVIN-1.zip', 'version': '20201124-090843/v1.9.0@57ac4ad8'}, 'SHBDUO-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBDUO-1.zip', 'version': '20201124-090917/v1.9.0@57ac4ad8'}, 'SHBTN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHBTN-1.zip', 'version': '20201124-092005/v1.9.0@57ac4ad8'}, 'SHIX3-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHIX3-1.zip', 'version': '20201124-092930/v1.9.0@57ac4ad8'}}}
> > 
> > homekit_release_info:[['X2.7.0-.*', {'version': '2.7.0-beta3', 'rel_notes': 'https://github.com/mongoose-os-apps/shelly-homekit/issues/330', 'rel_notesX': '', 'urls': {'Shelly1': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1.zip', 'Shelly1L': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1L.zip', 'Shelly1PM': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1PM.zip', 'Shelly2': 'https://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly2.zip', 'Shelly25': 'https://rojer.me/files/shelly/2.7.0-beta3/sh [... cut ...]
> > Thanks in advance for any hint!
> > are your devices on the same subnet ?
> 
> can you try the lastest beta script
> 
> [flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5797368/flash-shelly.py.zip)
> 
> can you run with following command, it should create a file called debug.log can you post it i'll take a look.
> `./flash-shelly.py -la -m=keep -v=5 --log-file=debug.log`

Today the script seems running fine, I will try next days and I will send you the output!
Thanks!

timoschilling, 2021-01-12:
I have the same issue on Mac, from time to time it doesn't find any or just some of my devises. immediately doing a rerun shows all devises.

andyblac, 2021-01-12:
> I have the same issue on Mac, from time to time it doesn't find any or just some of my devises. immediately doing a rerun shows all devises.

can you try the above, and post log file please.

morettoni, 2021-01-12:
> > I have the same issue on Mac, from time to time it doesn't find any or just some of my devises. immediately doing a rerun shows all devises.
> 
> can you try the above, and post log file please.

Today I got again the problem and you can find the log in the attachment. I used the command line you proposed with the Python script updated with the one in your zip file.

[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/5802332/debug.log)

andyblac, 2021-01-12:
> Today I got again the problem and you can find the log in the attachment. I used the command line you proposed with the Python script updated with the one in your zip file. [...]

hmm, the python dns browser did not find any devices after scanning for 20 seconds

```
2021-01-12 14:25:23,869 DEBUG 615 [Device Scan] automatic scan
2021-01-12 14:25:23,870 INFO 616 Scanning for Shelly devices...
2021-01-12 14:25:43,875 INFO 625 
2021-01-12 14:25:43,876 INFO 626 Devices found: 0 Upgradeable: 0 Flashed: 0
2021-01-12 14:25:43,876 INFO 628 Log file created: debug.log
```

what kind of network do you have ?

morettoni, 2021-01-12:
what do you mean?
I have a wifi network at home with 3 different access point, but all the device will get the IP inside the same network and there is only a WLAN…
Let me know if you need more informations...

> I have the same issue on Mac, from time to time it doesn't find any or just some of my devises. immediately doing a rerun shows all devises.
>
> can you try the above, and post log file please.
>
> Today [...]

andyblac, 2021-01-12:
> what do you mean? I have a wifi network at home with 3 different access point, but all the device will get the IP inside the same network and there is only a WLAN… Let me know if you need more informations...
> […](#)

it is a mesh, or are ap's linked via lan cable, what is manufacture ? do you use VLAN's ?, is mac on smame WiFi SSID or is i hardwired ?, do you use any special domian structure or standard .local ?

morettoni, 2021-01-12:
no mesh, each AP is wired to the switch and I have only one single SSID; also no special domain structure too...

> On 12 Jan 2021, at 14:37, Andy Blackburn <notifications@github.com> wrote:
> 
> it is a mesh, or are ap's linked via lan cable, what is manufacture ? do you use VLAN's ?, is mac on smame WiFi SSID or is i hardwired ?, do you use any special domian structure or standard .local ?
> 
> 

--
Luca Morettoni <luca(AT)morettoni.net> | http://www.morettoni.net
http://it.linkedin.com/in/morettoni/ | http://twitter.com/morettoni

andyblac, 2021-01-12:
> no mesh, each AP is wired to the switch and I have only one single SSID; also no special domain structure too...
> […](#)

hmm ok, and only happends every now and then, not everytime ?

can you run `timeout 20  dns-sd -B` post output.

morettoni, 2021-01-12:
not evertime; sorry I didn’t get the command I need to run...

> On 12 Jan 2021, at 14:43, Andy Blackburn <notifications@github.com> wrote:
> 
> hmm ok, and only happends every now and then, not everytime ?
> 
> can you run timeout 20 dns-sd -B post output.
> 
> 

--
Luca Morettoni <luca(AT)morettoni.net> | http://www.morettoni.net
http://it.linkedin.com/in/morettoni/ | http://twitter.com/morettoni

andyblac, 2021-01-12:
`timeout 20  dns-sd -B` run in terminal

morettoni, 2021-01-12:
maybe is not included in MacOS commands:

luca@MacBook-Pro ~ % timeout 20 dns-sd -B
zsh: command not found: timeout

> 
> 
> timeout 20 dns-sd -B run in terminal

andyblac, 2021-01-12:
ok try standard `dns-sd -B`. do a CTRL-C after 20 seconds

morettoni, 2021-01-12:
Same results, attached again the log file if needed...

[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/5802472/debug.log)

andyblac, 2021-01-12:
> Same results, attached again the log file if needed...
> 
> [debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/5802472/debug.log)

sorry you need to copy paste result from that command, it does not create a file, did it show any info ?

andyblac, 2021-01-12:
like 
```
Browsing for _http._tcp
DATE: ---Tue 12 Jan 2021---
13:54:39.586  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
13:54:39.587  Add        3   7 local.               _http._tcp.          shellyrgbw2-B075D3
13:54:39.587  Add        3   7 local.               _http._tcp.          shellydimmer-F3CD89
13:54:39.587  Add        3   7 local.               _http._tcp.          shellyrgbw2-B03CAE
13:54:39.587  Add        3   7 local.               _http._tcp.          shellydimmer2-40F520015A2D
13:54:39.587  Add        3   4 local.               _http._tcp.          UniFi Controller (192_168_2_14)
13:54:39.587  Add        3   7 local.               _http._tcp.          shellyrgbw2-B0C8BB
13:54:39.587  Add        3   7 local.               _http._tcp.          shellydw2-D8BFC01A7003
13:54:39.587  Add        3   7 local.               _http._tcp.          shellyrgbw2-B0A3EF
13:54:39.587  Add        3   7 local.               _http._tcp.          Bathroom-SW
13:54:39.587  Add        3   7 local.               _http._tcp.          shellydimmer2-40F52000B1C9
13:54:39.587  Add        3   7 local.               _http._tcp.          MediaRoom-Lamps-Button
13:54:39.587  Add        3   7 local.               _http._tcp.          Landing-Alarm-SW
13:54:39.587  Add        3   7 local.               _http._tcp.          Kitchen-Light-SW
13:54:39.587  Add        3   7 local.               _http._tcp.          shellyrgbw2-B09A29
13:54:39.587  Add        3   7 local.               _http._tcp.          Andys-Bedroom-SW
13:54:39.587  Add        3   7 local.               _http._tcp.          Garage-Light-SW
13:54:39.587  Add        3   7 local.               _http._tcp.          Lounge-Light-SW
13:54:39.587  Add        2   7 local.               _http._tcp.          Mums-Bedroom-SW
```

morettoni, 2021-01-12:
my output was just empty:

luca@MacBook-Pro ~ % dns-sd -B
Browsing for _http._tcp
DATE: ---Tue 12 Jan 2021---
14:56:02.527  ...STARTING...

> like
>
> Browsing for _http._tcp
> DATE: ---Tue 12 Jan 2021---
> 13:54:39.586  ...STARTING...
> Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
> 13:54:39.587  Add [...]

andyblac, 2021-01-12:
hmm, then it looks like a network issue. some how, try leaveing longer, maybe a minuet see if it helps.

morettoni, 2021-01-12:
is it correct to run that command (keeping it running) and in another console run the script? I will keep the dns command running for a quite long time and I will try it again...

> 
> 
> hmm, then it looks like a network issue. some how, try leaveing longer, maybe a minuet see if it helps.

andyblac, 2021-01-12:
> is it correct to run that command (keeping it running) and in another console run the script? I will keep the dns command running for a quite long time and I will try it again...
> […](#)

no need to run the script, we are just trying to detect any devices on network manually.

andyblac, 2021-01-12:
`dns-sd` is a browser that should list all device that has http access on you network.

morettoni, 2021-01-12:
sorry but I din’t understand how to run both… can you explain it to me?

> 
> 
> is it correct to run that command (keeping it running) and in another console run the script? I will keep the dns command running for a quite long time and I will try it again...
> … <x-msg://45/#>
> no need to run the script, we are just trying to detect any devices on network manually.

andyblac, 2021-01-12:
you dont, just run the `dns-sd -B`. that all i need atm, if it does not show anything after 60 seconds, something is strnage with your network.

morettoni, 2021-01-12:
i will do some investigations and I will come back to you. Thanks for the moment for your support!

> 
> 
> dns-sd is a browser that should list all device that has http access on you network.

andyblac, 2021-01-12:
> i will do some investigations and I will come back to you. Thanks for the moment for your support!
> […](#)

np

morettoni, 2021-01-12:
> you dont, just run the `dns-sd -B`. that all i need atm, if it does not show anything after 60 seconds, something is strange with your network.

I have done a simple test: I deactivated the WiFi of my MacBook and enabled it again, and after that I got this:

```
luca@MacBook-Pro ~ % dns-sd -B
Browsing for _http._tcp
DATE: ---Tue 12 Jan 2021---
15:15:00.491  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
15:15:00.685  Add        3   6 local.               _http._tcp.          shelly1-presa-terrazzo
15:15:00.685  Add        3   6 local.               _http._tcp.          shelly25-studio-bagno
15:15:00.685  Add        3   6 local.               _http._tcp.          shelly1-giardino-grande
15:15:00.685  Add        3   6 local.               _http._tcp.          shelly1-ingresso-pedonale
15:15:00.685  Add        3   6 local.               _http._tcp.          shelly1-giardino-auto
15:15:00.685  Add        3   6 local.               _http._tcp.          shelly1-soggiorno-1
15:15:00.685  Add        3   6 local.               _http._tcp.          shellyswitch25-soggiorno-2-3
15:15:00.685  Add        3   6 local.               _http._tcp.          shellyswitch25-giardino-piccolo
15:15:00.685  Add        3   6 local.               _http._tcp.          schelly1-scale-luce
15:15:00.685  Add        3   6 local.               _http._tcp.          shellyswitch25-lavanderia-scale
15:15:00.685  Add        2   6 local.               _http._tcp.          shelly1-cucina-luce
15:15:00.886  Add        2   6 local.               _http._tcp.          shellyplugs-9A34B6
15:15:28.346  Rmv        1   6 local.               _http._tcp.          shelly1-presa-terrazzo
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly25-studio-bagno
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-giardino-grande
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-ingresso-pedonale
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-giardino-auto
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-soggiorno-1
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shellyswitch25-soggiorno-2-3
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shellyswitch25-giardino-piccolo
15:15:28.347  Rmv        1   6 local.               _http._tcp.          schelly1-scale-luce
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shellyswitch25-lavanderia-scale
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-cucina-luce
15:15:28.347  Rmv        0   6 local.               _http._tcp.          shellyplugs-9A34B6
```

but if I run the script I get 0 devices listed and the debug.log file is exactly the same as before. I hope this could help you...

andyblac, 2021-01-12:
how long it it take to show ?, try running gain to confirm it still ok.

andyblac, 2021-01-12:
oh look, after 28 seconds devcies are being removed, no idea why
```
15:15:28.346  Rmv        1   6 local.               _http._tcp.          shelly1-presa-terrazzo
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly25-studio-bagno
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-giardino-grande
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-ingresso-pedonale
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-giardino-auto
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-soggiorno-1
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shellyswitch25-soggiorno-2-3
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shellyswitch25-giardino-piccolo
15:15:28.347  Rmv        1   6 local.               _http._tcp.          schelly1-scale-luce
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shellyswitch25-lavanderia-scale
15:15:28.347  Rmv        1   6 local.               _http._tcp.          shelly1-cucina-luce
15:15:28.347  Rmv        0   6 local.               _http._tcp.          shellyplugs-9A34B6
```

RMV mean remove.

andyblac, 2021-01-12:
@rojer any idea's i'm at aloss here ?

morettoni, 2021-01-12:
For example now I have only the "add" (they appear immediately) and no remove at all...

andyblac, 2021-01-12:
@rojer could it be related to https://github.com/mongoose-os-apps/shelly-homekit/issues/108

andyblac, 2021-01-12:
> For example now I have only the "add" (they appear immediately) and no remove at all...

and script ?

morettoni, 2021-01-12:
sorry… and the script works correctly!

> 
> 
> For example now I have only the "add" (they appear immediately) and no remove at all...
> 
> and script ?

andyblac, 2021-01-12:
> sorry… and the script works correctly!
> […](#)

ye, something strnage going on, on your network side, but what that is the question, as it seems your not the only one with the issue, does Home app seem effected ?, or is that perfectly ok?

morettoni, 2021-01-12:
Home app works fine, the Shelly are visible (and usable) even if not detected by the script; I have only two that sometimes disconnect (a Shelly1 and a Shelly25), but I have to investigate if the place is too far from the WiFi antenna...

> On 12 Jan 2021, at 15:28, Andy Blackburn <notifications@github.com> wrote:
> 
> ye, something strange going on, on your network side, but what that is the question, as it seems your not the only one with the issue, does Home app seem effected ?, or is that perfectly ok?
> 
> 

--
Luca Morettoni <luca(AT)morettoni.net> | http://www.morettoni.net
http://it.linkedin.com/in/morettoni/ | http://twitter.com/morettoni

andyblac, 2021-02-20:
closing as op has not responded.

closed 2021-02-20 by andyblac