mongoose-os-apps/shelly-homekit #397: Shelly 1 with HomeKit Firmware resented and can’t intigrate in my Home app anymore

Steve1468, 2021-01-10:
Hello I have reserved my Shelly 1 with HomeKit Firmeware 2.7.1, after resetting I can’t reach my Shelly, I see it as a WiFi in my iPhone and I have also the IP but I can’t intigrate in my Home App. What can I do

andyblac, 2021-01-10:
if you have rest, the devcie will be back in AP mode, you will have to resetup to your wifi, please see here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Device-webUI

Steve1468, 2021-01-10:
Thanks a lot, I got to the Shelly 1 IP Address then on the Webside https://192.168.33.1 but then I got stuck again. I tried to change the IP Address to get it on my Network nothing. I can also see it on the Shelly app, but I can't access to it because Shelly app says: WiFi connection problems.

andyblac, 2021-01-10:
once in the website do this https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-Settings

andyblac, 2021-01-10:
once that is done do this https://github.com/mongoose-os-apps/shelly-homekit/wiki/HomeKit-settings

andyblac, 2021-01-10:
once that is done follow this https://github.com/mongoose-os-apps/shelly-homekit/wiki/Adding-device

Steve1468, 2021-01-10:
I do not arrive to the WiFi Settings only the first one the Device web-UI and that's it

andyblac, 2021-01-10:
scroll down it all one page. show me what you have.

simonfield76, 2021-01-10:
> Thanks a lot, I got to the Shelly 1 IP Address then on the Webside https://192.168.33.1 but then I got stuck again. I tried to change the IP Address to get it on my Network nothing. I can also see it on the Shelly app, but I can't access to it because Shelly app says: WiFi connection problems.

Hard reset the shelly first by oppropriate button if available otherwise disconnect the device from the current and reconnect it. Within the first minute connect a wire to sw and with the other end of the wire touch 5 times either L or L1 the relay will click several times in order to confirm the hard reset.

Then open the web interface by connecting to ap created from the shelly and place wifi settings plus a static ip out of the dhcp pool. Then save and reconnect to the web server using the ip assigned. I've always used this method with all my shellys and to help a friend.

After this you can flash your.

Steve1468, 2021-01-10:
Tried it also with the Shelly app after another reset, but the Shelly 1 will not be included in my WiFi Network even when block the 5GHz network for a while

andyblac, 2021-01-10:
if your 2.4ghz on 20Mhz band ? shelly's do not support 40Mhz

andyblac, 2021-01-10:
> Tried it also with the Shelly app after another reset, but the Shelly 1 will not be included in my WiFi Network even when block the 5GHz network for a while

shelly app will not work when using HomeKit firmware.

Steve1468, 2021-01-10:
Thanks a lot, I will change the Shelly and I will use a new one. Thanks also for the HomeKit version it's a nice thing.

andyblac, 2021-01-10:
> Thanks a lot, I will change the Shelly and I will use a new one. Thanks also for the HomeKit version it's a nice thing.

if you can get access to the webUI it should be able to be configured correctly.

jsiegenthaler, 2021-01-10:
> is your 2.4ghz on 20Mhz band ? shelly's do not support 40Mhz

For clarification for readers: andy is referring to the channel bandwidth, not the carrier band.

Wifi operates on the 2.4GHz band with a channel bandwidth of various 5/10/20/40 and more MHz.

i didn't know that Shelly doesn't support channel bandwidth higher than 20MHz.
I did find out that they work well with my old 802.11a/b wifi router, but not with my new 802.11ac running at 40 or more MHz channel bandwidth.

I never bothered to research it further, I just setup a new access point for my Shelly stuff.

andyblac, 2021-01-25:
this still an issue ?, or can it be closed ?

Steve1468, 2021-01-25:
Hi Andy, I have changed the Shelly and solved it with that.
Today I saw, when I unplug the reseted Shelly and put it elsewhere the. I can reuse it.
Thanks a lot for your help.
And yes you can close it.
Stay safe and thanks again.

Stefan
Cell: +39 3406492025
email: stefanlanger@mac.com
WhatsApp: +39 3406492025

http://www.thelangerbooks.com
Author of:

   and

andyblac, 2021-01-25:
thanks for the update.

closed 2021-01-25 by andyblac