mongoose-os-apps/shelly-homekit #399: Shelly 1 dropping out of Wifi and shutting off

Rithord, 2021-01-11:
My Shelly 1 is not connecting to my home network anymore, as soon as I turn the power off for the device and turn it back on. It temporarily appears on the network and then disappears.

At first I was able to get to the web page for the settings but as soon as I would try to do anything it would timeout and disconnect.

This happened while I was trying to update the device I believe. I have reset the Shelly with the SW 5 times and it makes the relay noises to confirm. The AP is visible and I can connect, about 20 seconds later it disconnects completely again... 

Any ideas here?

andyblac, 2021-01-12:
sound like wifi issues are you on a "Mesh" network ?

andyblac, 2021-03-03:
is this still an issue ?

closed 2021-03-11 by github-actions[bot]