mongoose-os-apps/shelly-homekit #403: Bug in Firmware 2.71

hbednarek, 2021-01-13:
When I had 2.70 on my Shelly1 it was possible to make a "Automation". When Shelly1 in Bathroorm is "on" then another Shelly1 in Bathroom  starts and is "on" too. After loading the new firmware 2.71 zhis does not work. But switshing via Handy does work.

So I downgraded this Shelly and it just works again.

andyblac, 2021-01-16:
i have no issues here with 2.7.1 and automations, can you give an expample what does not work, so i can test it.

hbednarek, 2021-01-17:
It´s just as I wrote: One shelly 1 in the Bathroom and a second one in a lighted mirror in the bathroom. Both have a switch. Then I made a "automation": When Bathromm light is on then mirror shall light also. And when Bathroom light switches off then mirror light goes "off" too. That worked with 2.70 but wgen u upgraded to Version 2.71 the conncetion between the two lamps is broken. After downgrading to 2.70 the conncetion does work again. From my IPHONE both lamps are switchable via WLAN. So, the WLAN Connection is not the problem.  And the mirror shelly contains 2.71 all the time. Even now. Thats my example.

pivale, 2021-01-18:
Can you try recreating your automation?
I have a really similar setup (hallway, not bathroom) and all is working as intended with 2.7.1

rojer, 2021-01-18:
i'm struggling to think what change in 2.7.1 vs 2.7.0 could have caused it...

andyblac, 2021-01-18:
> i'm struggling to think what change in 2.7.1 vs 2.7.0 could have caused it...

me too, all my sutomations work np, and i have loads.

andyblac, 2021-01-18:
@hbednarek can you try deleting the automation and re-creating it, it could have gotten corrupted.

jobl1978, 2021-01-19:
I have the exact same automation in my kitchen with 2 shellys. In my case it works with 2.7.1.

andyblac, 2021-02-20:
still an issue ?

hbednarek, 2021-02-20:
No, 

Ihm using 2.73 and everything works. Thank you.

closed 2021-02-20 by andyblac