mongoose-os-apps/shelly-homekit #405: Use second input of Shelly1L

jeroendedeyne, 2021-01-15:
Is it possible to use the first input to control the relay and the second input as a detached input to control something else in your HomeKit?

rojer, 2021-01-16:
this is exactly how it works right now. second input is exported as a stateless switch to HAP.

andyblac, 2021-01-16:
@rojer any chance fixing it, as it broken atm, there is no "Input2"

rojer, 2021-01-16:
indeed, this was a regression introduced while adding BL0937 power measurement support in https://github.com/mongoose-os-apps/shelly-homekit/commit/ca0674d5905baa6d22de2ca5646678bebbc9439a.

Fixed in https://github.com/mongoose-os-apps/shelly-homekit/commit/a1044b907ad5f02a8b0e3820494255691319a169

closed 2021-01-16 by rojer