mongoose-os-apps/shelly-homekit #406: Single button issue

marienaz, 2021-01-16:
My alarm is an always openiswitch like my garage door opener... So when I setup shelly 1 as a normal switch (edge or any type, I have o toggle the button 2x before alarm is activated or deactivated.  

If I set the shelly connected to alarm as a "garage door opener it functions 100%, but that in homekit the icon shows garage doors.. Is there any way the garage door opener "settings" as on the homekit firmware can be copied but shows as a switch or something else?

andyblac, 2021-01-16:
please explain more, have you tried "Momentary" switch type.

marienaz, 2021-01-17:
> please explain more, have you tried "Momentary" switch type.

I have tried it but still nothing, i can here the switch closing, but it has to open again.. so when i toggle the momentary switch again, then the circuit opens and the alarm is activated.. the alarm works in the same manner as the garage door opener..
the system is looking for a impulse (contacts shorting for a small period of time) .

andyblac, 2021-01-17:
ok, try setting a auto off to 1 second, that should open the relay after second,

marienaz, 2021-01-17:
> ok, try setting a auto off to 1 second, that should open the relay after 1 second,

Hi Andy thanx for the help, i tried that allready, but then the switch goes back to the off position so i never know if its really on or off

andyblac, 2021-01-17:
try with the following

[image]

it should make the switch open = relay closed, switch closed = relay open.

marienaz, 2021-01-17:
> try with the following
> 
> [image]
> 
> it should make the switch open = relay closed, switch closed = relay open., it will work best with a push button type switch aka Momentary switch.
> 
> so when you push the button it opens the relay, and when you release it, it closes the relay aka ON.

Nope... then i have to toggle the switch on and off before the alarm is activated... and again on and off to deactivate alarm...

andyblac, 2021-01-17:
> Nope... then i have to toggle the switch on and off before the alarm is activated... and again on and off to deactivate alarm...

and with auto off enabled ?

marienaz, 2021-01-17:
auto off does not seem to work at all with this settings..

andyblac, 2021-01-17:
hmmm what time did you set it to ? it starts as soon as you press, so if your still holding after timeout it will not be activated, so do not set to short.

andyblac, 2021-01-17:
are you using a toggle switch aka rocker? or a push button ?

marienaz, 2021-01-17:
i use a button similar to the remote of a garage door opener.. but im trying this with the homekit software, so if tried the toggle switch on the firmware web interface and on my iphone.. auto off I tried 500ms and 5 seconds

andyblac, 2021-01-17:
ok your correct i have just tested auto off with "Activation" does not work, can you set to "Momentary" but leave "Inverted" and "Auto Off". enabled
 try that .

marienaz, 2021-01-17:
then it works but the switch state changes on and back off again... so i cant see on the homkit icon whether its on or off..  ...
so alarms is off, I click the button and when it auto off's the alarm is actavated, but then the switch still shows off...

andyblac, 2021-01-17:
ok then create a new ticket, asking for an "Inverted relay" option, and when describing say you would like the HomeKit icon to show as ON when the relay is OPEN (OFF), and HomeKit icon to show as OFF when relay is CLOSED (ON).

marienaz, 2021-01-17:
The garage door opener function works 100% on the alarm button shelly, but then it shows a garage door in homekit instead of a switch...

andyblac, 2021-01-17:
> The garage door opener function works 100% on the alarm button shelly, but then it shows a garage door in homekit instead of a switch...

yes, that fuction has those options, you just need then adding to SWITCH. link the issue to this discussion.

marienaz, 2021-01-17:
Open a ticket? same as this issue ? or where do I open a request or ticket?

andyblac, 2021-01-17:
> Open a ticket? same as this issue ? or where do I open a request or ticket?

yes sorry, Issue, i call them Tickets as in support tickets. open a new Issue asking for the above. and link it to this issue.

andyblac, 2021-01-17:
i'll do it for you.

andyblac, 2021-01-17:
are you using a Shelly1 ?? in low voltage or as dry contact relay ?

marienaz, 2021-01-17:
Its a shelly 1 yes... on 12v

andyblac, 2021-01-17:
thought so..

andyblac, 2021-01-17:
add you support to the issue i created for you.

andyblac, 2021-01-17:
i'll close this issue,

closed 2021-01-17 by andyblac