mongoose-os-apps/shelly-homekit #408: Add: Inverted Relay

andyblac, 2021-01-17:
can we add an option to the "Switch" mode for "Inverted Relay"

Description:
So the HomeKit icon shows as ON when the relay is OPEN (OFF), and HomeKit icon shows as OFF when relay is CLOSED (ON).

as per disscussion here https://github.com/mongoose-os-apps/shelly-homekit/issues/406

rojer, 2021-01-17:
so, an "inverted output" mode in addition to "inverted input" checkbox we have now?

andyblac, 2021-01-17:
> so, an "inverted output" mode in addition to "inverted input" checkbox we have now?

yup, thats what hes after.

Steve1468, 2021-02-06:
Okay thanks

closed 2021-02-22 by rojer

rojer, 2021-02-23:
please install 2.8.0-alpha2 and help test

marienaz, 2021-03-13:
I saw it was added like you asked for andy which is awesome, but it still does not work correctly...

i still have to toggle the shelly on and off before the alarm is activated or deactivated...

when the shelly is connected to my alarm and i set it to garage door opener it functions  correctly. But shows a garage door instead of the on off switch

andyblac, 2021-03-13:
> I saw it was added like you asked for andy which is awesome, but it still does not work correctly...
>
> i still have to toggle the shelly on and off before the alarm is activated or deactivated... [...]

please show me your setup, and describe again what your after.

marienaz, 2021-03-13:
I have an alarm system that activates via remote.  The same remote as a gate motor or garage door opener... the switch is a contact or  always open switch... the same as a garage door motor.. whereby the contact closes and opens again to activate the motor.. 

so when i use any switch type in the shelly software, and i toggle the switch (from off to on), it closes the contact of the alarm system, then i have to toggle it again, (from off to on) the contact opens and the alarm is activated...

so when i set the switch to auto off, it does the job, but then the switch shows off all the time whether the alarm
Is on or off

if i set my software to garage door, and connect the shelly to my alarm system it works 100% but then its not a switch on homekit its a garage door picture...

andyblac, 2021-03-13:
> I have an alarm system that activates via remote. The same remote as a gate motor or garage door opener... the switch is a contact or always open switch... the same as a garage door motor.. whereby [...]

ok i uderstand, then you need to add your suppor to this issue, https://github.com/mongoose-os-apps/shelly-homekit/issues/552, as homekit will not beable to know just by a relay what state the alarm is, you will need someway for the shely the read the state from your alarm panel.

marienaz, 2021-03-13:
No, i just need the same as a garage door opener setting, but let me be anle to change the hap type to switch instead of garage

andyblac, 2021-03-13:
> No, i just need the same as a garage door opener setting, but let me be anle to change the hap type to switch instead of garage

as i said, dont think it posible on a switch type.

PR #462 (2021-02-22): add invert output option
closes #408