mongoose-os-apps/shelly-homekit #409: Issue when using static IPs ?

iDevelopControlTechnology, 2021-01-17:
I think there is an issue with static IP's :
All my devices are configured with static IPs and after an update from V2.6.1 to V2.7.1 or V2.7.2 ,
I am not able to reach the devices anymore - Only a firmware downgrade to V2.6.1 fixes the issue.

Unfortunately, there is no menu for the IP settings, 
so I am not able to check or change settings after the update... 

But I think, the settings must be correct, because after downgrading, everythins works fine again...

andyblac, 2021-01-17:
> I think there is an issue with static IP's :
> All my devices are configured with static IPs and after an update from V2.6.1 to V2.7.1 or V2.7.2 ,
> I am not able to reach the devices anymore - Only a [...]

there is no option for static ip's, do you mean dhcp reservation ?

rojer, 2021-01-17:
there is no option in teh UI but the firmware will actually pick up and use static IP configured in the stock firmware (they use the same underlying configuration). so it should work. i'll check it out.

andyblac, 2021-01-17:
> there is no option in teh UI but the firmware will actually pick up and use static IP configured in the stock firmware (they use the same underlying configuration). so it should work. i'll check it out.

ahh, ok

iDevelopControlTechnology, 2021-01-17:
Yes I configured it in the original stock firmware before flashing to Homekit V2.6.1

rojer, 2021-01-19:
flashed 1L with configured static IP to HK 2.7.2 - ok, then downgraded to 2.6.1 - ok, 2.6.1 to 2.7.2 again - still ok.
seems to be working fine for me.

iDevelopControlTechnology, 2021-01-23:
Maybe only the Shelly 1 is effected by this issue ? I will double check this with a Shelly Plug S and Shelly 2.5

grizz479, 2021-01-28:
7 shelly1 up and running with static íps here, whith 2.7.2 up to yesterday, and now with 2.7.3

andyblac, 2021-02-08:
can this be closed? @iDevelopControlTechnology do you still have issues ?

andyblac, 2021-03-03:
closing as no responce

closed 2021-03-03 by andyblac