mongoose-os-apps/shelly-homekit #410: Shelly-homekit roller shutter wifi re-enable over SW input from physical switch

fahadalzoabi, 2021-01-18:
Hi there, 

I came across the following issue:
Shelly 2.5 homekit (roller shutter) working just fine, I changed the wifi router user name/ password, Shelly 2.5 disconnected from the wifi router due to the aforementioned reason.
I thought I will just look up into my phone for the Shelly wifi then go to the ip page and reconnect it to the wifi router with the new name/password, unfortunately I couldn’t find the shelly wifi - I’m not sure shelly wifi was actually disabled or it was connected at some level to the router but I’m sure it wasn’t there to control it or to reset it- 
On the manufacturer’s instructions they have two ways to reboot the shelly device 1) from the device click the button on the back 2) cutt off the power and turn it on again and within 3 minutes toggle one of the SW inputs five times (basically up or down), any how that didn’t work and I have 5 Shelly devices all mounted inside the wall.

I overcome this issue by renaming the wifi router back to the name / password that shelly recognize hoping that it will reconnect and they all did.

Maybe there is a way that I don’t know but how to reconnect or re-enable the wifi on the shelly with homekit features, if there is please a simple guidelines will do, if there is nothing how about in the comming update, physical routine of cutting the power then within 3-5 minutes toggling the SW inputs (up or down) # of times will reboot ?

Thanks

rojer, 2021-01-18:
shellies will try to connect to the configured wifi network and not revert to AP on their own.
shelly app is not really supported by this firmware.

what we need is a way to enter "fallback" credentials, so you can smoothly migrate devices from one network to another by entering new credentials as backup first, then changing the setup. mongoose os supports up to 3 wifi configs that it will try one after another, we just need to expose this in the UI.

andyblac, 2021-01-20:
i'll link this here for referance https://github.com/mongoose-os-apps/shelly-homekit/issues/313

andyblac, 2021-03-03:
no responce closing....

closed 2021-03-03 by andyblac