mongoose-os-apps/shelly-homekit #413: Setting fixed ip in flashed device

grizz479, 2021-01-18:
Hello,
I am trying to set a new fixed ip for the flashed shelly devices I installed through curl - POST command, but not getting any success:

`curl http://shellyxxx.local/rpc/Shelly.SetConfig -d "{"id": 1, "type": 0, "wifi_ip": 192.168.1.2}`
and 
`curl http://shellyxxx.local/rpc/Shelly.SetState -d "{"id": 1, "type": 0, "wifi_ip": 192.168.1.2}`

Is there a way to set config parameters once flashed or should I revert to stock, change ip and flash it back?

Thanks!

grizz479, 2021-01-18:
SOLVED!

curl http://192.168.1.198/rpc/Config.Set -d '{"config": {"wifi": {"sta": {"ip": "192.168.1.201"}}}, "reboot": true}'

Thanks and congratulations on your great job!

AnakinSpain, 2021-02-07:
There is another way to do that??

timoschilling, 2021-02-07:
@andyblac maybe we can add an `--ip/-i` flack to the python tool, like we do it with `--hap-code`?

andyblac, 2021-02-07:
> @andyblac maybe we can add an `--ip/-i` flack to the python tool, like we do it with `--hap-code`?

can you provide me a a fill output of `http://{host}/rpc/Shelly.GetInfo` ?

as when i only set sta ip i am getting

```
[Device Scan] manual
Valid Hostname: shelly1l-ADF044.local True
Hostname: shelly1l-ADF044.local is Online
Device URL: http://192.168.3.62/rpc/Shelly.GetInfo
Configuring static IP to 191.168.3.69...
requests.post(url='http://192.168.3.62/rpc/Config.Set', json={'config': {'wifi': {'sta': {'ip': '191.168.3.69'}}}}
{"code":-1,"message":"error saving config: Station static IP is set but no netmask provided"}
```

andyblac, 2021-02-07:
think it needs

`ip
gw
mask
dns
`

timoschilling, 2021-02-07:
Maybe you simply provide empty values like the UI does

andyblac, 2021-02-07:
> Maybe you simply provide empty values like the UI does

can you show me config please from static ? do you use ?, just my confg no sta section at all.

andyblac, 2021-02-07:
this is from stock
```
        "wifi_sta": {
            "enabled": true,
            "ssid": "ABFi-IoT-2.4",
            "ipv4_method": "dhcp",
            "ip": null,
            "gw": null,
            "mask": null,
            "dns": null
        },
```

andyblac, 2021-02-07:
and with stock on static
```
        "wifi_sta": {
            "enabled": true,
            "ssid": "ABFi-IoT-2.4",
            "ipv4_method": "static",
            "ip": "192.168.3.53",
            "gw": "192.168.3.126",
            "mask": "255.255.255.128",
            "dns": "192.168.3.29"
        },
```

andyblac, 2021-02-07:
but would like to see info on this fw with static ip working.

timoschilling, 2021-02-07:
Sorry I’m on the mobile device at the moment. Try to add ‘“gw”: null’ and so on to the request. 
But maybe only setting the IP didn’t work on the network layer without the gw & co

grizz479, 2021-02-07:
Mine with fixed ip and flashed shelly1:

`   "sta": {
      "enable": true,
      "ssid": "DOPAVIIs",
      "pass": "whateverpassword",
      "user": "",
      "anon_identity": "",
      "cert": "",
      "key": "",
      "ca_cert": "",
      "ip": "192.168.1.189",
      "netmask": "255.255.255.0",
      "gw": "192.168.1.1",
      "nameserver": "",
      "dhcp_hostname": "" `

timoschilling, 2021-02-07:
@grizz479 use 3 back ticks to fix the formatting

andyblac, 2021-02-07:
> @grizz479 use 3 back ticks to fix the formatting

fixed it for him :)

andyblac, 2021-02-07:
> Mine with fixed ip and flashed shelly1:
> 
> ```
>       "enable": true,
>       "ssid": "DOPAVIIs",
>       "pass": "whateverpassword",
>       "user": "",
>       "anon_identity": "",
>       "cert": "",
>       "key": "",
>       "ca_cert": "",
>       "ip": "192.168.1.189",
>       "netmask": "255.255.255.0",
>       "gw": "192.168.1.1",
>       "nameserver": "",
>       "dhcp_hostname": "" 
> ```

thx

andyblac, 2021-02-07:
@grizz479 i take it if ip is ""  ie empty it reverts to dhcp ?

timoschilling, 2021-02-07:
@andyblac maybe this helps https://mongoose-os.com/docs/mongoose-os/api/net/wifi.md

grizz479, 2021-02-07:
> @grizz479 i take it if ip is "" ie empty it reverts to dhcp ? or does it need `dhcp_hostname` ip

Setting “ip”: “” enables dchp client mode.
```
   "sta": {
      "enable": true,
      "ssid": "DOPAVIIs",
      "pass": "lambretta",
      "user": "",
      "anon_identity": "",
      "cert": "",
      "key": "",
      "ca_cert": "",
      "ip": "",
      "netmask": "255.255.255.0",
      "gw": "192.168.1.1",
      "nameserver": "",
      "dhcp_hostname": ""
    },
    ```

andyblac, 2021-02-07:
ye, got it all working, just need to workout how force a reboot

{"reboot": true} 

is not working

andyblac, 2021-02-07:
cmd will be `./flash-shelly.py --ip-type=static --ip=192.168.3.69 --mask=255.255.255.128 --gw=192.168.3.126 --dns=192.168.3.1 shelly1l-ADF044.local`

andyblac, 2021-02-07:
doing `./flash-shelly.py --ip-type=dhcp shelly1l-ADF044.local` removes static ip

andyblac, 2021-02-07:
@rojer how can i get the device to reboot via rpc ?

timoschilling, 2021-02-07:
@andyblac take a look on the UI it does a RPC reboot

timoschilling, 2021-02-07:
Does that new settings are working doting the stock -> hk update?

andyblac, 2021-02-07:
only found this https://github.com/mongoose-os-apps/shelly-homekit/blob/a3d9f7cf819ffdc213119d79d85d1a7d4a5eeb18/fs_src/index.html#L183

andyblac, 2021-02-07:
> Does that new settings are working doting the stock -> hk update?

atm it does not require a flash to use, but i can add it after hap_setup, so it can be done all in one

andyblac, 2021-02-07:
here is what i have for function
```
def write_static_ip(wifi_ip, network_type, ipv4_ip='', ipv4_mask='', ipv4_gw='', ipv4_dns=''):
  if network_type == 'static':
    logger.info(f"Configuring static IP to {ipv4_ip}...")
    value={'config': {'wifi': {'sta': {'netmask': ipv4_mask}}}}
    logger.debug(f"requests.post(url='http://{wifi_ip}/rpc/Config.Set', json={value}")
    response = requests.post(url=f'http://{wifi_ip}/rpc/Config.Set', json=value)
    logger.trace(response.text)
    value={'config': {'wifi': {'sta': {'nameserver': ipv4_dns}}}}
    logger.debug(f"requests.post(url='http://{wifi_ip}/rpc/Config.Set', json={value}")
    response = requests.post(url=f'http://{wifi_ip}/rpc/Config.Set', json=value)
    logger.trace(response.text)
    value={'config': {'wifi': {'sta': {'gw': ipv4_gw}}}}
    logger.debug(f"requests.post(url='http://{wifi_ip}/rpc/Config.Set', json={value}")
    response = requests.post(url=f'http://{wifi_ip}/rpc/Config.Set', json=value)
    logger.trace(response.text)
    value={'config': {'wifi': {'sta': {'ip': ipv4_ip}}}}
    logger.debug(f"requests.post(url='http://{wifi_ip}/rpc/Config.Set', json={value}")
    response = requests.post(url=f'http://{wifi_ip}/rpc/Config.Set', json=value)
    logger.trace(response.text)
  else:
    value={'config': {'wifi': {'sta': {'ip': ''}}}}
    logger.debug(f"requests.post(url='http://{wifi_ip}/rpc/Config.Set', json={value}")
    response = requests.post(url=f'http://{wifi_ip}/rpc/Config.Set', json=value)
    logger.trace(response.text)
  if response.text.find('"saved": true') > 0:
    logger.info(f"Saved, Rebooting...")
    value={'config': {'reboot': 'true'}}
    logger.debug(f"requests.post(url='http://{wifi_ip}/rpc/Config.Set', json={value}")
    response = requests.post(url=f'http://{wifi_ip}/rpc/Config.Set', json=value)
    logger.trace(response.text)
```

andyblac, 2021-02-07:
```
ipv4_ip: 192.168.3.69
ipv4_mask: 255.255.255.128
ipv4_gw: 192.168.3.126
ipv4_dns: 192.168.3.1
Do you wish to set your IP address to 192.168.3.69 (y/n) ? y
Configuring static IP to 192.168.3.69...
requests.post(url='http://192.168.3.69/rpc/Config.Set', json={'config': {'wifi': {'sta': {'netmask': '255.255.255.128'}}}}
{"saved": true}

requests.post(url='http://192.168.3.69/rpc/Config.Set', json={'config': {'wifi': {'sta': {'nameserver': '192.168.3.1'}}}}
{"saved": true}

requests.post(url='http://192.168.3.69/rpc/Config.Set', json={'config': {'wifi': {'sta': {'gw': '192.168.3.126'}}}}
{"saved": true}

requests.post(url='http://192.168.3.69/rpc/Config.Set', json={'config': {'wifi': {'sta': {'ip': '192.168.3.69'}}}}
{"saved": true}

Saved, Rebooting...
requests.post(url='http://192.168.3.69/rpc/Config.Set', json={'config': {'reboot': 'true'}}
{"saved": true}
```

andyblac, 2021-02-07:
> Does that new settings are working doting the stock -> hk update?

as it can't be used with scan, i think it better to leave as seperate from flashing.

timoschilling, 2021-02-07:
https://github.com/mongoose-os-apps/shelly-homekit/blob/c8defd1f3178543eb60fc64d0c567c0e5b9ebb22/fs_src/script.js#L340

andyblac, 2021-02-07:
from static to dhcp 😄 
```
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py -l shelly1l-ADF044.local

Host: http://shelly1l-ADF044.local
Device Name: shelly1l-ADF044
Device ID: shelly1l-ADF044
IP: 192.168.3.69
Model: Shelly1L
Sys Temp: 40˚c
Current: HomeKit 2.7.3
Latest: HomeKit 2.7.3
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py --ip-type=dhcp shelly1l-ADF044.local
Saved, Rebooting...
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py -l shelly1l-ADF044.local

Host: http://shelly1l-ADF044.local
Device Name: shelly1l-ADF044
Device ID: shelly1l-ADF044
IP: 192.168.3.62
Model: Shelly1L
Sys Temp: 40˚c
Current: HomeKit 2.7.3
Latest: HomeKit 2.7.3
```

andyblac, 2021-02-07:
back to static 
```
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py --ip-type=static --ip=192.168.3.69 --mask=255.255.255.128 --gw=192.168.3.126 --dns=192.168.3.1 shelly1l-ADF044.local
Do you wish to set your IP address to 192.168.3.69 (y/n) ? y
Configuring static IP to 192.168.3.69...
Saved, Rebooting...
andrews-mbp:tools andrewblackburn$ ./flash-shelly.py -l shelly1l-ADF044.local

Host: http://shelly1l-ADF044.local
Device Name: shelly1l-ADF044
Device ID: shelly1l-ADF044
IP: 192.168.3.69
Model: Shelly1L
Sys Temp: 40˚c
Current: HomeKit 2.7.3
Latest: HomeKit 2.7.3
```

andyblac, 2021-02-07:
@rojer you with those commands flags ?

timoschilling, 2021-02-07:
> as it can't be used with scan, i think it better to leave as seperate from flashing.

True and it only works with one device at a time. Maybe only if we flash one device?

andyblac, 2021-02-07:
> > as it can't be used with scan, i think it better to leave as seperate from flashing.
> 
> True and it only works with one device at a time. Maybe only if we flash one device?

atm it coded for hk fw, but i need to make sure it will work with stock too imo.

andyblac, 2021-02-07:
here is beta if you want try it. [flash-shelly.py 2.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5940151/flash-shelly.py.2.zip)

andyblac, 2021-02-07:
@timoschilling ok i have tweaked code so it now can set either after a flash or when just trying to set the ip, only 1 host is allowed

i still need to test stock fw, do that tomorrow

[flash-shelly.py 2.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5940422/flash-shelly.py.2.zip)

andyblac, 2021-02-07:
command lines are
`./flash-shelly.py --ip-type=static --ip=192.168.3.69 --mask=255.255.255.128 --gw=192.168.3.126 --dns=192.168.3.1 shelly1l-ADF044.local`

`./flash-shelly.py --ip-type=dhcp shelly1l-ADF044.local`

andyblac, 2021-02-07:
noticed a slight bug last minute 

[flash-shelly.py 3.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5940444/flash-shelly.py.3.zip)

grizz479, 2021-02-08:
Great job @andyblac!
You figured it out on your own, but ticks in 'true' were the problem with rebooting.

Testing it right now in several devices.

andyblac, 2021-02-08:
here is the latest version [flash-shelly.py 4.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5943892/flash-shelly.py.4.zip)

handles both stock and homekit fw type.

grizz479, 2021-02-08:
Works perfect in flashed 2.7.3.

But "list all" returns a problem:
````
./flash-shelly.py -la             
Scanning for Shelly devices...
Traceback (most recent call last):
  File "./flash-shelly.py", line 835, in <module>
    device_scan(args.hosts, action, args.dry_run, args.quiet_run, args.silent_run, args.mode, args.type, args.exclude, args.version, args.variant, args.hap_setup_code, args.network_type, args.ipv4_ip, args.ipv4_mask, args.ipv4_gw, args.ipv4_dns)
  File "./flash-shelly.py", line 699, in device_scan
    probe_device(device, action, dry_run, quiet_run, silent_run, mode, exclude, version, variant, hap_setup_code)
TypeError: probe_device() missing 5 required positional arguments: 'network_type', 'ipv4_ip', 'ipv4_mask', 'ipv4_gw', and 'ipv4_dns'

andyblac, 2021-02-08:
./flash-shelly.py --ip-type=static --ip=192.168.3.69 --mask=255.255.255.128 --gw=192.168.3.126 --dns=192.168.3.1 _hostname_`

`./flash-shelly.py --ip-type=dhcp _hostname_`

andyblac, 2021-02-08:
> Works perfect in flashed 2.7.3.
>
> But "list all" returns a problem:
>
> ```
> ./flash-shelly.py -la
> Scanning for Shelly devices...
> Traceback (most recent call last):
>   File [...]

try with ver4 zip

grizz479, 2021-02-08:
Yup, not happening with v4

andyblac, 2021-02-08:
> Yup, not happening with v4

cool, please report any issues, you should beable to use it when flahing too 😄 so it could flash from stock to hk, then setup ip, all in 1 commad line

andyblac, 2021-02-08:
this has been merged, i'll close again.

closed 2021-02-08 by andyblac