mongoose-os-apps/shelly-homekit #422: max number of connections

christi-g, 2021-01-21:
Hello, currently the maximum number of connections is limited to 9. 
Is it possible that more connections are possible? I have already reached the maximum number...

[image]

Thanks

andyblac, 2021-01-22:
i'm currently testing a version tht better handles the connections, dropping ones that are no used for a set period, so far seem to be working better. I'll keep you updated.

andyblac, 2021-01-24:
new upcoming version, they have been increased to 12 max connections, and better still old ilde connections get dropped, as they become stale.

rojer, 2021-01-24:
released in 2.7.3

closed 2021-01-24 by rojer

simonfield76, 2021-01-24:
Andy sorry for the question but could you give us more details about the old connections? After how long does the device will become stale?

During summer I usually return home every 7/9 days i don’t want arriving home and nothing works with my phone because I’ve been out.

In next release could you put an option to automatically cancel old connection if not used?

Inviato da iPhone

rojer, 2021-01-24:
@simonfield76 Andy was not quite correct in his description of the behavior.
in fact, we do not actively drop connections if there is no slot shortage, connections will stay open indefinitely.
but if we are at limit (used to be 9, now 12) and a new connection comes in, we will find the connection that has been idle the most time (time since last request) and evict it to make room for the new one.

timoschilling, 2021-01-24:
And dropping the connection even doesn’t mean unpairing. It only means that your 13. device isn’t connected and only controls the device indirectly through an other iOS device

rojer, 2021-01-24:
correct. this is about dropping a TCP connection, not pairing associaction.

andyblac, 2021-01-24:
Sorry for the confusion.

simonfield76, 2021-01-24:
Thank you very much for the clarification!!!

Your work is very much appreciated!!!

Inviato da iPhone