mongoose-os-apps/shelly-homekit #423: Flashing Shelly 1 (v3) to MOS/HomeKit = "disconnected" and no data in WebUI

knanyahu, 2021-01-21:
Noob here.  Trying to test MOS/HomeKit firmware on Shelly 1 for a project.  Got first unit, setup tested stock firmware (updated to latest version) everything works fine.  Shelly Cloud and app works fine, switch responds.   Followed the instructions for the 2.7.2 flash.  (Tried both OTA and using Andy's instructions on Glitter to use the py script.)  It appears the device flashes ok but it goes very quick and I dont see a confirmation that it was successful.  Browsing the WebUI shows flash was good but has graphic that says "Disconnected" and there is no data displayed.  See screen shot.  Log file also attached. 

Thank you for your help!
Best, Brian

[image]

[debug2.log](https://github.com/mongoose-os-apps/shelly-homekit/files/5850931/debug2.log)

andyblac, 2021-01-22:
@knanyahu has dicovered that it is an issue with Safari and BigSur, @everyone is anyone else having the same issue ?

pivale, 2021-01-22:
Still on Catalina, but will upgrade next week. I'll let you know it it still happens...

andyblac, 2021-03-03:
can this be closed ?

closed 2021-03-11 by github-actions[bot]