mongoose-os-apps/shelly-homekit #426: [Feature request] Autoupdate

LukasM937, 2021-01-22:
Since every release blocks the function.
It would be nice to either let it work after release,
or let it automatically check ever n-th day if there is some update.
For privacy concerned it could be toggled.

andyblac, 2021-01-22:
> Since every release blocks the function.
> It would be nice to either let it work after release,
> or let it automatically check ever n-th day if there is some update.
> For privacy concerned it could be toggled.

or just use the script in a cron job, runing on a daily basis ???

andyblac, 2021-01-22:
getting the device to update itself is impossble, it requires a client to domthe update.

closed 2021-01-22 by andyblac

LukasM937, 2021-01-22:
Nice Idea, sorry that I didn't think about it.
Also thanks for the quick reply

rojer, 2021-01-22:
first, there are technical reasons why we cannot do device-driven updates - free version of mongoose os does not allow that.

and secondly, it's risky to just auto-update everything in the background. any kind of bug would take out entire fleet...
when user is aware of it, it provides a kind of safety mechanism - if devices start to malfunction, user will at least stop updating.

andyblac, 2021-01-22:
> first, there are technical reasons why we cannot do device-driven updates - free version of mongoose os does not allow that.
>
> and secondly, it's risky to just auto-update everything in the [...]

i normally run a single device on a update for 24 hours, then if all is well, i run my script to update the rest.

LukasM937, 2021-01-22:
Again thank you for the advice on good practice!