mongoose-os-apps/shelly-homekit #428: Random reboot when switching lights

MichaelBischof, 2021-01-23:
Hi there,
I have a shelly 1 that reboots randomly when switching lights. It not only stops responding, the physical light switch also does not work anymore to control the light attached to the shelly.

It takes about 20s-30s for it to recover and work again.

Any idea what is going wrong here?

The logs show the following:
```Shelly.SetState via WS_in 192.168.2.101:61680
35852089 shelly_output.cpp:56    Output 1: off -> on (RPC)
35881228 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.101:61680
36167680 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.101:61680
37032530 mg_rpc.c:305            Shelly.SetState via WS_in 192.168.2.101:61680
37042114 shelly_output.cpp:56    Output 1: on -> off (RPC)
37070619 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.101:61680
37168384 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.101:61680
38171155 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.2.101:61680
38197868 mg_rpc.c:305            Shelly.SetState via WS_in 192.168.2.101:61680
100004 % c`E``a_c```dd,C`p:00    dd`1` 0  hdd )0 $(  @ �c
578334 mgos_rpc_channel_ua:313 0x3fff457c UART0
598506 shelly_main.cpp:304     === Creating accessories
400002 mgos_rpc_channel_mq:203 0x3fff3b84 shelly1-48076E/rpc
586276 mgos_rpc_channel_ua:313 0x3fff457c UART0
606436 shelly_main.cpp:304     === Creating accessories
617979 shelly_switch.cpp:191   Exporting 'Shelly Flur': type 0, state: 0
628817 shelly_main.cpp:331     === Starting HAP server (CN 1)
697572 HAPPlatformTCPStrea:185 Listening on 9000
711449 mgos_init.c:33          Init done, RAM: 40408 total, 28216 free, 28224 min free
723484 shelly_main.cpp:365     HAP server state: 1

mrlitchyb, 2021-01-23:
Reverted to stock firmware and the problem persists... probably nothing wrong with you code 😒 
No clue what is wrong though...

mrlitchyb, 2021-01-23:
Oh my god... found the issue. It was the bulb I attached to the Output. As soon as I removed the output everything worked like a charm. It seems like the shelly hardware has some kind of problem with the type of bulb / consumer attached. If I attach a different device to my output it just works.

closed 2021-01-23 by MichaelBischof