mongoose-os-apps/shelly-homekit #429: I can't calibrate

Setsuko34, 2021-01-23:
I can't calibrate my shelly2.5 on roller shutter mode to use it for my windows when I start the calibrating it only goes up and never stop so it forced on the motor.

can someone help me ?

andyblac, 2021-01-27:
@pivale can you help on this ?

pivale, 2021-01-28:
Hey @Setsuko34 ,

Calibration on both stock and Homekit firmware are bound the ability of the motor to stop at the end of movement (0w power consumption).
This will tell firmware for how long it needs to run and access percentages of open/closed.

If you say that if not stoping "forces" on the motor, it seems that you shutter motor doesn't stop on it's own.
This will render calibration (and roller shutter mode) useless and risky for you hardware.

Do you have any instruction set from your shutter model that can confirm this?

andyblac, 2021-03-03:
I'll close this one, maybe take a look here https://github.com/mongoose-os-apps/shelly-homekit/issues/404

closed 2021-03-03 by andyblac