mongoose-os-apps/shelly-homekit #430: Shelly 1 with door sensor

FXGPV, 2021-01-23:
Hello, as there is already the garage door with magnetic sensor for informing status, it is possible to add a second option for dor/windows with magnetic sensor? this of course would only repost the current state and not any action per self. Thanks

andyblac, 2021-01-24:
you can already do this in 2.7.2, 

1st chnage your shelly1 to INPUT MODE to "Detached", and then in the new setting that should appear "INPUT1" chnage HAP TYPE to "Contact Sensor"

[image]

it should give you this in HomeKit
[image]

FXGPV, 2021-01-24:
perfect... thank you very much

andyblac, 2021-01-24:
can i close this issue ?

closed 2021-01-25 by andyblac