mongoose-os-apps/shelly-homekit #432: homekit location based automation do not work

ari-ido-999, 2021-01-25:
Hello,

I tried a few scenarios where I leave/arrive home and none of them seem to to turn on/off the shelly 1 switches I have.  I have a homepod mini and all the locations settings seem correct. Are these supposed to work?

andyblac, 2021-01-25:
works here np, anyway this is HomeKit issue not of thise Firmware.

ari-ido-999, 2021-01-25:
Thanks, any idea to figure out where is the problem?

andyblac, 2021-01-25:
do you only have a single hub ?

andyblac, 2021-01-25:
do you location enabled in HomeKit in yours system settings ? (settings icon in main iOS)

ari-ido-999, 2021-01-25:
> do you location enabled in HomeKit in your system settings ? (settings icon in main iOS)

yes of course , all the settings that appear here are configured correctly https://techtechandmoretech.com/guides/how-to-fix-homekit-location-automation/

andyblac, 2021-01-25:
no AppleTV's if you do, disable HomeKit on them, they can cause issue with automations and HomeKit.

ari-ido-999, 2021-01-25:
I use homepod mini

andyblac, 2021-01-25:
it should work then, can you post some screenshots of your workflow ?,

ari-ido-999, 2021-01-25:
This automation for example is supposed to turn off the shelly 1 but it doesn't. Is it supposed to be immediate or can take time. I waited for about 10 minutes 100 meters from my home.

Whiterabbit64, 2021-01-25:
I think 100m is not enough. Location works with a radius of 300 or 400m. You can’t set less than that.

andyblac, 2021-01-25:
> 
> I think 100m is not enough. Location works with a radius of 300 or 400m. You can’t set less than that.

yeah IIRC the HomeKit specs are a Minimum of 200m is required, but the more distance assures better activation of automation iirc @Whiterabbit64 is correct 300m+ is recomended.

simonfield76, 2021-01-25:
I have disabled the function away/back to home to my gate for the same reason. Whenever I go out with my dog for his walk it take action too far on my way back and returning from job the distance is too short to keep the door open.  Moreover due to the fact that it is configured as garage door it needs an extra confirmation by logging into the phone.

Same happened with my old gate Bluetooth opener Solo.

Inviato da iPhone

andyblac, 2021-01-25:
so, can i close issue ?

ari-ido-999, 2021-01-26:
Yes, it works with my son's phone so it is something else not your software

closed 2021-01-26 by ari-ido-999