mongoose-os-apps/shelly-homekit #434: Issues and Labels

andyblac, 2021-01-25:
Guy's when creating an issue, please tag the device with the correct Device Label, and if anyother labels seem approiate please tagged them to it.

andyblac, 2021-01-25:
you can also now click on a label to see ALL issues in that label, please check no issue has been submitted before creating a new issue
thanks

rojer, 2021-01-26:
nice! thanks Andy!

andyblac, 2021-01-27:
@everyone please make sure when creating issues, you 1st check no existing exists, by clicking on the label associated with your issue, they maybe be an exisitng issue that could help you out.

for example there are loads of issues on Roller Shutters, as i dont have one or have used the setup i can't help, but it seems to me that the vast majority of them are all about the same issue "Calibration".

can people with Roller Shutters try to help people out in those issues.

oh, and please try to remeber to assing "Labels" when creating issues, 
Thanks.

closed 2021-02-08 by andyblac