mongoose-os-apps/shelly-homekit #435: webUI: relay button control.

andyblac, 2021-01-25:
@ZeevoX  suggestion for webUI for relays On / Off indicator, could you remove the on / off text and align the button up, and turn button GREEN button with ON text when ON, and BLUE button with OFF text, instead on ‘Turn On’ / ‘Turn Off’ ??, I think it would be more intuitive that way
 
This bit
https://files.gitter.im/5f786ddbd73408ce4ff09679/Fsro/Screenshot-2021-01-20-at-20.35.45.jpg

Or even BLUE ON / GREY OFF buttons

andyblac, 2021-01-26:
@timoschilling  maybe you take this one too ??

timoschilling, 2021-01-26:
Maybe, if I find some time in the future, I currently working on rgbw2 support.

I think gray isn't a good color for a button, it looks like disabled, maybe green / red.

andyblac, 2021-02-11:
how does this look ?

[image]

ZeevoX, 2021-02-11:
Aesthetically looks good for sure, however, not sure this is the clearest in terms of user interface. The status is not "Turn off", and because of the green colour it adds even further confusion as to whether it is currently "turned off" or whether pressing the button should turn it off.
Since this is a binary selection between on/off, could we have something similar to the toggle switches? (also in the screenshot you posted) – we could make them wider with on/off text in the middle to make it super clear.

andyblac, 2021-02-11:
I'll have a go 😄

rojer, 2021-02-11:
agree with Timothy here - green to turn off is confusing. and replacing with toggle switch is a good idea as well, i like it.

andyblac, 2021-02-12:
[image]

andyblac, 2021-02-12:
and with power meters
[image]

andyblac, 2021-02-12:
or
[image]

pivale, 2021-02-12:
I like these last two concepts most.
It's consistent with apple, homekit and with the rest of the UI that already had those toggles.

andyblac, 2021-02-12:
ye, me too now, atm i have an issue where the status is not auto updating, if i change the power state in HomeKit, it is not auto refresh, works if i manually refresh page.

still learning html5 and JS. 😄

andyblac, 2021-02-12:
PR submitted https://github.com/mongoose-os-apps/shelly-homekit/pull/486

PR #486 (2021-02-12): [webUI] change relay state control to a slider.
fixes #435
closed 2021-02-12 by rojer