mongoose-os-apps/shelly-homekit #436: Garage door: “open” position sensor possible?

stevenringo, 2021-01-26:
Hi,

Thank you for sharing this awesome firmware. You all have made it so easy to configure. Getting a stock Shelly 1 to work **properly** in HomeKit (with all 4 states) using HomeBridge or Home Assistant seems really tricky, and I am so glad this exists.

I installed this to my garage today using a **Shelly 1**, and it works brilliantly. The one thing I noticed is that the firmware assumes the reed switch is located at the “closed” position. This was fine for the garage as I could place the reed switch in a location such that it was able to make contact in the closed position.

I am now attempting to do this on my sliding gate, however having the reed switch detect the “closed” state is not really possible as it would mean needing to run a wire across the opening. I would rather have the reed switch detect “open”, and if the switch is not aligned it assumes “closed”.

I could “hack” it by reversing the “close sensor” setting from NC to NO (mine is NC), and it seems to work somewhat, however the opening/closing states are then inverted, which is not really desirable.

In other words, is it possible to have the “close sensor” be an “open sensor” instead?

Thanks so much.

Steven

rojer, 2021-01-26:
it's a bad idea but i guess i have no choice, as it just keeps coming up... :)

stevenringo, 2021-01-26:
Thank you for considering it. Why do you think it's a bad idea?

rojer, 2021-01-26:
when we have only one sensor, we only know one door position definitively and the other we just assume.
when the only sensor is a "fully closed" sensor, then when the app says "door closed" you know it is - the sensor says so.
however, when the sensor is "fully open", we know the door is not open, but when we say "closed", it may mean anywhere from "fully closed" to "just barely". it's a less secure configuration, if you close the door without looking at it,it may get suck in the middle and you wouldn't know because the app will happily report "closed".

stevenringo, 2021-01-26:
Makes sense, thank you. I think I need to find a way to have it sense "closed" (i.e. the way its currently supported).

andyblac, 2021-03-03:
i'll close this one.

closed 2021-03-03 by andyblac