mongoose-os-apps/shelly-homekit #437: Unable to Update a single Shelly 1

freddair, 2021-01-26:
6 other Shellies (1/1L/1PM) updated just fine.

Log:

315453175 mgos_ota_core.c:587     begin: waiting for user
315469786 mgos_ota_core.c:587     begin: waiting for user
315481456 mgos_ota_core.c:587     begin: waiting for user
315492687 mgos_ota_core.c:587     begin: waiting for user
315502976 mgos_ota_core.c:587     begin: waiting for user
315514114 mgos_ota_core.c:587     begin: waiting for user
315524383 mgos_ota_core.c:587     begin: waiting for user
315534570 mgos_ota_core.c:587     begin: waiting for user
315544756 mgos_ota_core.c:587     begin: waiting for user
315555979 mgos_ota_core.c:587     begin: waiting for user
315569251 mgos_ota_core.c:587     begin: waiting for user
315619663 mgos_ota_core.c:587     begin: waiting for user
315630204 mgos_ota_core.c:587     begin: waiting for user
315642388 mgos_ota_core.c:587     begin: waiting for user
315652660 mgos_ota_core.c:587     begin: waiting for user
315662849 mgos_ota_core.c:587     begin: waiting for user
315673581 mgos_ota_core.c:587     begin: waiting for user
315686509 mgos_ota_core.c:587     begin: waiting for user
315697772 mgos_ota_core.c:587     begin: waiting for user
315708279 mgos_ota_core.c:587     begin: waiting for user
315718474 mgos_ota_core.c:587     begin: waiting for user
315731892 mgos_ota_core.c:587     begin: waiting for user
315743065 mgos_ota_core.c:587     begin: waiting for user
315789163 mgos_ota_core.c:587     begin: waiting for user
315799470 mgos_ota_core.c:587     begin: waiting for user
315811082 mgos_ota_core.c:587     begin: waiting for user
315825272 mgos_ota_core.c:587     begin: waiting for user
315835495 mgos_ota_core.c:587     begin: waiting for user
315845701 mgos_ota_core.c:587     begin: waiting for user
315856261 mgos_ota_core.c:587     begin: waiting for user
315866476 mgos_ota_core.c:587     begin: waiting for user
315877988 mgos_ota_core.c:587     begin: waiting for user
315888414 mgos_ota_core.c:587     begin: waiting for user
315898970 mgos_ota_core.c:587     begin: waiting for user
315909166 mgos_ota_core.c:587     begin: waiting for user
315920393 mgos_ota_core.c:587     begin: waiting for user
315966090 mgos_ota_core.c:587     begin: waiting for user

[image][image]

andyblac, 2021-01-26:
i'll leave this one for @rojer

rojer, 2021-01-26:
device won't accept update if it's in "doing something" mode. that includes auto-off timer. do you use auto-off and is it active at this time?

freddair, 2021-01-26:
Yes this Shelly uses auto-off timer and obviously it was the issue. Thanks for pointing that out. I wasn't aware that the timer was still active even after reboots... maybe it would be helpful if we would get a corresponding warning / error message when trying to start an update. Thank you @rojer !

timoschilling, 2021-01-26:
Maybe we should disable the update button with a note if auto-off is on.

freddair, 2021-01-26:
Also it would be helpful to expose the countdown timer visible in the switch auto-off section

andyblac, 2021-01-26:
> Maybe we should disable the update button with a note if auto-off is on.

imo, no, only if auto-off is 'running', just having the option enabled does not harm, most of my devcies have it enabled, and i have never run into an issue updating.

timoschilling, 2021-01-26:
sorry if it's running, updated my post

andyblac, 2021-01-26:
@freddair would you like to create a issue for this feature request ?, i will then close this issue.

freddair, 2021-01-27:
https://github.com/mongoose-os-apps/shelly-homekit/issues/444

Done, this may be closed.

closed 2021-01-27 by andyblac