mongoose-os-apps/shelly-homekit #438: Request: allow user to determin LONG PRESS duration

andyblac, 2021-01-26:
can we add support to control how long the button has to be held down to action as a LONG PRESS, when in detached mode.

Some familly members and guests are holding the button to long and it is being detected as a LONG PRESS, but they just wanted a SHORT PRESS, it would be good if we can allow control of this in the webUI.

@ZeevoX is this something you could take a look at ?

rojer, 2021-01-26:
makes sense

andyblac, 2021-01-26:
yeah maybe @timoschilling or @ZeevoX could take a look.

timoschilling, 2021-01-26:
@rojer could you point out the place where the value is set/used?

andyblac, 2021-01-26:
i found these

https://github.com/mongoose-os-apps/shelly-homekit/blob/eb9dd677e677b9180a9f37eecf7b523cceeed81c/src/shelly_input_pin.cpp#L31
https://github.com/mongoose-os-apps/shelly-homekit/blob/47e0c4673637c3b7b5e5d6ec8022195de0a6c912/src/shelly_input_pin.hpp#L30

rojer, 2021-01-26:
default values are constants - set [here](https://github.com/mongoose-os-apps/shelly-homekit/blob/22a1d9a84f0083dd4edd8188459249ed8f71a0ff/src/shelly_input_pin.cpp#L30-L31)