mongoose-os-apps/shelly-homekit #444: Make running auto-off timer visible in UI

freddair, 2021-01-27:
Let's expose the countdown timer visible in the UI beneath each switch auto-off setting.

Let's also disable the update button with a note if auto-off is running (thanks @timoschilling ), because device won't accept update if it's in "doing something" mode, including auto-off timer (thanks @rojer ).

https://github.com/mongoose-os-apps/shelly-homekit/issues/437

rojer, 2021-01-27:
SGTM

rojer, 2021-01-27:
real solution for the auto-off timer is to persist its state and restore on reboot so it continues running, but for now disabling update will do.

andyblac, 2021-01-27:
@ZeevoX or @timoschilling could one of you guys take this on ?

andyblac, 2021-01-27:
and maybe this could be done at the same time ?  https://github.com/mongoose-os-apps/shelly-homekit/issues/209

timoschilling, 2021-01-27:
I can take a look on this issue, but #209 sound like a different feature.