mongoose-os-apps/shelly-homekit #445: Shelly 2.5 as roller shutter - calibration doen´t work propper

joha1971, 2021-01-27:
Hello, this is my 3rd shelly 2.5 which i want to use as roller shutter in homekit. The firts two are working very well with the SW version 2.7.1. The calibration ran well without issues. 
Now i wanted to install the 3rd shelly 2.5 which has issues in the calibration session. This shelly downloaded the SW version 2.7.3. This is my issue:
If i start the calibration, the shelly will NOT close the roller. It goes a few centimeter up and down, up and down. A couple of times. The time frame which will be identified is approx. 1 sec. I have the feeling that the calibration starts with the wrong relay output. First up and then down. But my feeling can be wrong.
My attention was to downgrade the SW to 2.7.1 because the other two are working well with this SW. But unfortunately this wasn´t successful. 
With the original SW it works. Now i run the roller shutter as 2 switcher in homekit.
Is it a software bug? I need help. Thank you
regards Jochen

joha1971, 2021-01-28:
Additional information: Here are my log. The calibration mode worls always for 1.08 sec. Then he is ready. That menas that the roller never shuts down completly.

132383880 shelly_output.cpp:56    Output 1: off -> on (move)
132393452 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
132420113 shelly_hap_window_c:544 P = 0.00 -> 0.00
132430119 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
132622250 shelly_hap_window_c:331 WC 1: Cur pos 7.31 -> 16.56, P = 0.00
132689663 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
133021441 shelly_output.cpp:56    Output 1: on -> off (moving)
133031458 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
133140532 mgos_sys_config.c:232   Loading conf2.json
133164370 mgos_sys_config.c:232   Loading conf3.json
133297926 mgos_sys_config.c:174   Saved to conf9.json
133309157 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
133327187 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
133425340 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
133524520 shelly_output.cpp:56    Output 2: off -> on (move)
133534613 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
133624732 shelly_hap_window_c:544 P = 0.00 -> 0.00
133634742 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
133676832 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
133725979 shelly_output.cpp:56    Output 2: on -> off (moving)
133735961 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
133845826 mgos_sys_config.c:232   Loading conf2.json
133869680 mgos_sys_config.c:232   Loading conf3.json
134004459 mgos_sys_config.c:174   Saved to conf9.json
134015708 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
134032953 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
134131014 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
134230269 shelly_output.cpp:56    Output 1: off -> on (move)
134240310 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
134330571 shelly_hap_window_c:544 P = 0.00 -> 0.00
134340582 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
134431633 shelly_output.cpp:56    Output 1: on -> off (moving)
134441659 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
134550886 mgos_sys_config.c:232   Loading conf2.json
134574706 mgos_sys_config.c:232   Loading conf3.json
134709491 mgos_sys_config.c:174   Saved to conf9.json
134720738 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
134740507 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
134761706 mgos_mongoose.c:66      New heap free LWM: 17264
134777287 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
134875259 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
134974482 shelly_output.cpp:56    Output 2: off -> on (move)
134984529 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
135074810 shelly_hap_window_c:544 P = 0.00 -> 0.00
135084842 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
135175915 shelly_output.cpp:56    Output 2: on -> off (moving)
135186895 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
135294928 mgos_sys_config.c:232   Loading conf2.json
135318774 mgos_sys_config.c:232   Loading conf3.json
135452655 mgos_sys_config.c:174   Saved to conf9.json
135463912 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
135481622 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
135579630 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
135678919 shelly_output.cpp:56    Output 1: off -> on (move)
135689965 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
135706014 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
135779219 shelly_hap_window_c:544 P = 0.00 -> 0.00
135789255 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
135881678 shelly_hap_window_c:331 WC 1: Cur pos 46.16 -> 53.47, P = 1.03
135890815 shelly_output.cpp:56    Output 1: on -> off (moving)
135900819 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
135998878 mgos_sys_config.c:232   Loading conf2.json
136022765 mgos_sys_config.c:232   Loading conf3.json
136156979 mgos_sys_config.c:174   Saved to conf9.json
136168253 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
136185967 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
136284055 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
136383275 shelly_output.cpp:56    Output 2: off -> on (move)
[... 117 lines cut ...]
144619922 shelly_main.cpp:543     Up 144.61, HAP 0/0/12 ns 0, RAM: 20480/37616; st 57; 4.1: c:1 mp:0.00 mt_ms:1081 cp:49.31 tp:49.31 lemd:0 lhmd:0
145676870 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
146692499 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
147695476 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
148672787 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
149696073 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
150675876 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.178.86:57677
151486761 mg_rpc.c:305            Shelly.SetConfig via WS_in 192.168.178.86:57677
151496617 shelly_rpc_service.:223 SetConfig ok, 0
151529561 mgos_sys_config.c:232   Loading conf2.json
[... cut ...]
158595219 mgos_http_server.c:180  0x3fff67ec HTTP connection from 192.168.178.86:57704
161745621 shelly_debug.cpp:203    Tailing
168618690 shelly_main.cpp:543     Up 168.61, HAP 0/0/12 ns 0, RAM: 21272/37616; st 57; 4.1: c:1 mp:0.00 mt_ms:1081 cp:49.31 tp:49.31 lemd:0 lhmd:0
176618709 shelly_main.cpp:543     Up 176.61, HAP 0/0/12 ns 0, RAM: 21272/37616; st 57; 4.1: c:1 mp:0.00 mt_ms:1081 cp:49.31 tp:49.31 lemd:0 lhmd:0
184618620 shelly_main.cpp:543     Up 184.61, HAP 0/0/12 ns 0, RAM: 21272/37616; st 57; 4.1: c:1 mp:0.00 mt_ms:1081 cp:49.31 tp:49.31 lemd:0 lhmd:0

joha1971, 2021-02-01:
Hi all,
is noboby able to support my issue? Now i have replaced my shelly 2.5 against a new one. But the issue is still the same. The curton down mode works only for 1,08 sec. and shelly thinks it´s closed, which is not. One additional information. I have two roller shuter motor connected to one switch. Could this be the problem? May be that the current flow is too high?
Need support. I am lost. Thank you.

andyblac, 2021-02-01:
have checked the other roller shutter isues, maybe they could help ?

joha1971, 2021-02-01:
Of course you are right. I saw that there is already an open issue with the same issue. I also saw that there is a request for an extention to use the roller shutter with typing in a calibration time manually. now i will wait until this issue is fixed and a new firmware will be released. Thank you

andyblac, 2021-02-01:
ok, i'll close this issue as duplicate. thx

closed 2021-02-01 by andyblac