mongoose-os-apps/shelly-homekit #448: Firmware 2.7.3 problem with Shelly1 "garage door opener" mode problem with close sensor

grizz479, 2021-01-28:
Hi there,
Since last update my -before totally functional- shelly1-garage door opener is giving false lectures from the close sensor. It warns  perfectly of the "open state" but after 5 seconds returns a "closed state" even though the door is still opening and the opening time is set to 20 secs. 
When closing it backs, it sends (and logs) the same inputs: First "open gate" followed closely by a "closed gate" while still closing.

I assume it is due to a problem in the new firmware.

Thanks

rojer, 2021-01-28:
actually, there haven't been much changes in the GDO recently. can you please enabled debug log and post it here while?

grizz479, 2021-01-29:
Hi Rojer,
After a couple reboots and repairing with home app everything seems to work fine now. I did not remember but when I installed I had some issues with the close sensor after first install. It appeared to me it took it several "runs" to take the input right, even though I suppose it is not even possible.

trolli-ch, 2021-01-30:
Thanks for reporting this. I have the same issue.
Before 2.7.3 the status was always correct, now its as described by grizz479.
So opening the door will show some second the status "opening" and then switch to ".closed"
So the door is open now, but the status is closed.

74298538656 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74299533423 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74299724188 shelly_output.cpp:56    Output 1: off -> on (GDO:open)
74299735703 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
74299747691 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
74299815890 shelly_output.cpp:56    Output 1: on -> off (pulse_off)
74300540563 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74301534445 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74302542520 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74303426997 shelly_main.cpp:543     Up 74303.42, HAP 0/8/12 ns 8, RAM: 18624/37912; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
74303539579 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74304538960 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74304823923 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
74304835960 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
74305535623 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74306534619 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74307533701 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74308537857 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74309533667 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.4.107:51802
74309738897 mgos_http_server.c:180  0x3fff72e4 HTTP connection from 192.168.4.107:51811
74309936621 shelly_debug.cpp:203    Tailing
74311426193 shelly_main.cpp:543     Up 74311.41, HAP 0/8/12 ns 8, RAM: 18696/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
74319426049 shelly_main.cpp:543     Up 74319.41, HAP 0/8/12 ns 8, RAM: 18696/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
74327426068 shelly_main.cpp:543     Up 74327.41, HAP 0/8/12 ns 8, RAM: 18696/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
74335426455 shelly_main.cpp:543     Up 74335.42, HAP 0/8/12 ns 8, RAM: 18696/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1

grizz479, 2021-02-02:
Had the same issue again. But it was after changing device ip. The first 5-6 times The gate is operated it takes wrong values from the sensor, but after that is working properly again.

mac-alx, 2021-02-06:
Hello,

I have a pretty similar problem. 
usecase: Shelly 1 as garage door opener with close sensor

When opening the garage door everything works fine. 
When closing the door the status is closing until the door is closed. Then the status changes to „closed“, back to „opening“ and again back to „closed“. 

The last changes are all in around 2 seconds. 

This happens when closing the door with the Shelly web access and when pressing the hardware button, not when closing the door via HomeKit. 

I already checked the close-sensor. It changes from open to close without any bouncing. 

This is not specific to 2.7.3

Do you have any idea how to solve that?

Many greetings
Marc

grizz479, 2021-02-16:
Still having this problem.
It worked fine for a few days but after a wifi issu troubleshooting it began guving wrong state again.

i just erased the accessy from homekit, rebout and linked again. Same problem. Here the log:

```
143260177 HAPPlatformTCPStrea:276 0x3fff6e34 192.168.1.11:54562 Accepted HAP connection, ns 0/3/12 ts 0x3fff594c
146855174 shelly_main.cpp:543     Up 146.84, HAP 0/3/12 ns 3, RAM: 21804/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
147746292 HAPPlatformTCPStrea:121 0x3fff695c 192.168.1.14:52368 Incoming HAP connection, ns 0/3/12
147760369 HAPPlatformTCPStrea:276 0x3fff695c 192.168.1.14:52368 Accepted HAP connection, ns 0/4/12 ts 0x3fff6ebc
152387080 shelly_output.cpp:56    Output 1: off -> on (GDO:open)
152398482 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
152409382 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
152697063 shelly_output.cpp:56    Output 1: on -> off (pulse_off)
155953690 shelly_main.cpp:543     Up 155.94, HAP 0/4/12 ns 4, RAM: 21336/37912; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
157448476 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
157460377 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
163953744 shelly_main.cpp:543     Up 163.94, HAP 0/4/12 ns 4, RAM: 21336/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
171953761 shelly_main.cpp:543     Up 171.94, HAP 0/4/12 ns 4, RAM: 21336/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
172320698 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.12:50252
173305336 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.12:50252
174303007 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.12:50252
174416745 mgos_http_server.c:180  0x3fff6c6c HTTP connection from 192.168.1.12:60117
174579813 shelly_debug.cpp:203    Tailing
179953274 shelly_main.cpp:543     Up 179.94, HAP 0/4/12 ns 4, RAM: 21420/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
187953214 shelly_main.cpp:543     Up 187.94, HAP 0/4/12 ns 4, RAM: 21420/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
195953275 shelly_main.cpp:543     Up 195.94, HAP 0/4/12 ns 4, RAM: 21420/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1

AG-dingoant, 2021-02-17:
Hi,
I have exactly the same issue but I couldn't find the variables. I open the door only from the HK applicaiton.

AG-dingoant, 2021-02-19:
I still have the problem. Here are the logs :

434856929142:1613726999875060 shelly_rpc_service.:223 SetConfig ok, 0
434856957123 mgos_sys_config.c:232   Loading conf2.json
434856980633 mgos_sys_config.c:232   Loading conf3.json
434856991220 mgos_config_util.c:110  Extra key: [relays]
434856999908 mgos_config_util.c:110  Extra key: [relays.r0]
434857010897 mgos_config_util.c:110  Extra key: [relays.r0.default_state]
434857020789 mgos_config_util.c:110  Extra key: [relays.r0.btn_reverse]
434857029543 mgos_config_util.c:110  Extra key: [relays.r0]
434857038948 mgos_config_util.c:110  Extra key: [relays]
434857164103 mgos_sys_config.c:174   Saved to conf9.json
434857195385 esp_main.c:218          Stack overflow! Last cb 0x4026d578, ptrs: 120:0x401019e4 56:0x401019e4 -736:0x4010194c -824:0x401019e4 -888:0x401019e4 -2776:0x40224c00 -3128:0x40224c00 -3536:0x40268bd9 -3728:0x4026ee20 -4048:0x40100a27 -4064:0x40100285
434858529487 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64574
434859227002 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64574
434860218948 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64574
434860309814 mgos_http_server.c:180  0x3fff6c24 HTTP connection from 192.168.0.45:64579
434860327099 mgos_http_server.c:180  0x3fff7194 HTTP connection from 192.168.0.45:64580
434860669413 shelly_debug.cpp:203    Tailing
434860820306 shelly_main.cpp:543     Up 434860.81, HAP 0/3/12 ns 3, RAM: 21580/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
434868820337 shelly_main.cpp:543     Up 434868.81, HAP 0/3/12 ns 3, RAM: 21580/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
434874095338 mgos_http_server.c:180  0x3fff6524 HTTP connection from 192.168.0.45:64595
434874125253 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434875219937 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434876218780 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434876819899 shelly_main.cpp:543     Up 434876.81, HAP 0/3/12 ns 3, RAM: 21500/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
434877219403 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434878232657 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434879231613 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434880218491 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434880946687 mg_rpc.c:305            Shelly.SetState via WS_in 192.168.0.45:64595
434880956886 shelly_output.cpp:56    Output 1: off -> on (GDO:open)
434880968947 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (RPC)
434880979939 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
434881023491 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434881042441 mgos_mongoose.c:66      New heap free LWM: 17016
434881226507 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434881267793 shelly_output.cpp:56    Output 1: on -> off (pulse_off)
434882225729 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434883236749 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434884219478 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434884819812 shelly_main.cpp:543     Up 434884.81, HAP 0/3/12 ns 3, RAM: 21500/37912; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
434885219414 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434886014839 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
434886026866 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
434886221224 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434887218405 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434888220715 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434889227465 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434890221445 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434891218248 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434892220120 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434892820055 shelly_main.cpp:543     Up 434892.81, HAP 0/3/12 ns 3, RAM: 21500/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
434893218984 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434894219283 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434895228462 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434895557688 mgos_http_server.c:180  0x3fff6e6c HTTP connection from 192.168.0.45:64598
434896679979 shelly_debug.cpp:203    Tailing
434900819642 shelly_main.cpp:543     Up 434900.81, HAP 0/3/12 ns 3, RAM: 20984/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
434905225333 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434906217112 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434907220510 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434908225857 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434908819419 shelly_main.cpp:543     Up 434908.81, HAP 0/3/12 ns 3, RAM: 20984/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
434909228840 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
434910221685 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64595
[... cut ...]
434996819756 shelly_main.cpp:543     Up 434996.81, HAP 0/3/12 ns 3, RAM: 21352/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
435004819844 shelly_main.cpp:543     Up 435004.81, HAP 0/3/12 ns 3, RAM: 21352/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
435012819758 shelly_main.cpp:543     Up 435012.81, HAP 0/3/12 ns 3, RAM: 21352/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
435020819371 shelly_main.cpp:543     Up 435020.81, HAP 0/3/12 ns 3, RAM: 21352/37912; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1

joddye2, 2021-02-21:
Same issue here.

andyblac, 2021-03-03:
@rojer any advice ?

rojer, 2021-03-06:
thanks for reporting. so what happens here is after 5 seconds of moving in the opening direction, closed sensor still remains active. we conclude that the door jammed or is otherwise refusing to move and reverse direction.

does it take the door more than 5 seconds to actually start moving and get sensor to deactivate?

rojer, 2021-03-06:
5 seconds is currently a constant, thinking being - "surely that ought to be enough for everybody". well, maybe not.

rojer, 2021-03-06:
i increased it to 7 seconds by default, please install 2.8.0-alpha4 and help test

#498

AG-dingoant, 2021-03-06:
Alpha installed. I will test this morning. Thanks for the update. When I use the normal press button of the door opener, the shelly detects correctly that the door is open and closed. The issue is only if the opening request comes from the Shelly

joddye2, 2021-03-06:
Tried 2.8.0 alpha but the problem still persist.
When closing the three notifications comes. Closed, open, closed.

rojer, 2021-03-06:
hm. ok, i'll need to see the logs again - when it does work properly (button press) and when it doesn't (web ui / HAP).

grizz479, 2021-03-06:
Installed 2.8.0 alpha.
Still working the same, with button press or web ui/HAP.
Even more, having trouble to reach de web ui.

andyblac, 2021-03-06:
> Installed 2.8.0 alpha.
> Still working the same, with button press or web ui/HAP.
> Even more, having trouble to reach de web ui.

yeah, seems alpha4 has connectivity issues, we're investigating.

andyblac, 2021-03-06:
@grizz479 if you can not connect to device webui at all try a hard reboot (cut power), aslso DO NOT try to reflash alpha3, if you do you will have to reset WiFi, as i found it doing so corupts wifi config.

afater a hard reboot, webui can be reached, but it can be hit and miss just keep trying.

AG-dingoant, 2021-03-06:
Same issue here as well.

joddye2, 2021-03-06:
> hm. ok, i'll need to see the logs again - when it does work properly (button press) and when it doesn't (web ui / HAP).

In my case both button press (RF remote) and HAP gives the closed/open/closed notifications.

My log:
_Log2

34033181272 shelly_main.cpp:554     Up 34033.11, HAP 0/1/12 ns 1, RAM: 23400/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34041117941 shelly_main.cpp:554     Up 34041.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34049118020 shelly_main.cpp:554     Up 34049.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34057118122 shelly_main.cpp:554     Up 34057.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34065117964 shelly_main.cpp:554     Up 34065.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34073117851 shelly_main.cpp:554     Up 34073.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34081117938 shelly_main.cpp:554     Up 34081.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34089117968 shelly_main.cpp:554     Up 34089.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34097117882 shelly_main.cpp:554     Up 34097.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34105117970 shelly_main.cpp:554     Up 34105.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34113117974 shelly_main.cpp:554     Up 34113.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34121117837 shelly_main.cpp:554     Up 34121.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34129118017 shelly_main.cpp:554     Up 34129.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34137118001 shelly_main.cpp:554     Up 34137.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34145117891 shelly_main.cpp:554     Up 34145.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34153117954 shelly_main.cpp:554     Up 34153.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34161118003 shelly_main.cpp:554     Up 34161.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34169117930 shelly_main.cpp:554     Up 34169.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34177117865 shelly_main.cpp:554     Up 34177.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34185117770 shelly_main.cpp:554     Up 34185.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34193117911 shelly_main.cpp:554     Up 34193.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34201117756 shelly_main.cpp:554     Up 34201.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34209117984 shelly_main.cpp:554     Up 34209.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34217117981 shelly_main.cpp:554     Up 34217.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34225117860 shelly_main.cpp:554     Up 34225.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34233117964 shelly_main.cpp:554     Up 34233.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34241117937 shelly_main.cpp:554     Up 34241.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34249117933 shelly_main.cpp:554     Up 34249.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34257118012 shelly_main.cpp:554     Up 34257.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34265117962 shelly_main.cpp:554     Up 34265.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34273117806 shelly_main.cpp:554     Up 34273.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34281117852 shelly_main.cpp:554     Up 34281.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34289118032 shelly_main.cpp:554     Up 34289.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34297118003 shelly_main.cpp:554     Up 34297.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34305117981 shelly_main.cpp:554     Up 34305.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34313117880 shelly_main.cpp:554     Up 34313.11, HAP 0/1/12 ns 1, RAM: 23088/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34321117881 shelly_main.cpp:554     Up 34321.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34329117984 shelly_main.cpp:554     Up 34329.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34337117887 shelly_main.cpp:554     Up 34337.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34345117945 shelly_main.cpp:554     Up 34345.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34353117885 shelly_main.cpp:554     Up 34353.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
[... cut ...]
34825117389 shelly_main.cpp:554     Up 34825.11, HAP 0/2/12 ns 2, RAM: 22460/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34833117595 shelly_main.cpp:554     Up 34833.11, HAP 0/2/12 ns 2, RAM: 22460/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34841117437 shelly_main.cpp:554     Up 34841.11, HAP 0/2/12 ns 2, RAM: 22460/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34849117610 shelly_main.cpp:554     Up 34849.11, HAP 0/2/12 ns 2, RAM: 22460/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1_

rojer, 2021-03-07:
@joddye2 it seems that your close sensor gives a pulse instead of maintaining a steady signal.
what kind of sensor is it?

joddye2, 2021-03-07:
@rojer Simple reed-relay. This one: https://www.tanealarm.com/products/surface-mount-contacts/fm-102/
Wiring similar as wiki: https://github.com/mongoose-os-apps/shelly-homekit/wiki/resources/GarageDoor-Shelly1.png

Before mongoose-os, I used with homebridge, and with that the notifications were ok (but nothing else :D)

grizz479, 2021-03-07:
For me alpha version is working properly after a day or so kinda “calibraring”?? First times gives wrong notifications bit after a fee times it works properly.
Can’t get tonthe log due to web ui connectivity  issues.

[image]

rojer, 2021-03-07:
@joddye2 hm. ok, let's walk through the logs step by step.

we are in a closed state when a HAP command to open comes:
```
34489 118127 shelly_main.cpp:554 Up 34489.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
34490 778556 shelly_output.cpp:58 Output 1: off -> on (GDO:open)
34490 789961 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
34490 801703 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
34491 088430 shelly_output.cpp:58 Output 1: on -> off (pulse_off)
```
this all takes about 2 seconds (timestamp on the left is in microseconds).
at this point we see transition of the sensor from 1 to 0 (with a bit of noise, but ok).
``` 
34491 490403 shelly_input.cpp:76 Input 1: change (state 1)
34491 581135 shelly_input.cpp:76 Input 1: change (state 0)
```
after 15 seconds (movement time) we transition to open state:
```
34497 117781 shelly_main.cpp:554 Up 34497.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
34505 117819 shelly_main.cpp:554 Up 34505.11, HAP 0/1/12 ns 1, RAM: 23404/37496; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
34506 843085 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> open (2 -> 0)
```
then we have about a minute of nothing much happening:
```
34513117551 shelly_main.cpp:554 Up 34513.11, HAP 0/1/12 ns 1, RAM: 23412/37496; st 0; 5.1: cur:open tgt:open cl:0 op:-1
34515026364 HAPPlatformTCPStrea:121 0x3fff6154 192.168.0.234:52957 Incoming HAP connection, ns 0/1/12
34515040481 HAPPlatformTCPStrea:276 0x3fff6154 192.168.0.234:52957 Accepted HAP connection, ns 0/2/12 ts 0x3fff5c5c
34517970024 esp_main.c:146 SDK: ip:192.168.0.248,mask:255.255.255.0,gw:192.168.0.1
34521117658 shelly_main.cpp:554 Up 34521.11, HAP 0/2/12 ns 2, RAM: 22964/37496; st 0; 5.1: cur:open tgt:open cl:0 op:-1
34529117483 shelly_main.cpp:554 Up 34529.11, HAP 0/2/12 ns 2, RAM: 22964/37496; st 0; 5.1: cur:open tgt:open cl:0 op:-1
34537117644 shelly_main.cpp:554 Up 34537.11, HAP 0/2/12 ns 2, RAM: 22964/37496; st 0; 5.1: cur:open tgt:open cl:0 op:-1
34545117698 shelly_main.cpp:554 Up 34545.11, HAP 0/2/12 ns 2, RAM: 22964/37496; st 0; 5.1: cur:open tgt:open cl:0 op:-1
34553117584 shelly_main.cpp:554 Up 34553.11, HAP 0/2/12 ns 2, RAM: 22964/37496; st 0; 5.1: cur:open tgt:open cl:0 op:-1
34561117521 shelly_main.cpp:554 Up 34561.11, HAP 0/2/12 ns 2, RAM: 22964/37496; st 0; 5.1: cur:open tgt:open cl:0 op:-1
```
when out of the blue sensor changes to 1 and we react by declaring that the door is closed
```
34567723120 shelly_input.cpp:76 Input 1: change (state 1)
34567771683 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
34567 782284 shelly_hap_garage_d:222 GDO 1: Cur State: open -> closed (0 -> 1)
```
except it then flips back to open almost immediately (50 ms later).
```
34567 826848 shelly_input.cpp:76 Input 1: change (state 0)
34567871840 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (ext)
34567 882731 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
```
and then to closed again, after 30 ms, and stays closed:
```
34567 916703 shelly_input.cpp:76 Input 1: change (state 1)
34568926645 shelly_input.cpp:76 Input 1: long (state 1)
34569117798 shelly_main.cpp:554 Up 34569.11, HAP 0/2/12 ns 2, RAM: 22956/37496; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
```
because we are in the opening state, after 7 seconds of waiting for close sensor to go to 0, we give up and declare the door closed.
```
34574 971791 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
34574 982711 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
```

the algorithm is working as intended, the problem here is the confusing behavior of the sensor.
why did it suddenly indicate "closed" when the door was open? was it closed via different means? why did it flip back to open then? is it just noise on the wires? maybe.

AG-dingoant, 2021-03-08:
Hi,
I did the update to alpha 5. I tested again and when the command comes from the shelly, the state remains wrong. When I tried as well to open the door with the normal button and it seems that the state is not updated anymore in on the shelly. In the past, the status was well updated when I pressed the normal button. If the sensor might be the problem, why is the sensor working properly whith the normal button?

I used a sensor from another door opener solution. As the shelly is configured in 220V and I connected the sensor to the door opener to get a low voltage and the feedback is on the shelly.

Here are my logs

179470567034 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179471504815 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179472509079 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179473510474 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179474071913 shelly_main.cpp:554     Up 179474.06, HAP 0/3/12 ns 3, RAM: 21764/37496; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
179474514606 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179475516531 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179476112759 mgos_http_server.c:180  0x3fff6ce4 HTTP connection from 192.168.0.60:54504
179476132533 mgos_ota_core.c:252     Starting, timeout 600, commit timeout 0, mem 18864
179476159362 mgos_ota_core.c:486     FW: switch1 esp8266 2.8.0-alpha5 20210307-174834/2.7.3-111-g06f6eb8-master
179476201709 shelly_main.cpp:365     == Stopping HAP service (1)
179476210762 mgos_ota_core.c:587     begin: waiting for user
179476227190 shelly_main.cpp:376     HAP server state: 2
179476242451 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5bdc nc 0x3fff5e04 0/2/12
179476255934 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff493c nc 0x3fff601c 0/1/12
179476269929 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff696c nc 0x3fff683c 0/0/12
179476279644 shelly_main.cpp:365     == Stopping HAP service (2)
179476288072 mgos_ota_core.c:587     begin: waiting for user
179476305411 HAPPlatformTCPStrea:105 0x3fff5704 0.0.0.0:9000 HAP connection closed, ns 0/0/12 i 1
179476317060 shelly_main.cpp:376     HAP server state: 0
179476327588 shelly_main.cpp:769     Starting firmware update
179476377592 esp_ota_backend.c:185   Slot 0, FW: switch1.bin -> 0x8000, FS fs.bin -> 0xbb000
179476395150 mgos_ota_core.c:504     0.18% total, esp_init_data_default_v08.bin 128 of 128
179476533080 esp_ota_backend.c:338   Start writing fs.bin (262144) @ 0xbb000
179476551245 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179476600846 mgos_ota_core.c:504     0.24% total, fs.bin 512 of 262144
179477554885 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179478555614 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179479560573 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179480309136 esp_ota_backend.c:376   Write finished, checksum ok
179480327312 mgos_ota_core.c:504     27.36% total, rboot.bin 512 of 2320
179480669572 esp_ota_backend.c:338   Start writing switch1.bin (699936) @ 0x8000
179480754961 mgos_ota_core.c:504     27.60% total, switch1.bin 512 of 699936
179480871442 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179481571349 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179482073817 shelly_main.cpp:554     Up 179482.06, HAP 0/0/12 ns 0, RAM: 19852/37496; st 0; disabled
179482577726 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179483578639 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179484339381 mgos_ota_core.c:504     54.71% total, switch1.bin 262656 of 699936
179484581617 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179485631328 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179486588889 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179487538483 mgos_ota_core.c:504     81.82% total, switch1.bin 524800 of 699936
179487603578 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179488616106 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179489643543 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179490135305 esp_ota_backend.c:376   Write finished, checksum ok
179490155404 shelly_main.cpp:554     Up 179490.15, HAP 0/0/12 ns 0, RAM: 20388/37496; st 0; disabled
179490165384 mgos_ota_core.c:630     Reached the end of archive
179490175562 esp_ota_backend.c:410   Resetting RF calibration data @ 0x1fb000
179490248696 esp_ota_backend.c:442   New rboot config: prev_rom: 1, current_rom: 0 current_rom addr: 0x8000, current_rom size: 699936, current_fs addr: 0xbb000, current_fs size: 262144
179490260257 mgos_ota_core.c:811     Update finished, result 1 (Update applied, rebooting)
179490277830 mgos_ota_core.c:854     Update requires reboot
179490285894 mgos_system.c:58        Rebooting in 500 ms
179490625777 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.60:54474
179490744065 esp_main.c:146          SDK: state: 5 -> 0 (0)
179490752026 esp_main.c:146          SDK: rm 0
179490760046 esp_main.c:146          SDK: pm close 7
179490773666 mgos_wifi.c:119         WiFi STA: Disconnected, reason: 8
179490783182 mgos_net.c:82           WiFi STA: disconnected
[... cut ...]
248002114 mgos_http_server.c:180  0x3fff6974 HTTP connection from 192.168.0.60:54541
248017219 mgos_http_server.c:187  0x3fff6974 GET /
249192636 shelly_main.cpp:554     Up 249.18, HAP 0/3/12 ns 3, RAM: 21372/37448; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
257192672 shelly_main.cpp:554     Up 257.18, HAP 0/3/12 ns 3, RAM: 21372/37448; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
265192908 shelly_main.cpp:554     Up 265.18, HAP 0/3/12 ns 3, RAM: 21372/37448; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1

rojer, 2021-03-09:
@AG-dingoant i don't see any input activity in the log whatsoever

we get a command to open, pulse output, then 7 seconds late sensor is still closed, so we assume door has not moved, give up and return to "closed" state (which is what the sensor is telling us).

```
82363383 shelly_output.cpp:58 Output 1: off -> on (GDO:open)
82376232 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
82387863 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
82419151 mgos_mongoose.c:66 New heap free LWM: 14440
82674847 shelly_output.cpp:58 Output 1: on -> off (pulse_off)
89192487 shelly_main.cpp:554 Up 89.18, HAP 0/3/12 ns 3, RAM: 21884/37448; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
89433111 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
89445532 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
```

AG-dingoant, 2021-03-17:
Here are the logs of a test I did this afternoon. It is wired because it seams that the sensor is not working properly. I couldn't see in my home app the status changing while using the wall button. I will check the wiring as well.

100268381976 shelly_debug.cpp:227    No log file, sending new entries
100268786239 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100269800911 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100270798479 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100271795764 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100272795190 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100273791747 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100274299190 shelly_main.cpp:560     Up 100274.29, HAP 0/3/12 ns 3, RAM: 23012/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100274790621 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100275791150 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100276786827 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100277795518 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100278784044 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.0.45:64629
100282299133 shelly_main.cpp:560     Up 100282.29, HAP 0/3/12 ns 3, RAM: 23608/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100290299096 shelly_main.cpp:560     Up 100290.29, HAP 0/3/12 ns 3, RAM: 23608/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100298299055 shelly_main.cpp:560     Up 100298.29, HAP 0/3/12 ns 3, RAM: 23608/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100306299014 shelly_main.cpp:560     Up 100306.29, HAP 0/3/12 ns 3, RAM: 23608/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100314299250 shelly_main.cpp:560     Up 100314.29, HAP 0/3/12 ns 3, RAM: 23608/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100318315574 shelly_output.cpp:58    Output 1: off -> on (GDO:open)
100318323302 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
100318330172 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
100318622374 shelly_output.cpp:58    Output 1: on -> off (pulse_off)
100322298926 shelly_main.cpp:560     Up 100322.29, HAP 0/3/12 ns 3, RAM: 23584/38552; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
100325380107 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
100325387291 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
100330299083 shelly_main.cpp:560     Up 100330.29, HAP 0/3/12 ns 3, RAM: 23560/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100335513637 shelly_output.cpp:58    Output 1: off -> on (GDO:open)
100335521329 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (HAPopen)
100335528206 shelly_hap_garage_d:222 GDO 1: Cur State: closed -> opening (1 -> 2)
100335820399 shelly_output.cpp:58    Output 1: on -> off (pulse_off)
100338299106 shelly_main.cpp:560     Up 100338.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:opening tgt:open cl:1 op:-1
100342580267 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
100342587404 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
100346299161 shelly_main.cpp:560     Up 100346.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100354298997 shelly_main.cpp:560     Up 100354.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100362299347 shelly_main.cpp:560     Up 100362.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100370298915 shelly_main.cpp:560     Up 100370.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100378299071 shelly_main.cpp:560     Up 100378.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100386299020 shelly_main.cpp:560     Up 100386.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100394298973 shelly_main.cpp:560     Up 100394.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100402298995 shelly_main.cpp:560     Up 100402.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100410298965 shelly_main.cpp:560     Up 100410.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100418299061 shelly_main.cpp:560     Up 100418.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100426299175 shelly_main.cpp:560     Up 100426.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100434298960 shelly_main.cpp:560     Up 100434.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100442299195 shelly_main.cpp:560     Up 100442.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1
100450299091 shelly_main.cpp:560     Up 100450.29, HAP 0/3/12 ns 3, RAM: 23920/38552; st 0; 5.1: cur:closed tgt:closed cl:1 op:-1

grizz479, 2021-03-18:
Update of situation: after working properly with alpha version I updated to stable  2.8.1 and 2.8.2. With both updates, it stopped getting the proper state of the close sensor, and only suddenly it worked for a little while yesterday, getting proper door state and updating with no issues. 
Today, once again, it has stopped receiving proper input from close sensor, and know I do not have a clue of how to manage it or where the problem comes frome.

The wiring and current are checked and the connection to homekit is up and running, too.

Any ideas?

grizz479, 2021-03-29:
Hi there!
Have been making some research and tests. I flashed the shelly to stock and have been trying as garage door opener with homebridge. And... the problem remains the same, with the same "chatter" of sensor state and, sometimes working properly with no clue of what changes these times. 

I wonder if it has to do with wifi connectivity? I use a custom "nest" wifi network but everything works properly in every floor.

joddye2, 2021-03-30:
> Hi there!
> Have been making some research and tests. I flashed the shelly to stock and have been trying as garage door opener with homebridge. And... the problem remains the same, with the same [...]

I used stock shelly + homebridge for 2-3 months and as I remember I didn't have this problem. 
Otherwise I thinking about we use inapropriate contact sensor.(?)

closed 2021-05-08 by github-actions[bot]

infiums, 2022-01-05:
Same problem. When everything is ok (door closed, sensor shows that door closed) I send command to open, status changes to opening, but after few seconds I receive notification that door has been closed (while gate moves to open). Movement time not passed and contact sensor magnet detached. Then it always show Thad door is closed, also when it’s move or opened.