mongoose-os-apps/shelly-homekit #452: New Feature: below Wh - Switch off

Rappelsack, 2021-01-30:
Hello

I do not know how to describe it correctly.
It would e a awesome feature it f.e. a Shelly Plug S could be set up to: below XY Wh, cut the current.

So I could plug my TV, activate the Shelly and when I set the TV to standby the current drops below a limit and after that the shelly switches off.

Hope u understand.

Kind regards.

andyblac, 2021-01-30:
plesse add your support to this issue https://github.com/mongoose-os-apps/shelly-homekit/issues/26, i'll close this one as duplicate thx.

Rappelsack, 2021-01-30:
Thanks!

I do not mean the power consumption readout in homekit.

I mean a simmilar function like „auto off delay“:
[image]

But „auto off if the consumption drops below 0,5W“ for example.

timoschilling, 2021-04-21:
@rojer what do you think about this issue? I think thats something which should be handelt by homekit after implementing #26

Rappelsack, 2021-05-01:
Any news about this?

I guess it should be not difficult to implement?

rojer, 2021-05-01:
well, homekit does not provide any means of exporting consumption, let alone controlling it...
there are no fundamental objections, but we need to persist consumption across reboots.
other than that, it would indeed be pretty easy to do.

andyblac, 2021-05-01:
> well, homekit does not provide any means of exporting consumption, let alone controlling it...

not exactly true, Home app desn't, but homekit does, via 3rd party app's (Eve, Home+, etc)

Rappelsack, 2021-05-01:
Well, it doesn't have to go over HomeKit. It's via the web GUI. Just like the auto-Off. 

That's how you could connect your TV. If the power consumption falls below XY watts, the Shelly turns off. It could be switched on again with a HomeKit button.

timoschilling, 2021-06-14:
@rojer are you still think we should implement this? After #690 this can be done by and HomeKit automatisation.

rojer, 2021-06-14:
but that's for eve only, isn't it?
i think it's still a good idea to have locally, since it's a safety mechanism as well.

timoschilling, 2021-06-14:
Not only for eve, there are many apps that can create automatisations for that.
Why is below power cutoff a safety mechanism? Keep in mind over power is requested in a different issue #339

rojer, 2021-06-14:
oh, sorry. indeed, i got confused, i thought this was for overpower protection.
but i guess if we are to implement the overpower cutoff, might as well add a check for below certain threshold.

Rappelsack, 2022-11-18:
I see that the development is a little bit behind? But is there any news about this function?
It would be so great to cut the current via Shelly S when the washing machine is ready (consumption below XY Wh) f.e.