mongoose-os-apps/shelly-homekit #454: Shelly2 Auto on

Marcusg562, 2021-01-31:
Hey guys, has anyone the same issue? Every morning my Shelly turn on at an specific time. There is no HomeKit automation set and the initial stage is for both switches off. Any ideas?

simonfield76, 2021-01-31:
Could you give more details?

Inviato da iPhone

Marcusg562, 2021-01-31:
What exactly do you wanna know? 

HAP Type: Switch
Input Mode: Toggle
Inverted input: off
Initial State: off
Auto Off: off

Version: 2.7.3 (build 20210124-130626/2.7.3-g883914a-master)

rojer, 2021-01-31:
there is nothing in the device that can possibly cause this, there is no scheduling functionality at all.

andyblac, 2021-01-31:
He has a poltergeist, playing games with him 😄

Marcusg562, 2021-01-31:
> there is nothing in the device that can possibly cause this, there is no scheduling functionality at all.
> 
> if you unpair the device from homekit, does it still do it?

I will test it tomorrow - since this only happens in the morning

Marcusg562, 2021-02-01:
It seems as this issue was caused by an Eve Button in my Homekit

closed 2021-02-01 by Marcusg562