mongoose-os-apps/shelly-homekit #456: Shelly 1 as impulse relay

ToblerCGN, 2021-01-31:
Hi folks! I'm using a Shelly 1 as a replacement for an impulse relay. It mostly works fine, when I manually switch on or off. When I use an automation to switch off the relay it switches off the light and instantly switches it on again. Because of this I can't use automations with it. Input mode is momentary.

andyblac, 2021-01-31:
what type of physical switch you using ? for momentary you need a momentary switch ie push button.

jsiegenthaler, 2021-01-31:
You mean an Eltako? I have a Shally1 wired up to the Eltako

andyblac, 2021-01-31:
in UK we also call them "RETRACTIVE". ie spring loaded

ToblerCGN, 2021-01-31:
I am using the Eltako S12 100 and would like to replace it. And I am using push buttons. The Shelly works fine, when I manually switch it on or off. When I use an automation to switch on, ist works fine as well. But when the automation says switch off, ist switches off and after a second ist switches on again.

andyblac, 2021-01-31:
i mean what type wall switch are you using ???

andyblac, 2021-01-31:
if you turn off manually in homekit does it turn on again ?

ToblerCGN, 2021-01-31:
When I turn it off manually in HomeKit, it doesn‘t turn on again.

andyblac, 2021-01-31:
then it sounds like your automation is at fault.

andyblac, 2021-01-31:
i have loads of shelly1's being used with automations, and have no issues.

andyblac, 2021-01-31:
can you post screenshots of your automation ?

ToblerCGN, 2021-01-31:
I am using automations and Shellys as well and they all work fine. But in this special wiring ist doesn‘t.

andyblac, 2021-01-31:
> I am using automations and Shellys as well and they all work fine. But in this special wiring ist doesn‘t.

can you post it maybe i can spot something ?

ToblerCGN, 2021-01-31:
I already deleted it and unwired the Shelly. But ist was an ordinary automation. On sunset switch on, on sunrise switch off, as ist did. But on sunrise and and other automation to switch off, ist switched off and after a second or so switched on again. I googled the problem and found another User on Youtube, who had the same problem.

ToblerCGN, 2021-01-31:
Maybe jsiegenthaler can contact me. He is using the Eltako as well.

andyblac, 2021-01-31:
did you not remove and replace it with the shelly ?, so then it just becomes a normal relay, nothig speacial. if it was switching on / off in homekit via manully press of the tile in homekit, then shelly is working correctly, then it down to HomeKit automation / HomeKit hub.

andyblac, 2021-03-03:
no responce closing...

closed 2021-03-03 by andyblac