mongoose-os-apps/shelly-homekit #457: Problem using Shelly 2.5 and 1 outside my house.

theflexa1, 2021-01-31:
Hello to everyone.

I can´t reach my Shllies from 4G... I can with wi-fi but I cannot reach them by 4G....

Is there something I am doing wrong?

andyblac, 2021-01-31:
do you have a homekit hub ?

theflexa1, 2021-01-31:
I have an Ipad, an Apple TV and a Homepod....

The strange thing... is that I set another Shelly 1 1 month ago... and i works with 4G...
Really rare...

andyblac, 2021-01-31:
should be no reason why then, check in home app that at least one hub is connected,

theflexa1, 2021-01-31:
I´ve already check that!

Conected is the status, but... I cannot think about what it´s wrong. In the photo you can see two Shelly 1. One instaled 1 month ago, the other today. Same process... (I think I did the same...)

[image]

theflexa1, 2021-01-31:
The only strange thing I have done is changing the name of the device....
[image]
[image]

theflexa1, 2021-01-31:
No, this last thing is not the problem.... I´ve just tested... no way to make it work outside the wi-fi :(

theflexa1, 2021-01-31:
More info:

I hace reset the firmware to original. The shelly app let me control the Shelly 1 and 2.5 from the app runing 4G. So the issue must be on the 2.7.3 of the new firmware of homekit...

If someone know how to fix this... just write hehe

andyblac, 2021-01-31:
can you enable debug log please, then reboot device, open log clicking on log link in webui on a pc/Mac, wait few mins for clients and hubs to connect. then try on 4g and copy paste log to a file and post here.

theflexa1, 2021-01-31:
The debug log... where can I access that in the homekit software?

theflexa1, 2021-01-31:
[image]
This is the log you ask me :)

simonfield76, 2021-01-31:
In the web server pointing the ip address of the shelly in the browser

Inviato da iPhone

andyblac, 2021-01-31:
look for the ip of the main hub, does it get a connection ?

theflexa1, 2021-01-31:
> look for the ip of the main hub, does it get a connection ?

The main hub... you mean the Apple TV?

andyblac, 2021-01-31:
> > look for the ip of the main hub, does it get a connection ?
> 
> The main hub... you mean the Apple TV?

tbh i wpuld disable homekit on appletv imo you will get nothing but problems, as you said you have homepod use that.

theflexa1, 2021-01-31:
I´m going to test that. ;)

> > > look for the ip of the main hub, does it get a connection ?
> > 
> > 
> > The main hub... you mean the Apple TV?
> 
> tbh i would disable homekit on appletv imo you will get nothing but problems, as you said you have homepod use that.

andyblac, 2021-01-31:
here is mine
[image]

andyblac, 2021-01-31:
> I´m going to test that. ;)
> 
> > > > look for the ip of the main hub, does it get a connection ?
> > > 
> > > 
> > > The main hub... you mean the Apple TV?
> > 
> > 
> > tbh i would disable homekit on appletv imo you will get nothing but problems, as you said you have homepod use that.

it can take 5 mins for a hub to change status, so give it time after disabling appletv, check in home app for a disabled status, then find the connected hub ip, make sure it gets a connection in the shelly log.

andyblac, 2021-03-03:
can this be closed ?

theflexa1, 2021-03-03:
Yes ! It was the apple tv

Atentamente, 

Álvaro Domínguez.

closed 2021-03-03 by andyblac

ma-jkl, 2021-03-04:
Hi, can you tell me how did you fix it? I have the same issue. Michal