mongoose-os-apps/shelly-homekit #461: Can't Inject RPC Event to stateless switch configured as Toggle

POFerro, 2021-02-04:
Hi :),

I'm trying to inject an event through rpc to a stateless switch to activate the associated homekit scenario. My stateless switch is configured as Toggle so I'd like to inject a single press event to homekit.
I tried every combination ( sending a kChange event, kSingle, kToggle, etc.) but nothing seems to work, my event is allways ignored, looking at the code it seems that kChange events are filtered directly in rpc handler and the stateless switch ignores events != kChange if the switch input mode is != kMomentary.
Is this a dead end? Is my scenario not supported or am I missing something?
Would be nice to be able to inject this type of event, can you help me?

Thanks and best regards
POFerro

andyblac, 2021-02-04:
show me the url you are using.

andyblac, 2021-02-04:
for more help see here: https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API

POFerro, 2021-02-04:
Hi @andyblac ,

Yes that was the documentation I was following.
Here is my url: http://192.168.1.109/rpc/Shelly.InjectInputEvent?id=1&event=1
I was hopping that this triggered the homekit event associated with single press but it doesn't.

I managed to get it working by: setting switch InputMode = Momentary->Sending event->setting switch back to toggle.
Seems like the button configured as toggle ignores the event.

Thanks and best regards
POFerro

andyblac, 2021-02-04:
@POFerro glad you got it working, all be it with a work around.

@rojer is this expected behaviour ? or could it be a bug ?

rojer, 2021-02-08:
this is working as intended but in your scenario i don't understand one thing: why do you have the stateless switch input set to toggle? it seems that just leaving it in momentary mode would allow you to inject events, so... why not?

POFerro, 2021-02-08:
Hi @rojer ,
That’s because my physical switch is of type toggle, if I configure it as momentary my wife kills me since the way the switch works becomes weirder for her 😂.
The system that injects the events is just a playground for me to be able to simulate the switch press from my pc, so it’s secondary to my wife’s needs ;).

Since it’s as designed, no worries, I’ll use the workaround.
Thanks and best regards.

andyblac, 2021-02-08:
ok, i'll close.

closed 2021-02-08 by andyblac

rojer, 2021-02-09:
@POFerro ok, i understand :)

we do not currently support change events injection... i don't remember exactly why, but it seemed not trivial to it correctly at the time and i didn't need it.