mongoose-os-apps/shelly-homekit #465: Shelly1 as Switch for my Alarm system

Steve1468, 2021-02-06:
Hello all, I use a Shelly1 to switch in and off my Alarm System, it works but I have to click to times in the HomeKit switch icon. Is there a possibility to configurate the Shelly1 that when it shows "on" on the icon that the alarm in on, and when it shows "off" on the icon that the alarm is off. All that with only one click?

jsiegenthaler, 2021-02-06:
You can do that with a simple Homekit automation:
When Alarm is on, turn on the switch
When Alarm is off, turn off the switch

Steve1468, 2021-02-06:
Yes, but my alarm needs two impulses not only one.

Inviato da iPad

jsiegenthaler, 2021-02-06:
Two pushed of a switch? Do that in the Automation
Two switches? Add another Shelly1

Steve1468, 2021-02-06:
> Two pushes of a switch? Do that in the Automation

That could be working.
What I saw now, the “sw” on my Shelly is not connected.
Maybe is that, otherwise I connect another Shelly or an rele 

Stefan
Cell: +39 3406492025
email: stefanlanger@mac.com
WhatsApp: +39 3406492025

http://www.thelangerbooks.com
Author of:

   and

andyblac, 2021-02-06:
@Steve1468 why do you need to click it 2 times ?   on, then off ?

andyblac, 2021-02-06:
sounds like this https://github.com/mongoose-os-apps/shelly-homekit/issues/406 if so i have made a request here https://github.com/mongoose-os-apps/shelly-homekit/issues/408, please add you support to it, and report back here, and i'll close this issue.

Steve1468, 2021-02-06:
Now it works with one click thanks to your info jsiegenthaler, problem is now the switch turns on and after some seconds turns off and remains always turned off, so I don't see in the status of HomeKit app if the alarm is on or not.

jsiegenthaler, 2021-02-06:
Adapt your Home Automation as you need to. You can do whatever you want in your automation.

andyblac, 2021-02-06:
@Steve1468 i think you need the option i posted above.

andyblac, 2021-02-06:
> Adapt your Home Automation as you need to. You can do whatever you want in your automation.

it can't be done with automations, he need "Inverted Output". ie relay open but shows in HomeKt as ON.

Steve1468, 2021-02-06:
Yes right.
Now I have status: off
HAP Type: switch
Input Mode: Momentary
Inverted input: enabled 
Initial state: off
Auto off: enabled 
Auto off delay: 0:00:00:01.100

No possibility right now for "inverted output"
Where can I find that?

Steve1468, 2021-02-06:
Do I have to update the Shelly1 Software?

andyblac, 2021-02-06:
> Yes right.
> Now I have status: off
> HAP Type: switch
> Input Mode: Momentary
> Inverted input: enabled
> Initial state: off
> Auto off: enabled
> Auto off delay: 0:00:00:01.100
> 
> No possibility right now for "inverted output"
> Where can I find that?

no, it is waiting to be merged, and then built, should not be long.

andyblac, 2021-02-06:
anyway, i'll close this issue, as it is a duplicate, pleaes add your support the other issue here https://github.com/mongoose-os-apps/shelly-homekit/issues/408

andyblac, 2021-02-06:
once it meged, and built, test and if it does not fix your issue, please feel free to reopen this issue.

closed 2021-02-06 by andyblac

Steve1468, 2021-02-06:
Okay thanks