mongoose-os-apps/shelly-homekit #467: Improve update  mode

AnakinSpain, 2021-02-06:
Will be possible to improve the way to update. I am not usually turn on my mac. 

Could it be possible? 

Maybe from one shelly to others...

andyblac, 2021-02-06:
sorry what do you mean?

andyblac, 2021-02-06:
from stock to this firmware ? or update this firmware to latest ?, all can be done on your phone.

AnakinSpain, 2021-02-06:
Update to latest. 

From my phone I enter every static IP in my browser and update from web gui. But I have 15 shellys

andyblac, 2021-02-06:
unforunatly there is no other way without your mac, with your mac there is a script that update all of them at once.

andyblac, 2021-02-06:
ok you could also try it on https://pyto.app, i managed to get it to run

[image]

it not free but only 2.99 for basic version, and free for 3 day to trial it.

andyblac, 2021-02-06:
i'll close this issue, as not much can be done.

closed 2021-02-06 by andyblac